RfidResearchGroup/proxmark3 #3: lf t55xx detect failing IF t55 is set to EM modulation

TomHarkness, 2018-08-14:
**When writing an EM ID to a t55 the command "lf t55 detect" no longer detects the modulation automatically**

**Command output below:**

pm3 --> lf sear
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

HID Prox TAG ID: 1122334455 (41514) - Format Len: 37bit - FC: 4643 - Card: 107050

[+] Valid HID Prox ID Found!

pm3 --> lf t5 det
Chip Type  : T55x7
Modulation : FSK2a
Bit Rate   : 4 - RF/50
Inverted   : Yes
Offset     : 31
Seq. Term. : No
Block0     : 0x00107060

pm3 --> lf em 410x_write 0F0368568B 1
Writing T55x7 tag with UID 0x0f0368568b (clock rate: 64)
pm3 --> #db# Started writing T55x7 tag ...
#db# Clock rate: 64
#db# Tag T55x7 written with 0xff83c03322a646e4

pm3 --> lf sear
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

EM410x  pattern found

EM TAG ID      : 0F0368568B

Possible de-scramble patterns
Unique TAG ID  : F0C0166AD1
HoneyWell IdentKey {
DEZ 8          : 06837899
DEZ 10         : 0057169547
DEZ 5.5        : 00872.22155
DEZ 3.5A       : 015.22155
DEZ 3.5B       : 003.22155
DEZ 3.5C       : 104.22155
DEZ 14/IK2     : 00064481678987
DEZ 15/IK3     : 001034014845649
DEZ 20/ZK      : 15001200010606101301
}
Other          : 22155_104_06837899
Pattern Paxton : 259822731 [0xF7C948B]
Pattern 1      : 9750181 [0x94C6A5]
Pattern Sebury : 22155 104 6837899  [0x568B 0x68 0x68568B]

[+] Valid EM410x ID Found!

pm3 --> lf t5 det
[!] Could not detect modulation automatically. Try setting it manually with 'lf t55xx config'
pm3 -->

Latest pull, clean build, RDV3 hardware.

RfidResearchGroup, 2018-09-09:
Try now.  It should work better.

RfidResearchGroup, 2018-09-11:
And now there is a new command  **lf t55xx deviceconfig**  which allows for custom setting of t55xx timings from client.  On RDV40 it is also persisted on flash memory.   No need to recompile every time in order to try different timings

RfidResearchGroup, 2018-09-13:
@TomHarkness   Did you have time to test?

TomHarkness, 2018-09-13:
I did some tests yesterday - still the same issue with stock timing settings. Will have more time to do some in depth testing tonight.

TomHarkness, 2018-09-13:
Just checked this in the original pm3 fork and the issue does not present with rdv4 hardware.

RfidResearchGroup, 2018-09-14:
I have tried it with RDV40,   using a T5577 configure to EM4105 ,    and using 
````` lf t55xx deviceconfig a 29 b 17 c 15 d 47 e 15 `````

and it works.

TomHarkness, 2018-09-15:
Will confirm when I can flash - issue with latest pull.

RfidResearchGroup, 2018-09-16:
you should be able to compile

TomHarkness, 2018-09-16:
Yeah this seemed to sort the issue. This might actually be better stock timings.

closed 2018-09-16 by RfidResearchGroup