RfidResearchGroup/proxmark3 #4: MacOS Install Instructions

miketweaver, 2018-08-14:
I followed the MacOS instructions, and they did not work.

```
$ brew tap RfidResearchGroup/proxmark3
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
openssl ✔                  azure-cli                  beast                      cquery                     geoipupdate                influxdb                   lxc                        protobuf-c                 wtf

==> Tapping rfidresearchgroup/proxmark3
Cloning into '/usr/local/Homebrew/Library/Taps/rfidresearchgroup/homebrew-proxmark3'...
remote: Counting objects: 5, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 5 (delta 0), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (5/5), done.
Tapped 1 formula (30 files, 28KB).
```

```
$ brew install proxmark3
Error: rfidresearchgroup/proxmark3/proxmark3 is a head-only formula
Install with `brew install --HEAD rfidresearchgroup/proxmark3/proxmark3`
```

```
$ brew install --HEAD proxmark3
==> Installing proxmark3 from rfidresearchgroup/proxmark3
Error: No available formula with the name "rfidresearchgroup/proxmark3/arm-none-eabi-gcc" (dependency of rfidresearchgroup/proxmark3/proxmark3)
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
```
I'm not sure what to do here.

TomHarkness, 2018-08-15:
Brew install seems broken. Looking in to it now. 

For now you can do manual install if you like. Try the below  steps and let me know how you go.

First clone the repository locally - your users home directory is good:

1. "mkdir ~/proxmark3"
2. "git clone https://github.com/RfidResearchGroup/proxmark3.git ~/proxmark3"

Install libs and dependencies:

1.  "brew tap nitsky/stm32"
2. " brew install readline libusb p7zip libusb-compat wget qt5 pkgconfig arm-none-eabi-gcc"

IF you have problems with #1 it's because that repo currently has an issue. It's easy to resolve but I have not had time to try and do a PR. Go here and follow my workaround for manual install if required:
https://github.com/nitsky/homebrew-stm32/issues/11#issuecomment-412529620

Now some path stuff that QT needs to have in place:

1.  "export PKG_CONFIG_PATH=/usr/local/Cellar/qt/5.11.1//lib/pkgconfig/"
2. "export QT_PKG_CONFIG_QT5CORE=$(find /usr -name Qt5Core.pc 2>/dev/null)"
3. "chmod 666 $QT_PKG_CONFIG_QT5CORE"
4. " echo "moc_location=\${prefix}/bin/moc" >> $QT_PKG_CONFIG_QT5CORE"
5. "chmod 444 $QT_PKG_CONFIG_QT5CORE"

Next link redline to avoid compilation errors:
"brew link --force readline"

Now to compile:

1. "cd ~/proxmark3"
2. "make clean; make"

Verify client works:
 
"./client/proxmark3 /dev/cu.usbmodem****" 

If you got this far without issues you might want to add in a bash alias that points the command "pm3" to the executable, and your likely good to flash the latest bootloader and main rom at this point:

Issue the command: "echo "'alias pm3='~/proxmark3/client/proxmark3'" >> ~/.bash_profile"

All commands are surrounded in quotes. Remove them from the beginning and end of each command before issuing.

RfidResearchGroup, 2018-08-15:
There was a missing file,   I have added it.  Try again :)

TomHarkness, 2018-08-15:
Confirmed to be working. Just tested.

miketweaver, 2018-08-15:
```
brew install --HEAD rfidresearchgroup/proxmark3/proxmark3
```
Confirmed working as well. Thank you.

closed 2018-08-15 by miketweaver