RfidResearchGroup/proxmark3 #5: LF StandAlone mode is broken. Field active but no tag read

TomHarkness, 2018-08-15:
This one should be easy to fix but I'm unsure where to look sorry!

RfidResearchGroup, 2018-08-15:
too little information to be useful for debugging purposes.
...which standalone mode?

TomHarkness, 2018-08-15:
SamyRun - default. No others tested so far, I'll get back with more info asap.

TomHarkness, 2018-08-31:
Tested all LF StandAlone. None work, field active as I have LED test cards and can see easily. But no data read or played back.

RfidResearchGroup, 2018-09-01:
hm.. lf sim seems to be off ..   Lets see if there was a update to fpga images..

RfidResearchGroup, 2018-09-01:
nop,    right now its FSK based simulation that is bad like these
lf hid sim
lf pyramid sim

TomHarkness, 2018-09-02:
Check:

`https://github.com/iceman1001/proxmark3/issues/239`

RfidResearchGroup, 2018-09-05:
This one should be fixed now.    My LF samyrun works as expected.  @TomHarkness  Would you mind verify it?

TomHarkness, 2018-09-06:
Confirmed working.

closed 2018-09-06 by TomHarkness