RfidResearchGroup/proxmark3 #6: Bug and partial Fix - update.sh for OS X users

TomHarkness, 2018-08-15:
**Update.sh does not find pm3 ion osx due to the "?" switch not working**

Along with the above we have no symlink in place and I think it may be good to put one in.

function wait4proxmark {
	echo >&2 "Waiting for Proxmark to appear..."
	while [ ! -c /dev/ttyACM? -a ! -L /dev/pm3-? ]; do
		sleep .1
	done
	local PM3=`ls -1 /dev/pm3-? /dev/ttyACM? 2>/dev/null | head -1`
	echo >&2 -e "Found proxmark on ${PM3}\n"
	echo $PM3
}

**The following works fine but line 5 also throws a missing ']' error.** Not sure about linux yet. Maybe it could be formatted better?

function wait4proxmark {
	echo >&2 "Waiting for Proxmark to appear..."
	while [ ! -c /dev/ttyACM? || /dev/tty.usbmodem* -a ! -L /dev/pm3-? ||  /dev/pm3* ]; do
		sleep .1
	done
	local PM3=`ls -1 /dev/pm3-? /dev/ttyACM? dev/tty.usbmodem* /dev/pm3*  2>/dev/null | head -1`
	echo >&2 -e "Found proxmark on ${PM3}\n"
	echo $PM3
}

Also I think somewhere it would be good to check OS, detect OS X and put some symlinks in place. Working code:

 	if [[ $OSTYPE == darwin* ]]; then
		echo "I am a Mac - creating symlink to /dev/pm3"
			ln -s $PM3 /dev/pm3
		;fi

TomHarkness, 2018-08-16:
Seems like the best structure to make this cross compatible with Linux and OS X without having issues with conditional OR is something like:

	while [ ! -c /dev/ttyACM? || /dev/tty.usbmodem* ] -a [ ! -L  /dev/pm3-? || /dev/pm3* ] ; do

This works but still throws missing '[' error.

RfidResearchGroup, 2018-08-16:
@joanbono   just made solution for one script over at iceman fork.   Its been merged.

No need to over complicate things.   I would also suggest implementing the same dual idea into the proxmark.sh  script since its uses the look up comport aswell inside.

and don't forget to add a entry in changelog.md   for eternal glory.

@TomHarkness  said something about making a symbolic link  to /dev/pm3 as it is done for linux ppl with udev rules.  (sudo make udev)    making things a bit easier for future support questions about my comport is not showing.

TomHarkness, 2018-08-16:
I'll do what I can to get a PR ready for this. Just currently stuck on this line throwing a missing ']' error. I'm sure its simple but there is something regarding syntax that I just don't quite understand yet. 

while [ ! -c /dev/ttyACM? || /dev/tty.usbmodem* ] -a [ ! -L  /dev/pm3-? || /dev/pm3* ] ; d

If someone figures out how to get this line formatted correctly so that no error is thrown that would be very helpful., Apart from that everything else including OS detection / symlink is simple.

At the moment with the following script that I am using for testing:

#!/bin/bash

function wait4proxmark {
	echo >&2 "Waiting for Proxmark to appear..."
	while [ ! -c /dev/ttyACM? || /dev/tty.usbmodem* ] -a [ ! -L  /dev/pm3-? || /dev/pm3* ]; do
		sleep .1
	done
	local PM3=`ls -1 /dev/pm3-? /dev/ttyACM? /dev/tty.usbmodem* /dev/cu.usbmodem* 2>/dev/null | head -1`
	echo >&2 -e "Found proxmark on ${PM3}\n"
	echo $PM3
}
$(wait4proxmark)
# flash bootroom & system image
#client/flasher $(wait4proxmark) -b bootrom/obj/yiuopt[hjkootrom.elf armsrc/obj/fullimage.elf

I get the below output: 

$ sudo ./update.sh
Waiting for Proxmark to appear...
./update.sh: line 5: [: missing `]'
./update.sh: line 5: /dev/tty.usbmodem881: Permission denied
./update.sh: line 5: /dev/pm3*: No such file or directory
Found proxmark on /dev/cu.usbmodem881

./update.sh: line 12: /dev/cu.usbmodem881: Permission denied

**Ignore the permission issue. I know exactly what that relates to.**

TomHarkness, 2018-08-16:
My thoughts are that we have two options here.

**Option #1** 

Do OS detect at the beginning of update.sh (BEFORE device discovery), create symlink and issue the correctly formatted commands based on what OS is detected.

**Option #2**

Format the commands so that regardless of what OS is running they are compatible and throw no errors. Do OS detect after device discovery and only use OS detection for creating a symlink rather than having seperate statements for different  OS's.

Does this make sense? What are peoples thoughts on this and what would be the better option in the long term?

Thanks guys!

joanbono, 2018-08-16:
Hi @RfidResearchGroup  I can do the same for the proxmark.sh script, it wouldn't take me more than 2 or 3 minutes.

RfidResearchGroup, 2018-08-16:
I have tested on ubuntu16.04
- update.sh
- proxmark3sh

Went as smooth as I ever could wish for.  Really simple.   No more looking for /dev...

RfidResearchGroup, 2018-08-16:
function name in proxmark.sh is wrong...
you forgot a _linux..

RfidResearchGroup, 2018-08-16:
the libcanberra gtk needed for ubuntu 18.04 ...

joanbono, 2018-08-17:
@RfidResearchGroup libcanberra needed to update it, you mean??? :/

RfidResearchGroup, 2018-08-17:
for install.sh,  there were a PR where 18.04 needs libcanberragtk in order to compile the source. 
I haven't use install.sh  since I do it manually when I have to set a dev-env

joanbono, 2018-08-17:
Working on it 👍

TomHarkness, 2018-08-17:
Three others that I know with varying OS X versions are reporting both scripts working. Nice!

closed 2018-08-17 by TomHarkness