RfidResearchGroup/proxmark3 #14: iClass chk seems to be broken

TomHarkness, 2018-08-17:
Valid key is definitely in the dictionary as seen from the below output:

pm3 --> hf iclass chk f client/default_keys.dic
[+] Reading tag CSN
[+] Loaded 88  keys from client/default_keys.dic
[+] Generating diversified keys, MAC
[+] Searching for DEBIT key
[+] Tag info
[+] CSN     | 97 AD 1E 01 F8 FF 12 E0
[+] CCNR    | FE FF FF FF FF FF FF FF 00 00 00 00
-----+------------------+---------
#key | key              | mac
-----+------------------+---------
[ 0] | 0000ffffffffffff | e341d798
[ 1] | 0000000000000000 | 2510f8ce
[ 2] | 0000a0a1a2a3a4a5 | faaed79d
[ 3] | 0000d3f7d3f7d3f7 | 332a78ae
[ 4] | 00005a1b85fce20a | 8d6603cd
[ 5] | 000000000ffe2488 | 0138fa4f
[ 6] | 00000297927c0f77 | 32d0e015
[ 7] | 000026940b21ff5d | 172dd122
[ 8] | 00002ba9621e0a36 | 7b594a4f
[ 9] | 0000434f4d4d4f41 | cbd798ff
[+] ... skip printing the rest
.....
[-] Chunk [0/88] : 11.7s [debit]

[+] Time in iclass checkkeys: 12 seconds

pm3 --> hf iclass dump k AEA684A6DAB23278
.------+--+-------------------------+

CSN   |00| 97 AD 1E 01 F8 FF 12 E0 |

------+--+-------------------------+

      |01| 12 FF FF FF 7F 1F FF 3C | .......<

      |02| FE FF FF FF FF FF FF FF | ........

      |03| 99 0B 05 0E 84 A6 B3 0A | ........

      |04| FF FF FF FF FF FF FF FF | ........

      |05| FF FF FF FF FF FF FF FF | ........

      |06| 03 03 03 03 00 03 E0 17 | ........

      |07| 2E D7 86 81 51 19 7B 36 | ....Q.{6

      |08| 2A D4 C8 21 1F 99 68 71 | *..!..hq

      |09| 2A D4 C8 21 1F 99 68 71 | *..!..hq

      |0A| FF FF FF FF FF FF FF FF | ........

      |0B| FF FF FF FF FF FF FF FF | ........

      |0C| FF FF FF FF FF FF FF FF | ........

      |0D| FF FF FF FF FF FF FF FF | ........

      |0E| FF FF FF FF FF FF FF FF | ........

      |0F| FF FF FF FF FF FF FF FF | ........

      |10| FF FF FF FF FF FF FF FF | ........

      |11| FF FF FF FF FF FF FF FF | ........

      |12| FF FF FF FF FF FF FF FF | ........

------+--+-------------------------+

[+] saving dump file - 19 blocks read
[+] saved 152 bytes to binary file iclass_tagdump-97ad1e01f8ff12e0-1.bin
pm3 -->

RfidResearchGroup, 2018-08-17:
...that would be because someone is using a dictionary file with mifare classic keys instead of a dictionary file with iClass keys.

closed 2018-08-17 by RfidResearchGroup

TomHarkness, 2018-08-18:
Oh shoot my bad, how did I miss that. Need to catch up on sleep I think!