RfidResearchGroup/proxmark3 #19: LF broken again - not rdv4 specific, all hardware.

TomHarkness, 2018-08-29:
Only happening with this firmware. original and iceman are ok. See below: 

pm3 --> hw tune

[=] measuring antenna characteristics, please wait...

....

[+] LF antenna: 67.13 V - 125.00 kHz
[+] LF antenna: 41.50 V - 134.00 kHz
[+] LF optimal: 69.11 V - 126.32 kHz
[+] LF antenna is OK

[+] HF antenna: 47.02 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

No GUI in this build!
pm3 --> lf sear
#db# transfer to client failed ::  | bytes between 0 - 512 (512)
#db# transfer to client failed ::  | bytes between 512 - 1024 (512)
#db# transfer to client failed ::  | bytes between 1024 - 1536 (512)
#db# transfer to client failed ::  | bytes between 1536 - 2048 (512)
#db# transfer to client failed ::  | bytes between 2048 - 2560 (512)
#db# transfer to client failed ::  | bytes between 2560 - 3072 (512)
#db# transfer to client failed ::  | bytes between 3072 - 3584 (512)
#db# transfer to client failed ::  | bytes between 3584 - 4096 (512)
#db# transfer to client failed ::  | bytes between 4096 - 4608 (512)
#db# transfer to client failed ::  | bytes between 4608 - 5120 (512)
#db# transfer to client failed ::  | bytes between 5120 - 5632 (512)
#db# transfer to client failed ::  | bytes between 5632 - 6144 (512)
#db# transfer to client failed ::  | bytes between 6144 - 6656 (512)
#db# transfer to client failed ::  | bytes between 6656 - 7168 (512)
#db# transfer to client failed ::  | bytes between 7168 - 7680 (512)
#db# transfer to client failed ::  | bytes between 7680 - 8192 (512)
#db# transfer to client failed ::  | bytes between 8192 - 8704 (512)
#db# transfer to client failed ::  | bytes between 8704 - 9216 (512)
#db# transfer to client failed ::  | bytes between 9216 - 9728 (512)
#db# transfer to client failed ::  | bytes between 9728 - 10240 (512)
#db# transfer to client failed ::  | bytes between 10240 - 10752 (512)
#db# transfer to client failed ::  | bytes between 10752 - 11264 (512)
#db# transfer to client failed ::  | bytes between 11264 - 11776 (512)
#db# transfer to client failed ::  | bytes between 11776 - 12288 (512)
#db# transfer to client failed ::  | bytes between 12288 - 12800 (512)
#db# transfer to client failed ::  | bytes between 12800 - 13312 (512)
#db# transfer to client failed ::  | bytes between 13312 - 13824 (512)
#db# transfer to client failed ::  | bytes between 13824 - 14336 (512)
#db# transfer to client failed ::  | bytes between 14336 - 14848 (512)
#db# transfer to client failed ::  | bytes between 14848 - 15360 (512)
#db# transfer to client failed ::  | bytes between 15360 - 15872 (512)
#db# transfer to client failed ::  | bytes between 15872 - 16384 (512)
#db# transfer to client failed ::  | bytes between 16384 - 16896 (512)
#db# transfer to client failed ::  | bytes between 16896 - 17408 (512)
#db# transfer to client failed ::  | bytes between 17408 - 17920 (512)
#db# transfer to client failed ::  | bytes between 17920 - 18432 (512)
#db# transfer to client failed ::  | bytes between 18432 - 18944 (512)
#db# transfer to client failed ::  | bytes between 18944 - 19456 (512)
#db# transfer to client failed ::  | bytes between 19456 - 19968 (512)
#db# transfer to client failed ::  | bytes between 19968 - 20480 (512)
#db# transfer to client failed ::  | bytes between 20480 - 20992 (512)
#db# transfer to client failed ::  | bytes between 20992 - 21504 (512)
#db# transfer to client failed ::  | bytes between 21504 - 22016 (512)
#db# transfer to client failed ::  | bytes between 22016 - 22528 (512)
#db# transfer to client failed ::  | bytes between 22528 - 23040 (512)
#db# transfer to client failed ::  | bytes between 23040 - 23552 (512)
#db# transfer to client failed ::  | bytes between 23552 - 24064 (512)
#db# transfer to client failed ::  | bytes between 24064 - 24576 (512)
#db# transfer to client failed ::  | bytes between 24576 - 25088 (512)
#db# transfer to client failed ::  | bytes between 25088 - 25600 (512)
#db# transfer to client failed ::  | bytes between 25600 - 26112 (512)
#db# transfer to client failed ::  | bytes between 26112 - 26624 (512)
#db# transfer to client failed ::  | bytes between 26624 - 27136 (512)
#db# transfer to client failed ::  | bytes between 27136 - 27648 (512)
#db# transfer to client failed ::  | bytes between 27648 - 28160 (512)
#db# transfer to client failed ::  | bytes between 28160 - 28672 (512)
#db# transfer to client failed ::  | bytes between 28672 - 29184 (512)
#db# transfer to client failed ::  | bytes between 29184 - 29696 (512)
#db# transfer to client failed ::  | bytes between 29696 - 30000 (304)
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

HID Prox TAG ID: 20041400cd (102) - Format Len: 26bit - FC: 10 - Card: 102

[+] Valid HID Prox ID Found!

#db# transfer to client failed ::  | bytes between 0 - 512 (512)
#db# transfer to client failed ::  | bytes between 512 - 1024 (512)
#db# transfer to client failed ::  | bytes between 1024 - 1536 (512)
#db# transfer to client failed ::  | bytes between 1536 - 2048 (512)
#db# transfer to client failed ::  | bytes between 2048 - 2560 (512)
#db# transfer to client failed ::  | bytes between 2560 - 3072 (512)
#db# transfer to client failed ::  | bytes between 3072 - 3584 (512)
#db# transfer to client failed ::  | bytes between 3584 - 4096 (512)
#db# transfer to client failed ::  | bytes between 4096 - 4608 (512)
#db# transfer to client failed ::  | bytes between 4608 - 5120 (512)
#db# transfer to client failed ::  | bytes between 5120 - 5632 (512)
#db# transfer to client failed ::  | bytes between 5632 - 6000 (368)
#db# transfer to client failed ::  | bytes between 0 - 512 (512)
#db# transfer to client failed ::  | bytes between 512 - 1024 (512)
#db# transfer to client failed ::  | bytes between 1024 - 1536 (512)
#db# transfer to client failed ::  | bytes between 1536 - 2048 (512)
#db# transfer to client failed ::  | bytes between 2048 - 2560 (512)
#db# transfer to client failed ::  | bytes between 2560 - 3072 (512)
#db# transfer to client failed ::  | bytes between 3072 - 3584 (512)
#db# transfer to client failed ::  | bytes between 3584 - 4096 (512)
#db# transfer to client failed ::  | bytes between 4096 - 4608 (512)
#db# transfer to client failed ::  | bytes between 4608 - 5120 (512)
#db# transfer to client failed ::  | bytes between 5120 - 5632 (512)
#db# transfer to client failed ::  | bytes between 5632 - 6144 (512)
#db# transfer to client failed ::  | bytes between 6144 - 6656 (512)
#db# transfer to client failed ::  | bytes between 6656 - 7168 (512)
#db# transfer to client failed ::  | bytes between 7168 - 7679 (511)
pm3 -->

RfidResearchGroup, 2018-08-29:
Dang,  one of my upcoming changes got pushed to repo..

https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/appmain.c#L1285

Three lines needs to be reverted back.

RfidResearchGroup, 2018-08-29:
Should work now.   Check and close!

TomHarkness, 2018-08-29:
Confirmed - fixed.

closed 2018-08-29 by TomHarkness