RfidResearchGroup/proxmark3 #26: EM Marine card read is broken

Defensor7, 2018-09-06:
**Describe the bug**
EM Marine card read is broken after last commits.

**Without card attached:**
pm3 --> lf em 410x_read
LF Signal properties:          
  high..........15          
  low...........-7          
  mean..........2          
  amplitude.....13          
  is Noise......No          
  THRESHOLD noice amplitude......10          
[#] DEBUG: (setClockGrid) demodoffset 0, clk 0          
[#] DEBUG: (ASKDemod_ext) Bitlen from grphbuff: 8192          
getHiLo fuzzed: High 37 | Low 20          
getHiLo fuzzed: High 43 | Low 26          
getHiLo fuzzed: High 43 | Low 26          
DEBUG: (askdemod_ext) CLEAN: startIdx 0, alignPos 0          
[#] DEBUG: (ASKDemod_ext) No data found errors:-1, invert:0, bitlen:0, clock:8 

**With card attached:**
pm3 --> lf em 410x_read
LF Signal properties:          
  high..........125          
  low...........-128          
  mean..........-5          
  amplitude.....130          
  is Noise......No          
  THRESHOLD noice amplitude......10          
[#] DEBUG: (setClockGrid) demodoffset 0, clk 0          
[#] DEBUG: (ASKDemod_ext) Bitlen from grphbuff: 8192          
getHiLo fuzzed: High 125 | Low -76          
getHiLo fuzzed: High 125 | Low -64          
getHiLo fuzzed: High 125 | Low -64          
DEBUG: (askdemod_ext) CLEAN: startIdx 0, alignPos 0          
[#] DEBUG: (ASKDemod_ext) No data found errors:-1, invert:0, bitlen:0, clock:32 

pm3 --> hw version

Proxmark3 RFID instrument
          

 [ CLIENT ]          
 client: iceman build for RDV40 with flashmem; smartcard;  
          
 [ ARM ]
 bootrom: iceman/master/ 2018-09-05 14:29:59
      os: iceman/master/ 2018-09-06 10:17:49

 [ FPGA ]
 LF image built for 2s30vq100 on 2017/10/25 at 19:50:50
 HF image built for 2s30vq100 on 2018/ 8/10 at 11:48:34          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 242267 bytes (46) Free: 282021 bytes (54)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory  

pm3 --> hw status
#db# Memory          
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing          
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image          
#db#   mode.................... LF image built for 2s30vq100 on 2017/10/25 at 19:50:50          
#db# Flash memory          
#db#   init....................OK          
#db#   Memory size.............2 mbits / 256kb          
#db#   Unique ID...............0xd567a882a7bb8e25          
#db# Smart card module (ISO 7816)          
#db#   version.................v2.06          
#db# LF Sampling config          
#db#   [q] divisor.............95 (125 KHz)          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# USB Speed          
#db#   Sending USB packets to client...          
#db#   Time elapsed............1500ms          
#db#   Bytes transferred.......780800          
#db#   USB Transfer Speed PM3 -> Client = 520533 Bytes/s          
#db# Various          
#db#   MF_DBGLEVEL.............1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mods          
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)   

pm3 --> data tune
          
[=] measuring antenna characteristics, please wait...
          
...
          
[+] LF antenna: 57.64 V - 125.00 kHz          
[+] LF antenna: 42.77 V - 134.00 kHz          
[+] LF optimal: 61.47 V - 127.66 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 48.01 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

RfidResearchGroup, 2018-09-06:
How close was the card to the antenna,   I have noticed a problem when I have my card direct on antenna but with distance it becomes good again

Defensor7, 2018-09-06:
No result at any distance.
HID card works.
Em Marine card does't.

Last Iceman's fork works well.

RfidResearchGroup, 2018-09-09:
Try latest source.,  same as #3

RfidResearchGroup, 2018-09-13:
@Defensor7   have you had the chance to test?

Defensor7, 2018-09-13:
Yes, it works now.

closed 2018-09-13 by RfidResearchGroup