RfidResearchGroup/proxmark3 #28: Unable to build

miketweaver, 2018-09-07:
While installing to Mac, it throws this error based on the commit added 20 minutes ago.

```
gcc -MT obj/cmdhflegic.o -MMD -MP -MF obj/cmdhflegic.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I../common/polarssl -I../zlib -I../uart -I/opt/local/include -I../liblua -Wall -g -O3 -DHAVE_GUI -DWITH_FLASH -DWITH_SMARTCARD -DWITH_FPC -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/cmdhflegic.o cmdhflegic.c
cmdhflegic.c:506:25: error: expected expression
        if ( strlen(Cmd) = 0 | |cmdp == 'h' ) return usage_legic_sim();
                               ^
1 error generated.
make[1]: *** [obj/cmdhflegic.o] Error 1
make: *** [client/all] Error 2
```

https://github.com/RfidResearchGroup/proxmark3/commit/b7d0786ab818ab277f4923d6e044bd1e6c93e3a8#diff-4a54e359a85cf26ddb848e4b65dd2cd3R506

TomHarkness, 2018-09-08:
I'm not getting this with the latest pull.

iceman1001, 2018-09-08:
pull latest..   its fixed already :)

closed 2018-09-08 by RfidResearchGroup