RfidResearchGroup/proxmark3 #29: i2c  potential hang issue

RfidResearchGroup, 2018-09-13:
From:  https://github.com/iceman1001/proxmark3/issues/247
Sometime it is useful to be able to unhang the i2c.   Normally would be unplug pm3 but that is a too blunt method to use.  

Link to suggested code that deals with a hung i2c.
https://github.com/merlokk/SmartHome/blob/master/ESP8266CO2PM25/ESP8266CO2PM25.ino#L75

@cjbrigato   any takers on this?

TomHarkness, 2018-09-13:
Friend of mine is looking in to it right now.

RfidResearchGroup, 2018-09-13:
Great!

iceman1001, 2018-10-27:
@merlokk   Its better you answer in this thread than over at the iceman fork repo where I posted.

merlokk, 2018-10-27:
yes)
hi) it may be problem... in some rare cases but....
ill fix it in near future after end fido part.

merlokk, 2018-11-14:
close? e4cb74fedb165ea5d24e1b567d88dd3d29d63893

merlokk, 2018-11-14:
by code it should work.
if you have proxmark wo i2c module...
it needs a resistor to ground on any i2c pin to check if it works

closed 2019-01-06 by iceman1001