RfidResearchGroup/proxmark3 #30: No USB connection after update

Peterthegreat, 2018-09-14:
Hi everyone,

I was running my proxmark3 rdv 4 for the second time and decided to follow the instructions for this repo, so I proceeded with the image upgrade.

[ 6552.518021] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
[ 6570.767352] usb 1-3: USB disconnect, device number 77
[ 6573.085096] usb 1-3: new full-speed USB device number 78 using xhci_hcd
[ 6578.237065] usb 1-3: device descriptor read/64, error -110
[ 6583.493272] usb 1-3: New USB device found, idVendor=9ac4, idProduct=4b8f
[ 6583.493274] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6583.493276] usb 1-3: Product: PM3 Device
[ 6583.493277] usb 1-3: Manufacturer: proxmark.org
[ 6583.493278] usb 1-3: SerialNumber: 888
[ 6583.493814] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
[ 6689.121873] usb 1-3: USB disconnect, device number 78
[ 6691.348203] usb 1-3: new full-speed USB device number 79 using xhci_hcd
[ 6696.712160] usb 1-3: new full-speed USB device number 80 using xhci_hcd
[ 6702.020367] usb 1-3: New USB device found, idVendor=9ac4, idProduct=4b8f
[ 6702.020370] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6702.020371] usb 1-3: Product: PM3 Device
[ 6702.020372] usb 1-3: Manufacturer: proxmark.org
[ 6702.020373] usb 1-3: SerialNumber: 888
[ 6702.020907] cdc_acm 1-3:1.0: ttyACM1: USB ACM device
root@kali:~/proxmark3RDV4# client/flasher /dev/ttyACM1 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf
Loading ELF file bootrom/obj/bootrom.elf
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000d60->0x00000d60) [R X] @0x298

Loading ELF file armsrc/obj/fullimage.elf
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00037900->0x00037900) [R X] @0x94
1: V 0x00200000 P 0x00139900 (0x000012bc->0x000012bc) [RW ] @0x37994
Note: Extending previous segment from 0x37900 to 0x38bbc bytes

[+] Waiting for Proxmark to appear on /dev/ttyACM1           
..........................................[=] UART Setting serial baudrate 115200 [FPC enabled]
.Found 
Entering bootloader... 
(Press and release the button only to abort)
[+] Waiting for Proxmark to appear on /dev/ttyACM1           
......................[=] UART Setting serial baudrate 115200 [FPC enabled]
. Found 

Flashing... 
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks].OK 
 0x00100200..0x00100f5f [0xd60 / 7 blocks].......OK 

Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0013abbb [0x38bbc / 454 blocks].......................................................Foo 128 | 128 (will loop)          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          

And then I waited for about 10 minutes with no change. After reconnecting, all I can see on dmesg | grep -i usb
[ 6949.214132] usb 1-3: new full-speed USB device number 88 using xhci_hcd
[ 6966.117994] usb 1-3: new full-speed USB device number 89 using xhci_hcd
[ 6983.021859] usb 1-3: new full-speed USB device number 90 using xhci_hcd
[ 6999.929716] usb 1-3: new full-speed USB device number 91 using xhci_hcd
[ 7016.833577] usb 1-3: new full-speed USB device number 92 using xhci_hcd
[ 7033.737437] usb 1-3: new full-speed USB device number 93 using xhci_hcd

I did received some warnings at first for a armsrc file
[image]

Any help is apreciated

TomHarkness, 2018-09-15:
Hmm, I'm hesitant to flash now that I see this. I just opened another issue but I think its the same but I am running OS X.

RfidResearchGroup, 2018-09-15:
@Peterthegreat    Does the latest PR fix your issue?

Peterthegreat, 2018-09-15:
Not yet, still trying to upload the image. How much time should I press the power button? If i do it too quickly, it says 

Waiting for Proxmark to appear on /dev/ttyACM0           
...............................[=] UART Setting serial baudrate 115200 [FPC enabled]
.Found 
[!] sending bytes to proxmark failed     

Here is the current status, after rebuild.
[image]

Peterthegreat, 2018-09-15:
OK, so I noticed a problem. When entering bootloader mode (maybe even just connecting to USB -as I see the same blinking pattern on the same time interval- ), the device resets after about 20 seconds (maybe a watchdog timer?)

>[31974.321822] usb 1-2: new full-speed USB device number 11 using xhci_hcd
[31974.463357] usb 1-2: New USB device found, idVendor=9ac4, idProduct=4b8f
[31974.463359] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[31974.463360] usb 1-2: Product: proxmark3
[31974.463361] usb 1-2: Manufacturer: proxmark.org
[31974.463362] usb 1-2: SerialNumber: iceman
[31974.463848] cdc_acm 1-2:1.0: ttyACM1: USB ACM device
[31993.220571] usb 1-2: USB disconnect, device number 11

And I guess it does not have enough time to upload the image to the FPGA

RfidResearchGroup, 2018-09-15:
you need to fix the modemmanger  / udev rules...

Peterthegreat, 2018-09-16:
Ugh, you are right. For some reason I skipped that step. Really sorry about that

But even after a logout and a OS restart and rebuilding, I still have the exact same problem.

RfidResearchGroup, 2018-09-16:
now you flash fullimage using the button trick.
press button,  keep it pressed, while you flash fullimage,   once finished,  release button, wait for restart of device.   Its a one-hand operation :)

TomHarkness, 2018-09-16:
KEEP THE BUTTON HELD THE ENTIRE TIME!!!! Before and while plugging in, then while flashing and then until the entire process of flashing is done. Seriously do not let the button go!!!!

iceman1001, 2018-09-18:
Kali,   there seem to be issues with kali.   if @Peterthegreat could test on a ubuntu / windows OS

RfidResearchGroup, 2018-09-18:
To suggest from Proxmark Wiki,  about Kali

[image]

Peterthegreat, 2018-09-18:
Thanks for the fast support! I really appreciate it.

Everything appears to be OK. I uninstalled modemmanager, did the image upgrade and installed it back.

closed 2018-09-18 by Peterthegreat