RfidResearchGroup/proxmark3 #35: Broken Proxmark3 Rv4.0 after initial flash (Kickstarter Edition)

andreashappe, 2018-10-21:
I was redirected here by Dennis from the Kickstarter messaging board.

**My original Message from 2018/7/31:**

I do have have problem with my proxmark. I followed the update instructions at https://github.com/Proxmark/proxmark3/wiki/Kali-Linux

When flashing the new bootrom the following output was produced:

``` shell
andy@CargoCult:~/Tools/proxmark3/client$ sudo ./flasher /dev/ttyACM0 ../armsrc/obj/fullimage.elf
Loading ELF file '../armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002c398->0x0002c398) [R X] @0x94
1: V 0x00200000 P 0x0012e398 (0x00001938->0x00001938) [RW ] @0x2c42c
Note: Extending previous segment from 0x2c398 to 0x2dcd0 bytes

Waiting for Proxmark to appear on /dev/ttyACM0 .
Found.
#db# unknown command:: 0xc20d540d
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM0 ............
Found.

Flashing...
Writing segments for file: ../armsrc/obj/fullimage.elf
0x00102000..0x0012fccf [0x2dcd0 / 367 blocks]....................Waiting for a response from the proxmark...
You can cancel this operation by pressing the pm3 button
```

After half an hour I canceled the operation. now the proxmark is not detected when plugging in (the A and C lights are red, all four blue lights are lit).

Error-wise I get the following dmesg output:

``` shell
[ 413.180442] usb 3-4: new full-speed USB device number 26 using xhci_hcd
[ 413.584453] usb 3-4: device descriptor read/64, error -71
[ 416.404493] usb 3-4: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
[ 416.404501] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 416.404504] usb 3-4: Product: PM3 Device
[ 416.404506] usb 3-4: Manufacturer: proxmark.org
[ 416.405276] usb 3-4: can't set config #1, error -71
[ 416.405346] usb 3-4: USB disconnect, device number 26
```

After that Dennis told me on 2018/8/16:

> Sorry for delay. Change the cable as we realise there is 1% faulty cables within this KS shipment

As well as (on 2018/9/18):

> https://github.com/RfidResearchGroup/proxmark3
> 
> Try the new one.
> 
> I think your issue is the jumping ports ?
> 
> Hold the white button when flashing so the ports will not jump. As I can see now, the proxmark3 is in brick mode. 

This was my answer when I tried the suggested fix:

still the same (initially all red and blue lights were blinking), now the A/C lights are red.

Output during flashing:

```
~/Tools/proxmark3$ sudo client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000c8c->0x00000c8c) [R X] @0x298

Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002bb98->0x0002bb98) [R X] @0x94
1: V 0x00200000 P 0x0012db98 (0x00001964->0x00001964) [RW ] @0x2bc2c
Note: Extending previous segment from 0x2bb98 to 0x2d4fc bytes

Waiting for Proxmark to appear on /dev/ttyACM0 ................
Found.
Waiting for a response from the proxmark...
You can cancel this operation by pressing the pm3 button

```
This will not finish. Dmesg shows, that the USB device disconnects:

```
[ 1997.250509] usb 3-4: New USB device found, idVendor=9ac4, idProduct=4b8f, bcdDevice= 1.00
[ 1997.250513] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1997.250515] usb 3-4: Product: PM3 Device
[ 1997.250517] usb 3-4: Manufacturer: proxmark.org
[ 1997.250519] usb 3-4: SerialNumber: 888
[ 1997.251321] cdc_acm 3-4:1.0: ttyACM0: USB ACM device
[ 2015.077421] cdc_acm 3-4:1.0: failed to set dtr/rts
[ 2015.319739] usb 3-4: USB disconnect, device number 11
[ 2015.734344] usb 3-4: new full-speed USB device number 12 using xhci_hcd
```
To this Dennis answered (2018/9/18):

> https://github.com/RfidResearchGroup/proxmark3
> 
> Test it on a windows computer first.
> 
> If not, post an issue here on github.
> 
> Sorry for delay.

I did test it under Windows (also on 2018/9/18), found out the following:

> Hi, just tested it with Windows: problem is, that the proxmark is not even detected by windows (so no driver is installed). Should I post my Linux findings (and the windows problem) to github or should I test something else before?
> 
> I'm still very disturbed that my proxmark seems to be in a bricked state (and I was following the gitlab instructions to the point).

As there was no feedback in the month since I'm moving this to github. I'm still very disturbed that my proxmark3 seems to be bricked after I was using your supplied cable and your instructions to the letter. Not very happy about this.

RfidResearchGroup, 2018-10-21:
Regarding  Kali:    
> possible error
> if device gets disconnected while getting enumerated, it is usally the modem-manager which are spooking. Some users has tried killing it with some success. A more permanent solution must be found for Kali users. 

Usually killing modem-manager and don't forget the udev rules .   In this repo you can run

andreashappe, 2018-10-22:
wow, uninstalling modemmanager did the trick!

This happened during flashing:

```
ah@IrregularApocalypse:~/Tools/proxmark3$ sudo client/flasher /dev/ttyACM1 armsrc/obj/fullimage.elf 
Loading ELF file armsrc/obj/fullimage.elf
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00037a48->0x00037a48) [R X] @0x94
1: V 0x00200000 P 0x00139a48 (0x000012bc->0x000012bc) [RW ] @0x37adc
Note: Extending previous segment from 0x37a48 to 0x38d04 bytes

[+] Waiting for Proxmark to appear on /dev/ttyACM1           
[=] UART Setting serial baudrate 460800
.Found 

Flashing... 
Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0013ad03 [0x38d04 / 455 blocks].......................................................................................................................................................................................................................................................................................................................................................................................................................................................................OK 

Resetting hardware... 
All done. 

Have a nice day!
```

And this is the output of the client:

```
[=] UART Setting serial baudrate 460800

Proxmark3 RFID instrument
          

 [ CLIENT ]          
 client: iceman build for RDV40 with flashmem; smartcard;  
          
 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-873-gc15e755e-dirty-unclean 2018-05-26 08:47:16
      os: iceman/master/1deaab5f 2018-10-22 10:19:36

 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 240898 bytes (46%) Free: 283390 bytes (54%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory  
```

..and i was able to read some RFID cards! Thanks! (: should I close the issue?

RfidResearchGroup, 2018-10-22:
Got a mail that it solved your problem. Closing.

closed 2018-10-22 by RfidResearchGroup