RfidResearchGroup/proxmark3 #38: Mess with information about proxmark3 rdv40

zhovner, 2018-10-22:
Okay, I'm want to read more about proxmark3 rdv40 and make a decision is it worth buying or not.

First I go to rfidresearchgroup.com and what I see. It's a instagram  with fashionable color PCB's? 
[image]

Not a single line of text with specification. I see black, blue and red PCB's. Is this a different models of device? Where should I click? Or click on every model and read the difference? 
Okay, I click on random device that look similar to proxmark3. And carousel gallery appears with PCB photos. What?!
[image]

Okay I finally find a menu on the top and click to `Our Projects --> PROXMARK3 RDV4.0` and get to proxgrind.com. I click to `Shop --> Hardware` to read more information about proxmark3 RDV4.0 

[image]

And here is what I see. Where is the proxmark3 RDV4.0?!

[image]

 Okay, I found this page http://proxgrind.com/prototyping/proxmark3-rdv4-0-development/ and read finally read some useful info. 

Then I got to https://lab401.com/collections/hardware/products/proxmark-3-rdv4 to read  more information. I still don't get what the SIM card socket made for and what the difference between external red HF antenna and assembly blue HF antenna even after watch a video. What is the long range and medium rage? 
[image]

So I look at  Assembly Instructions link in resources section. Guess what I get? 

[image]

I got this: proxmark elechouse RDV version assembly instruction.
Installation / Introduction Manual opens elechouse documents too.
[image]

#### Conclusion

Proxmark3 is a highly specialized device for experts that expects a schemes, accurate technical documents and so on. RDV4.0 looks very cool but it lack of clear documentation and accurate structure of information on all affiliated sites. I want to see all necessary information on single page, without the scrolling of tons a slides from kickstarter presentation. I want to see a real comparisons with old revision and real usage examples. Not a PCB photo gallery. 

Here is the examples of good product page, take a look for inspiration:

https://hydrabus.com/hydrabus-1-0-hw-detailed/

zhovner, 2018-10-22:
I will better post this here http://www.proxmark.org/forum/viewtopic.php?pid=32590

closed 2018-10-22 by zhovner

iceman1001, 2018-10-27:
You sound very confused.  Let me try to clear it for you a bit,  reading on the forum thread also helps.

The PM3 rdv4  is a _normal_ Pm3 with some extra features as

- onboad memory
- sim card module
- fpc connector

All neatly packaged in a small formfactor and nice case. 
As mention on those sites the RDV4 is at the moment only sold via two distributors .  Hackerwarehouse and Lab401. 

These extra hardware features enables you to make more interesting functionality with your pm3 rdv4. 
I among others have added basic support for the hardware,  making it more rapid developement for new features.   It offers you a new platform of possibilities to make use of. 

for instance you can now take your rfid enabled credit card and look at the rfid part and hook into the sim card via a extender (sim size -> full size)  the card and see that part of the ISO-7816 comms.  
All in same client.  

The new extra onboard memory will help for all kinds of times you want save / load stuff on the device,  some ppl has already started adding it to standalone modes..   Making the device in standalone mode much more efficent in the field.