RfidResearchGroup/proxmark3 #46: hf mf hardnested.. broken?

vdwel, 2018-11-03:
When I try to do a hardnested attack, I get: 
`Apply bit flip properties                               |             nan |  nand`
 
I am not sure if it has something to to with the os, but I am using OSX. The nan|nand, goes on forever.

The iceman repo does not have this problem.

I didn't have to to dive into this yet. But I thought it might be good to share the issue.

merlokk, 2018-11-03:
try to launch `hf mf hardnested t 1 000000000000`

vdwel, 2018-11-04:
This gives:

      0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 120 million (2^26.8) keys/s      | 140737488355328 |   14d          
       0 |       0 | Using 0 precalculated bitflip state tables              | 140737488355328 |   14d          
       2 |       0 | Simulating key 000000000000, cuid 51601b95 ...          | 140737488355328 |   14d          
       2 |     113 | Apply bit flip properties                               | 140737488355328 |   14d          
       2 |     451 | Apply bit flip properties                               | 140737488355328 |   14d          
       2 |     788 | Apply bit flip properties                               | 140737488355328 |   14d          
       2 |    1125 | Apply bit flip properties                               | 140737488355328 |   14d          
       2 |    1347 | Apply Sum property. Sum(a0) = 136                       |   2279249281024 |    5h          
       3 |    1457 | Apply bit flip properties                               |   2279192657920 |    5h          
       3 |    1570 | Apply bit flip properties                               |   2279249281024 |    5h          
       3 |    1679 | Apply bit flip properties                               |   1000147320832 |    2h          
       3 |    1790 | Apply bit flip properties                               |   1000147320832 |    2h          
       3 |    1900 | Apply bit flip properties                               |   1000147320832 |    2h          
       3 |    2007 | Apply bit flip properties                               |   1000147320832 |    2h          
       3 |    2007 | (1. guess: Sum(a8) = 256)                               |   1000147320832 |    2h          
      15 |    2007 | Apply Sum(a8) and all bytes bitflip properties          |   1000147320832 |    2h          
      15 |    2007 | (Test: Key NOT found)                                   |               0 |    0s    

This is what I do when things go wrong:
`hf mf hardnested 1 A a0a1a2a3a4a5 4 A`

       0 |       0 | Start using 8 threads and AVX2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 120 million (2^26.8) keys/s      | 140737488355328 |   14d          
       0 |       0 | Using 0 precalculated bitflip state tables              | 140737488355328 |   14d          
       3 |     112 | Apply bit flip properties                               | 140737488355328 |   14d          
       4 |     224 | Apply bit flip properties                               | 140737488355328 |   14d          
       5 |     336 | Apply bit flip properties                               | 140737488355328 |   14d          
       6 |     447 | Apply bit flip properties                               | 140737488355328 |   14d          
       7 |     559 | Apply bit flip properties                               | 140737488355328 |   14d          
       8 |     670 | Apply bit flip properties                               | 140737488355328 |   14d          
       8 |     782 | Apply bit flip properties                               | 140737488355328 |   14d          
       9 |     893 | Apply bit flip properties                               | 140737488355328 |   14d          
      10 |    1003 | Apply bit flip properties                               | 140737488355328 |   14d          
      11 |    1115 | Apply bit flip properties                               | 140737488355328 |   14d          
      12 |    1227 | Apply bit flip properties                               | 140737488355328 |   14d          
      13 |    1337 | Apply bit flip properties                               | 140737488355328 |   14d          
      14 |    1445 | Apply bit flip properties                               | 140737488355328 |   14d          
      14 |    1556 | Apply bit flip properties                               | 140737488355328 |   14d          
      15 |    1663 | Apply bit flip properties                               | 140737488355328 |   14d          
      16 |    1772 | Apply bit flip properties                               | 140737488355328 |   14d          
      17 |    1878 | Apply bit flip properties                               | 140737488355328 |   14d          
      18 |    1986 | Apply Sum property. Sum(a0) = 128                       |             nan |  nand          
      19 |    2096 | Apply bit flip properties                               |             nan |  nand          
      20 |    2205 | Apply bit flip properties                               |             nan |  nand          
      21 |    2315 | Apply bit flip properties                               |             nan |  nand          
      21 |    2425 | Apply bit flip properties                               |             nan |  nand          
      22 |    2531 | Apply bit flip properties                               |             nan |  nand          `

merlokk, 2018-11-04:
Try to disable avx. Look at command help. Im from phone

vdwel, 2018-11-04:
I couldn't find it in the help. So I just disabled AVX512 in the Makefile (client), and recompiled. And this  fixes the problem.

vdwel, 2018-11-04:
Ok.. it seems that the nan problem occurs when I start the proxmark3 executable from a link in /usr/local/bin

The fact that it seemed to be linked to the log file, is probably a coincidence.

The segmentation fault problem is another problem.
Somehow the segmentation fault problem disappeares when I change:
`#define NUM_REDUCTION_WORKING_THREADS	(num_CPUs())`
to
`#define NUM_REDUCTION_WORKING_THREADS	1`
In the file: cmdhfmfhard.c

I am running this on a MacBookPro with 2.7Ghz Core i7

merlokk, 2018-11-04:
```
Options:
      w: Acquire nonces and write them to binary file nonces.bin
      s: Slower acquisition (required by some non standard cards)
      r: Read nonces.bin and start attack
      iX: set type of SIMD instructions. Without this flag programs autodetect it.
        i5: AVX512
        i2: AVX2
        ia: AVX
        is: SSE2
        im: MMX
        in: none (use CPU regular instruction set)
```

option `in`

vdwel, 2018-11-04:
I can't find the iX option. What is going wrong here??

merlokk, 2018-11-04:
ill look at work. maybe it needs to move from proxmark repo

RfidResearchGroup, 2018-11-05:
...on mingw it works as expected.     The extra input param iX might not be in this repo yet.

```
pm3 --> hf mf hard 0 a  fc00018778f7 58 b
--target block no: 58, target key type:B, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0

 time    | #nonces | Activity                                                | expected to brute force
         |         |                                                         | #states         | time
------------------------------------------------------------------------------------------------------
       0 |       0 | Start using 4 threads and AVX2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 366 million (2^28.4) keys/s      | 140737488355328 |    4d
       3 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    4d
       8 |     112 | Apply bit flip properties                               |    238295285760 | 11min
       9 |     224 | Apply bit flip properties                               |    124342550528 |  6min
      10 |     335 | Apply bit flip properties                               |     81319583744 |  4min
      11 |     447 | Apply bit flip properties                               |     75882856448 |  3min
      12 |     559 | Apply bit flip properties                               |     74114990080 |  3min
      13 |     670 | Apply bit flip properties                               |     73130893312 |  3min
      14 |     782 | Apply bit flip properties                               |     59038093312 |  3min
      15 |     892 | Apply bit flip properties                               |     59038093312 |  3min
      16 |    1003 | Apply bit flip properties                               |     48376852480 |  2min
      17 |    1113 | Apply bit flip properties                               |     42143924224 |  2min
      18 |    1222 | Apply bit flip properties                               |     26413299712 |   72s
      19 |    1334 | Apply bit flip properties                               |     20383420416 |   56s
      20 |    1444 | Apply bit flip properties                               |     20383420416 |   56s
      22 |    1554 | Apply Sum property. Sum(a0) = 96                        |       685771328 |    2s
      22 |    1662 | Apply bit flip properties                               |       702249856 |    2s
      23 |    1769 | Apply bit flip properties                               |       702249856 |    2s
      24 |    1877 | Apply bit flip properties                               |       372186464 |    1s
      25 |    1877 | (1. guess: Sum(a8) = 128)                               |       372186464 |    1s
      30 |    1877 | Apply Sum(a8) and all bytes bitflip properties          |       180477328 |    0s
      30 |    1877 | Brute force phase completed. Key found: 776974687573    |               0 |    0s
pm3 -->
```

RfidResearchGroup, 2018-11-05:
@merlokk     I added the cpu-commad set,     but its used with a space.   i 5  == i5

vdwel, 2018-11-05:
Tested with the update from today with the AVX options. All options generate the segmentation fault. Running it on Linux on a VM, does not generate problem. I am running this on a MacBook Pro and an iMac Pro (both have the same problem). Maybe a threading issue??
```
pm3 --> hf mf hardnested 0 A a0a1a2a3a4a5 4 A i n
--target block no:  4, target key type:A, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0           
Using no SIMD core.          

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 8 threads and no SIMD core                  |                 |          
       0 |       0 | Brute force benchmark: 167 million (2^27.3) keys/s      | 140737488355328 |   10d          
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   10d          
       5 |     112 | Apply bit flip properties                               |    145549918208 | 15min          
       6 |     223 | Apply bit flip properties                               |     36698669056 |  4min          
       6 |     335 | Apply bit flip properties                               |     36411879424 |  4min          
       7 |     446 | Apply bit flip properties                               |     35838304256 |  4min          
       8 |     557 | Apply bit flip properties                               |     35838304256 |  4min          
       9 |     665 | Apply bit flip properties                               |     35838304256 |  4min          
       9 |     776 | Apply bit flip properties                               |     35838304256 |  4min          
      10 |     886 | Apply bit flip properties                               |     35838304256 |  4min          
      11 |     994 | Apply bit flip properties                               |     35838304256 |  4min          
      12 |    1103 | Apply bit flip properties                               |     35838304256 |  4min          
      13 |    1212 | Apply bit flip properties                               |     35838304256 |  4min          
      15 |    1320 | Apply Sum property. Sum(a0) = 128                       |     10010160128 |   60s          
      15 |    1430 | Apply bit flip properties                               |     10010160128 |   60s          
      16 |    1541 | Apply bit flip properties                               |     10010160128 |   60s          
      17 |    1648 | Apply bit flip properties                               |      6578807296 |   39s          
      18 |    1756 | Apply bit flip properties                               |      6578807296 |   39s          
      19 |    1865 | Apply bit flip properties                               |      3735252224 |   22s          
      20 |    1974 | Apply bit flip properties                               |      3831299584 |   23s          
      20 |    2085 | Apply bit flip properties                               |      3831299584 |   23s          
      21 |    2194 | Apply bit flip properties                               |      1922544512 |   12s          
      22 |    2303 | Apply bit flip properties                               |      2567988480 |   15s          
      23 |    2409 | Apply bit flip properties                               |      3488799488 |   21s          
      23 |    2409 | (1. guess: Sum(a8) = 64)                                |      3488799488 |   21s          
Segmentation fault: 11
```

cjbrigato, 2018-11-05:
@vdwel you mind explaining upon the closing ? thx

vdwel, 2018-11-05:
Sorry.. I accidentally pressed "close and comment"..

RfidResearchGroup, 2018-11-05:
the segmentation fault,     can you get a better description running gdc or similar?

vdwel, 2018-11-05:
```
      21 |    2201 | Apply bit flip properties                               |      4827253248 |    5s          
      22 |    2312 | Apply bit flip properties                               |      3460939520 |    3s          
      23 |    2312 | (1. guess: Sum(a8) = 192)                               |      3460939520 |    3s          
proxmark3 was compiled with optimization - stepping may behave oddly; variables may not be available.
Process 70914 stopped
* thread #7, stop reason = EXC_BAD_ACCESS (code=1, address=0x12a800000)
    frame #0: 0x00000001000945a3 proxmark3`bitarray_to_list [inlined] next_state(bitarray=0x000000012a600000, state=<unavailable>) at cmdhfmfhard.c:172 [opt]
   169 			line <<= 1;
   170 		}
   171 		index++;
-> 172 		while (bitarray[index] == 0x00000000 && state < 1<<24) {
   173 			index++;
   174 			state += 0x20;
   175 		}
Target 2: (proxmark3) stopped.
(lldb) 
```

iceman1001, 2018-11-05:
can you see which method who was calling next_state() when it crashed?

vdwel, 2018-11-05:
I am not a debugging master, but I think it is the method: bitarray_to_list

iceman1001, 2018-11-05:
how much memory do you have in your computer?

vdwel, 2018-11-05:
16Gb

iceman1001, 2018-11-05:
And you get this error every time you run hardnested?

vdwel, 2018-11-05:
Nope, only after the message:
```(1. guess: Sum(a8) = XX)```
Sometimes it finishes without this message.
It looks like it runs into the problem after entering the method:
```static void generate_candidates(uint8_t sum_a0_idx, uint8_t sum_a8_idx)```

vdwel, 2018-11-05:
:-)

vdwel, 2018-11-05:
So do we change the worstcase_size to 1<<19 or the bittarray size to 1<<20 ??

vdwel, 2018-11-05:
Thanks for the help. 👍 I learned a lot.

iceman1001, 2018-11-05:
I pushed the fix to match worstcase_size.   Test it all out to see you don't get this again!

Thank you for reporting and figuring out a fix!

closed 2018-11-05 by RfidResearchGroup