RfidResearchGroup/proxmark3 #51: hf mf cload

iceman1001, 2018-11-11:
**Describe the bug**
Since the name template  change of filenames when saving dump files etc,   the template starts with h ,  as in  "hf-mf-UID.bin"  or similar.     All older commands still not adapted,  they assume if first char is 'h' its a request for helptext.

**To Reproduce**
Steps to reproduce the behavior:
1. create a dump,   hf mf dump
2. convert from  bin to emul,    script run dumptoemul -h
3. try restoring onto magic card,    hf mf cload hf-mf-ZZZZ

the help text is shown..

**Expected behavior**
Expected behavior is to load the file

**Additional context**
All older commands should have a param 'f'  as for indicating a filename entry.
this is a breaking change from old style command which uses without a param but only filename.

Also all commands should be able to load eml/bin without having to choose.  in fileutils.c should have support for auto-detect and load of those two kinds of formats.   In the future I see also support for reading json,

merlokk, 2018-11-11:
json will be here with #50 and can be easily integrated into old parts of code

RfidResearchGroup, 2018-11-11:
I'm adding some function stubs for you,  in the process of unifying the load / save files from proxmark client.

merlokk, 2018-11-11:
ok)

RfidResearchGroup, 2018-11-12:
alright then,  Got a beginning for it.  Load EML / BIN ,  but how to make a swap based on file existence?  or?

merlokk, 2018-11-14:
#54

iceman1001, 2018-11-24:
how are we doing with the json stuff?

merlokk, 2018-11-24:
not now( i was very busy at work. i think from Monday))))

RfidResearchGroup, 2018-11-24:
Don't stress it.