RfidResearchGroup/proxmark3 #56: How to unbrick rdv4?

sp4rrows, 2018-11-19:
Hi,
How is the procedure of unbricking?
Which tools are needed-same as for pm3rdv2?

-PROXMARK 3 RDV4 - FLASHING SUPPORT (
And which other tool is needed?(Segger J-LINK, Proxmark 3 RDV J-Link adaptor, AVRISP mk2,...)

Would be great to get a short howto.

Thanks

bogiton, 2018-11-20:
Oooh, I like this adaptor! Very fancy!
I think it should be included in RDV4 and Chameleon rebooted wikis.
Btw, any idea how these smaller pin headers are called?

RfidResearchGroup, 2018-11-20:
Yes, the adapter is slick.   Feel free to update all the Wikis..  (Rdv4,  official pm3,  chameleon rebooted and the chameleon rebooted gui )    :)

bogiton, 2018-11-20:
Can't create a new page in rdv4 wiki. I updated the respective page in Chameleon.

sp4rrows, 2018-11-20:
I has some issues while testing with my raspi.
Thanks the De-bricking-Segger has helped.

closed 2018-11-20 by sp4rrows

RfidResearchGroup, 2018-11-21:
@bogiton   I can't see the wiki changes...  I added a page on this wiki.