RfidResearchGroup/proxmark3 #66: #db# unknown command:: 0xd32d410d

gtpy, 2018-12-04:
**Describe the bug**
Get #db#messages at first start of client:
[=] UART Setting serial baudrate 460800
#db# unknown command:: 0xd32d410d 
If I exit client next time this message doesn't appear

**To Reproduce**
1. Compiled and flashed fresh  RDV4
2. Connect it
3. Start client
4. Message appears

**Expected behavior**
#db# unknown command will not be shown.

**Screenshots**
*will follow*

**Desktop (please complete the following information):**
 - OS: ubuntu 18.04.1 LTS x64
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
`Proxmark3 RFID instrument`

` [ CLIENT ]          `
` client: iceman build for RDV40 with flashmem; smartcard;  `
          
```
 [ ARM ]
 bootrom: iceman/master/259dbadb 2018-12-04 13:30:56
      os: iceman/master/259dbadb 2018-12-04 13:30:58
```
```
 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23    
```      

```
 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 241920 bytes (46%) Free: 282368 bytes (54%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```
 - hw status
```
#db# Memory          
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing          
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image          
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
#db# Flash memory          
#db#   Baudrate................24MHz          
#db#   Init....................OK          
#db#   Memory size.............2 mbits / 256kb          
#db#   Unique ID...............0xd567a882a70f7f26          
#db# Smart card module (ISO 7816)          
#db#   version.................v2.06          
#db# LF Sampling config          
#db#   [q] divisor.............95 (125 KHz)          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config          
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............15*8 (120)          
#db# USB Speed          
#db#   Sending USB packets to client...          
#db#   Time elapsed............1500ms          
#db#   Bytes transferred.......809984          
#db#   USB Transfer Speed PM3 -> Client = 539989 Bytes/s          
#db# Various          
#db#   MF_DBGLEVEL.............1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mode          
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)   
```    
 - data tune
```
pm3 --> data tune
          
[=] measuring antenna characteristics, please wait...
          
....
          
[+] LF antenna: 72,94 V - 125.00 kHz          
[+] LF antenna: 39,23 V - 134.00 kHz          
[+] LF optimal: 72,94 V - 125,00 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 48,15 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
```

**Additional context**
n/a

iceman1001, 2018-12-04:
try removing this row  https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/Makefile#L26

gtpy, 2018-12-04:
Try to remove it. Compiled.
Flashing hangs:
```
client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf
Loading ELF file bootrom/obj/bootrom.elf
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000d60->0x00000d60) [R X] @0x298

Loading ELF file armsrc/obj/fullimage.elf
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00037bf8->0x00037bf8) [R X] @0x94
1: V 0x00200000 P 0x00139bf8 (0x000012b4->0x000012b4) [RW ] @0x37c8c
Note: Extending previous segment from 0x37bf8 to 0x38eac bytes

[+] Waiting for Proxmark to appear on /dev/ttyACM0           
[=] UART Setting serial baudrate 460800
.Found 
#db# unknown command:: 0xd32d410d          
Entering bootloader... 
(Press and release the button only to abort)
[+] Waiting for Proxmark to appear on /dev/ttyACM0           
[=] UART Setting serial baudrate 460800
. Found 
[!] sending bytes to proxmark failed          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button   
```

gtpy, 2018-12-04:
While added again -> flashing works again:

```
client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf
Loading ELF file bootrom/obj/bootrom.elf
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000d60->0x00000d60) [R X] @0x298

Loading ELF file armsrc/obj/fullimage.elf
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00037e48->0x00037e48) [R X] @0x94
1: V 0x00200000 P 0x00139e48 (0x000012b8->0x000012b8) [RW ] @0x37edc
Note: Extending previous segment from 0x37e48 to 0x39100 bytes

[+] Waiting for Proxmark to appear on /dev/ttyACM0           
[=] UART Setting serial baudrate 460800
.Found 
Entering bootloader... 
(Press and release the button only to abort)
[+] Waiting for Proxmark to appear on /dev/ttyACM0           
............[=] UART Setting serial baudrate 460800
. Found 

Flashing... 
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks].OK 
 0x00100200..0x00100f5f [0xd60 / 7 blocks].......OK 

Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0013b0ff [0x39100 / 457 blocks].........................................................................................................................................................................................................................................................................................................................................................................................................................................................................OK 

Resetting hardware... 
All done. 

Have a nice day!

```

gtpy, 2018-12-04:
Maybe a stupid question, but the FPC Connector is intended for the High Range antenna?

iceman1001, 2018-12-05:
FPC connector could among others be used for an external antenna  LF/HF 
Also for a uart interface,  like a BL dongel - to serial
Also for external battery

iceman1001, 2018-12-05:
Strange,  I have now tested on the following,  I don't get that message.

- Ubuntu 18.04  (x86_64) 
- Ubuntu 16.04 (686) 
- WSL (win10)

gtpy, 2018-12-05:
I’ll try it again.
Maybe there was a mistake in my handling.

gtpy, 2018-12-05:
Strange. This message already comes up at flashing (actual compiled version without deleted L26 in armsrc/Makefile):
1st try (see error below):
```
~/proxmark3$ client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf
Loading ELF file bootrom/obj/bootrom.elf
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000d60->0x00000d60) [R X] @0x298

Loading ELF file armsrc/obj/fullimage.elf
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00037e48->0x00037e48) [R X] @0x94
1: V 0x00200000 P 0x00139e48 (0x000012b8->0x000012b8) [RW ] @0x37edc
Note: Extending previous segment from 0x37e48 to 0x39100 bytes

[+] Waiting for Proxmark to appear on /dev/ttyACM0           
[=] UART Setting serial baudrate 460800
.Found 
#db# unknown command:: 0xd32d410d        
Entering bootloader... 
(Press and release the button only to abort)
[+] Waiting for Proxmark to appear on /dev/ttyACM0           
[=] UART Setting serial baudrate 460800
. Found 
[!] sending bytes to proxmark failed          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
^C
```

2nd try:
```
~/proxmark3$ client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf
Loading ELF file bootrom/obj/bootrom.elf
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000d60->0x00000d60) [R X] @0x298

Loading ELF file armsrc/obj/fullimage.elf
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x00037e48->0x00037e48) [R X] @0x94
1: V 0x00200000 P 0x00139e48 (0x000012b8->0x000012b8) [RW ] @0x37edc
Note: Extending previous segment from 0x37e48 to 0x39100 bytes

[+] Waiting for Proxmark to appear on /dev/ttyACM0           
[=] UART Setting serial baudrate 460800
.Found 
Entering bootloader... 
(Press and release the button only to abort)
[+] Waiting for Proxmark to appear on /dev/ttyACM0           
............[=] UART Setting serial baudrate 460800
. Found 

Flashing... 
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks].OK 
 0x00100200..0x00100f5f [0xd60 / 7 blocks].......OK 

Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0013b0ff [0x39100 / 457 blocks].........................................................................................................................................................................................................................................................................................................................................................................................................................................................................OK 

Resetting hardware... 
All done. 

Have a nice day!
```

iceman1001, 2018-12-05:
Doubt it to be defect since you can flash and run it.   And hw tune shows good values for your antennas.

The only thing is that your device seems to get a usb command directly after the serial port is opened on the first run...   Which really doesn't make sense right now.  If you got it always,  it could be detectable.
There are only the getversion sent to the device in the startup...

gtpy, 2018-12-05:
with official repo same: 
```
#db# unknown command:: 0xd32d410d          
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.1.0-30-ga015ef3-suspect 2018-12-05 12:12:55
os: master/v3.1.0-30-ga015ef3-suspect 2018-12-05 12:12:56
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2018/09/12 at 15:18:46
SmartCard Slot: available
          
uC: AT91SAM7S512 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 196281 bytes (37%). Free: 328007 bytes (63%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> 
```

I also cannot believe that it is a defect, because all other seems to bee good.

> Doubt it to be defect since you can flash and run it. And hw tune shows good values for your antennas.
>
> The only thing is that your device seems to get a usb command directly after the serial port [...]
I always have it at the first start right after the boot up

iceman1001, 2018-12-05:
...somewhere in the code around openproxmark I guess,  but since I have no idea to replicate your problem I can't fix it.

gtpy, 2018-12-06:
Thanks a lot.
I’ll try to understand more details about PM And try to work my way into the code.

gtpy, 2018-12-06:
Thanks a lot.
I’ll try to understand more details about PM And I will working my way into the code.

closed 2018-12-21 by gtpy