RfidResearchGroup/proxmark3 #82: Recovery of ECDSA keys

iceman1001, 2019-01-11:
**Recovery of ECDSA keys,**

Now, question is how many ECDSA keys are there...  I see in Amiibo that they seem to use eliptic curve signatures. 

See:
https://github.com/tintinweb/ecdsa-private-key-recovery

merlokk, 2019-01-11:
Needs to have a card with this.
Visa Master UnionPay by now have only RSA keys(

iceman1001, 2019-04-17:
Maybe @aczid have an solution for this?

aczid, 2019-04-27:
It's like the PS3 signatures right? I'm not sure I have anything to add @iceman1001

iceman1001, 2019-04-27:
yeah.  Something like PS3 hack.
Don't be shy! 

How did it go with hashcat?

iceman1001, 2020-10-06:
We can retried public keys from signatures now. 
Don't know if we will get any further any time soon.  
I am closing this one meanwhile.

closed 2020-10-06 by iceman1001