RfidResearchGroup/proxmark3 #83: hf legic sim / Legic Simulation does not work properly

raphCode, 2019-01-11:
I have the Proxmark 3 RDV4 and tried simulation of a legic prime card.
I used the steps
* hf legic dump (I used a MIM1024 card and had to increase timeouts in the client, otherwise reads would fail)
* hf legic eload 2 <filename>
* hf legic sim 2
If i recall correctly, at this point LED A is activated and as soon as I get close to the reader LED B and C flash fast, they appear in different intensities. The reader does not react at all.

From older forum posts and issues I saw "timing errors" mentioned, but I thought with  https://github.com/RfidResearchGroup/proxmark3/pull/25 this has been fixed.

Can I somehow help with the debugging/development of this feature? I have experience with programming and electronics as well as some tools like oscilloscope.
I do not know where to start or what to check because I'm not familiar with the project yet. Maybe someone has an idea?

iceman1001, 2019-01-11:
Output from 
**hw version**
**hw status**
**hw tune**

raphCode, 2019-01-11:
<details>
  <summary>Click to expand console output</summary>
`pm3 --> hw version

Proxmark3 RFID instrument
          

 [ CLIENT ]          
 client: iceman build for RDV40 with flashmem; smartcard;  
          
 [ ARM ]
 bootrom: iceman/master/a17bf8e-dirty-unclean 2019-01-08 15:12:19
      os: iceman/master/a17bf8e-dirty-unclean 2019-01-08 15:12:23

 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 248123 bytes (47%) Free: 276165 bytes (53%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
pm3 -->  hw status
#db# Memory          
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing          
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image          
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
#db# Flash memory          
#db#   Baudrate................24MHz          
#db#   Init....................OK          
#db#   Memory size.............2 mbits / 256kb          
#db#   Unique ID...............0xd567a882a784b326          
#db# Smart card module (ISO 7816)          
#db#   version.................v2.06          
#db# LF Sampling config          
#db#   [q] divisor.............95 (125 KHz)          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config          
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............15*8 (120)          
#db# USB Speed          
#db#   Sending USB packets to client...          
#db#   Time elapsed............1500ms          
#db#   Bytes transferred.......825856          
#db#   USB Transfer Speed PM3 -> Client = 550570 Bytes/s          
#db# Various          
#db#   MF_DBGLEVEL.............1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mode          
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)          
pm3 -->  hw tune
          
[=] measuring antenna characteristics, please wait...
          
....
          
[+] LF antenna: 73,08 V - 125.00 kHz          
[+] LF antenna: 42,49 V - 134.00 kHz          
[+] LF optimal: 74,64 V - 126,32 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 49,14 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
`
</details>

iceman1001, 2019-01-11:
Good start,  you have latest firmware from Repo,  and your HF antenna works.
Just standard questions.

You can read the legic card well?   Your eload command seems to be missing the filename....  Is that because you edited your post or did you forgot?

raphCode, 2019-01-11:
I could read the card with the info and reader commands and get data in the first two blocks, the rest are 00. I think this is a plausible result because there is not much data that needs to be stored on the card.
Again, I had to increase the timeout in `client/cmdhflegic.c:736`, function `legic_read_mem` to not have it abort before the wohle card is read.

I thought I added a filename to the eload in my first post, maybe the edit was not sent properly.
I am pretty sure that I did it correctly in the client since a confirmation appeared that 1024 byte were read from the file.

Is there a possibility to use the info command on the emulator memory to confirm dump and eload work as when directly reading the card?

iceman1001, 2019-01-11:
hm..  the rdmem ?  **hf legic dump** doesn't use **legic_read_mem**   so that is strange.  or did you mean that you tried to just read out all 1024 bytes with **rdmem** command to look?

You would need to fiddle with the info command in order to look at a dump file / emulator memory

uhei, 2019-01-21:
I second this.

I've also had to increase the timeout in the legic_read_mem() and CmdLegicDump() functions from 3000 to 7000 to have stable read success with MIM1024 cards.

Also the simulation fails. After eloading the card dump and starting sim when getting closer to the reader LED B and C flash, but the reader doesn't react.

I'm running RDV 4 with latest RfidResearchGroup/proxmark3 repo (firmware/client/bootrom).

How can I help to debug this issue?

iceman1001, 2019-01-21:
I hear ya,
I added a long waiting for reading memory.  Remember that read mem isn't suppose to read the whole memory,  use dump cmd instead.
Also adapted the output,

How about you guys show me some output from using legic commands?

uhei, 2019-01-21:
Thanks iceman1001!

Just gave it another try with your commits from today. Dump fails with timeout:
```tyrell2:proxmark3 uh$ client/proxmark3 /dev/cu.usbmodemiceman1 

██████╗ ███╗   ███╗ ████╗     ...iceman fork
██╔══██╗████╗ ████║   ══█║      ...dedicated to RDV40
██████╔╝██╔████╔██║ ████╔╝
██╔═══╝ ██║╚██╔╝██║   ══█║    iceman@icesql.net
██║     ██║ ╚═╝ ██║ ████╔╝  https://github.com/iceman1001/proxmark3
╚═╝     ╚═╝     ╚═╝ ╚═══╝ pre v4.0

Keep iceman fork alive with a donation!           https://paypal.me/iceman1001/
MONERO: 43mNJLpgBVaTvyZmX9ajcohpvVkaRy1kbZPm8tqAb7itZgfuYecgkRF36rXrKFUkwEGeZedPsASRxgv4HPBHvJwyJdyvQuP

[=] UART Setting serial baudrate 115200

Proxmark3 RFID instrument
          

 [ CLIENT ]          
 client: iceman build for RDV40 with flashmem; smartcard;  
          
 [ ARM ]
 bootrom: iceman/master/v1.1.0-616-g13ed4f46 2019-01-19 16:23:29
      os: iceman/master/v1.1.0-616-g13ed4f46 2019-01-19 16:23:31

 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 248101 bytes (47%) Free: 276187 bytes (53%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
```

Info command works and shows all data of all three segments (output omitted):
```pm3 --> hf legic info
Reading tag memory 1024 b...          
.
          

CDF: System Area          
------------------------------------------------------          
MCD: xx, MSN: xx xx xx, MCC: xx OK     
....
```

---
Dump fails:

```
pm3 --> hf legic dump o ../legic_xxx_20190121
[+] TYPE : MIM1024 card (1002 bytes)          
[+] Reading tag memory 1024 b...          
[!] Command execute time-out          
pm3 --> 
```

---

I've tweaked the timeout:

```
tyrell2:proxmark3 uh$ git diff
diff --git a/client/cmdhflegic.c b/client/cmdhflegic.c
index 0f580da9..63efbdab 100644
--- a/client/cmdhflegic.c
+++ b/client/cmdhflegic.c
@@ -899,7 +899,7 @@ int CmdLegicDump(const char *Cmd){
        clearCommandBuffer();
        SendCommand(&c);
        UsbCommand resp;
-       if (!WaitForResponseTimeout(CMD_ACK, &resp, 3000)) {
+       if (!WaitForResponseTimeout(CMD_ACK, &resp, 7000)) {
                PrintAndLogEx(WARNING, "Command execute time-out");
                return 1;
        }
@@ -1263,4 +1263,4 @@ int CmdHFLegic(const char *Cmd) {
 int CmdHelp(const char *Cmd) {
        CmdsHelp(CommandTable);
        return 0;
-}
\ No newline at end of file
+}
```
and gave it another try:

```
pm3 --> hf legic dump o ../legic_xxx_20190121
[+] TYPE : MIM1024 card (1002 bytes)          
[+] Reading tag memory 1024 b...          
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
[+] Wrote 1024 bytes to ../legic_xxx_20190121.bin          
pm3 -->
```

I will try the simulation tomorrow.

How can I turn on more verbose/debug logging?

iceman1001, 2019-01-21:
You turn on the debugging with **hf mf dbg 4**  (or 1,2,3,4)     I know,  not the obvious place.. 

I missed the timeout for the dump command.  I pushed a fix for it.  Try again, please!

iceman1001, 2019-01-21:
I also changed  the timeouts on

- hf legic write
- hf legic restore
- hf legic wipe

Pull and test :)

drandreas, 2019-01-21:
Looking at the code
 - LED A indicates that the simulator is on
 - LED B indicates that a frame is reserved
 - LED C indicates that a frame is sent (this implies that, the received frame is valid)
As you mentioned you see B and C so that's a good start.

For your Question about debugging:
 

1.  You should run the Simulation for a couple of seconds. The simulator stores the communication in a trace log.
2. after a couple of seconds you should stop the simulation by pressing the button on the Proxmark3.
3. Now you should be able to print a com log using `trace list legic`

This will tell you at what point the communication fails.

As a side note, the official repos simulator might be more stable, since that code uses a more precise time source. However, the official repo has no trace function.

iceman1001, 2019-01-22:
Any news @uhei  @raphCode  ?

uhei, 2019-01-22:
Yes @iceman1001 , there are news.

Dumping the card works now:

```
pm3 --> hf legic dump o ../legic_xxx_20190122
[+] TYPE : MIM1024 card (1002 bytes)          
[+] Reading tag memory 1024 b...          
.
          
[+] Wrote 1024 bytes to ../legic_xxx_20190122.bin          
pm3 --> 
```

Simulating the card still fails. LED B and C are flashing fast:

```
pm3 --> hf legic eload 2 ../legic_xxx_20190122
          
[+] Loaded 1024 bytes from file: ../legic_xxx_20190122.bin  to emulator memory
          
pm3 --> hf legic sim 2
pm3 --> #db# Starting Legic emulator, press button to end          
#db# Stopped          

pm3 --> trace list legic
Recorded Activity (TraceLen = 10513 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
LEGIC - Reader Mode: Timings are in ticks (1us == 1.5ticks)
        Tag Mode: Timings are in sub carrier periods (1/212 kHz == 4.7us)          
          
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |        342 | Rdr |16  2a  20  08                                                           |     |           
       1740 |       2082 | Rdr |16  2a  20  08                                                           |     |           
       5493 |       5617 | Rdr |07  61  00  00                                                           |     | IV 0x61          
       5687 |       5813 | Tag |06  3d  00                                                               |     | MIM1024          
       8083 |       8207 | Rdr |07  31  00  00                                                           |     | IV 0x31          
       8277 |       8403 | Tag |06  3d  00                                                               |     | MIM1024          
      16476 |      16818 | Rdr |16  2a  20  08                                                           |     |           
      18216 |      18558 | Rdr |16  2a  20  08                                                           |     |           
      21932 |      22056 | Rdr |07  0b  00  00                                                           |     | IV 0x0B          
      22126 |      22252 | Tag |06  3d  00                                                               |     | MIM1024          
      22305 |      22415 | Rdr |06  39  00  00                                                           |     | ACK 256/1024          
      22473 |      22677 | Rdr |0b  01  00  00                                                           |     | READ Byte(0)          
      22747 |      22999 | Tag |0c  77  02                                                               |     |           
      23223 |      23427 | Rdr |0b  03  00  00                                                           |     | READ Byte(1)          
      23497 |      23749 | Tag |0c  0e  04                                                               |     |           
      23973 |      24177 | Rdr |0b  05  00  00                                                           |     | READ Byte(2)          
      24247 |      24499 | Tag |0c  4c  09                                                               |     |           
      24723 |      24943 | Rdr |0b  07  00  00                                                           |     | READ Byte(3)          
      25013 |      25265 | Tag |0c  5e  09                                                               |     |           
      25489 |      25701 | Rdr |0b  09  00  00                                                           |     | READ Byte(4)          
      25771 |      26023 | Tag |0c  80  06                                                               |     |           
      29748 |      29872 | Rdr |07  45  00  00                                                           |     | IV 0x45          
      29942 |      30068 | Tag |06  3d  00                                                               |     | MIM1024          
      30121 |      30231 | Rdr |06  39  00  00                                                           |     | ACK 256/1024          
      30289 |      30509 | Rdr |0b  01  00  00                                                           |     | READ Byte(0)          
      30579 |      30831 | Tag |0c  77  02                                                               |     |           
      31055 |      31267 | Rdr |0b  03  00  00                                                           |     | READ Byte(1)          
      31337 |      31589 | Tag |0c  0e  04                                                               |     |           
      31813 |      32001 | Rdr |0b  05  00  00                                                           |     | READ Byte(2)          
      32071 |      32323 | Tag |0c  4c  09                                                               |     |           
      32547 |      32767 | Rdr |0b  07  00  00                                                           |     | READ Byte(3)          
      32837 |      33089 | Tag |0c  5e  09                                                               |     |           
[... cut ...]
     110240 |     110428 | Rdr |0b  d8  00  00                                                           |     | WRITE Byte(108)          
     110974 |     111138 | Rdr |0b  01  00  00                                                           |     | READ Byte(0)          
     111516 |     111696 | Rdr |0b  38  00  00                                                           |     | WRITE Byte(28)          
...
```

Any hints how I can get the time source of the official repo to this one? Which file should I look at?

iceman1001, 2019-01-22:
That would be the fpga code...   
Looking at your trace shows that the simulation works for some bytes,  and the reader seems to both want to read / write to your pm3.

Just try the offical repo and see if that works better against the reader,  ie the reader reacts?

uhei, 2019-01-23:
I gave it a try today with the official repo.

More or less same behavior: LEC C and D flashing, but reader doesn't react (door doesn't open).

```
tyrell2:proxmark3 uh$ client/proxmark3 /dev/cu.usbmodem14201 
Prox/RFID mark3 RFID instrument          
bootrom: iceman/master/v1.1.0-616-g13ed4f46 2019-01-19 16:23:29
os: master/v3.1.0-52-g1511ea2-suspect 2019-01-22 20:45:15
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2018/11/28 at 08:33:11
SmartCard Slot: available
          
uC: AT91SAM7S512 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 206126 bytes (39). Free: 318162 bytes (61).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          

proxmark3> hf legic reader
proxmark3> #db# Reading card ...          
#db# Card (MIM 1024) read, use 'hf legic decode' or          
#db# 'data hexsamples 1024' to view results          

hf legic sim 2
proxmark3> #db# Starting Legic emulator, press button to end          
#db# Stopped          

proxmark3> 
```

I've also tried to enable more debugging (`hf mf dbg 4 `) but not more output has been shown.

drandreas, 2019-01-24:
There is no output that can be enabled with `dbg 4`, since adding serial output,
would make the code too slow.

The come log looks surprisingly clean. But it puzzles me that the received frame length is always 0x0b (11). 11 bits is correct for read, but writes should be 23 bits. I assume 1 or several bit errors.

Let us look at the timings read of byte 0, 1, 2, 3, 4 look fine, tag to reader frame gabs are constant (224 ticks). But the next gap is significantly longer (1442). My best guess would be that the does not handle that case. You could try to double `RWD_CMD_TIMEOUT` and see how it goes.

iceman1001, 2019-01-25:
Pull latest, I added @drandreas  suggestion of doubling the timeout.

uhei, 2019-01-25:
> Pull latest, I added @drandreas suggestion of doubling the timeout.

Some minutes ago I've already modified the code.
Now more bytes are read. However door still doesn't open.

```
tyrell2:proxmark3 uh$ git diff
diff --git a/armsrc/legicrfsim.c b/armsrc/legicrfsim.c
index 1816a29c..7417325a 100644
--- a/armsrc/legicrfsim.c
+++ b/armsrc/legicrfsim.c
@@ -46,7 +46,7 @@ static uint32_t last_frame_end; /* ts of last bit of previews rx or tx frame */
 #define RWD_TIME_PAUSE        4 /* 18.9us */
 #define RWD_TIME_1           21 /* RWD_TIME_PAUSE 18.9us off + 80.2us on = 99.1us */
 #define RWD_TIME_0           13 /* RWD_TIME_PAUSE 18.9us off + 42.4us on = 61.3us */
-#define RWD_CMD_TIMEOUT      40 /* 40 * 99.1us (arbitrary value) */
+#define RWD_CMD_TIMEOUT      80 /* 40 * 99.1us (arbitrary value) */
 #define RWD_MIN_FRAME_LEN     6 /* Shortest frame is 6 bits */
 #define RWD_MAX_FRAME_LEN    23 /* Longest frame is 23 bits */
 

----
pm3 --> hf legic eload 2 ../legic_xxx_20190122
          
[+] Loaded 1024 bytes from file: ../legic_xxx_20190122.bin  to emulator memory
          
pm3 --> 
pm3 --> hf legic sim 2
pm3 --> #db# Starting Legic emulator, press button to end          
#db# Stopped          

pm3 --> trace list legic
Recorded Activity (TraceLen = 19385 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
LEGIC - Reader Mode: Timings are in ticks (1us == 1.5ticks)
        Tag Mode: Timings are in sub carrier periods (1/212 kHz == 4.7us)          
          
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |        140 | Rdr |07  4f  00  00                                                           |     | IV 0x4F          
        210 |        336 | Tag |06  3d  00                                                               |     | MIM1024          
       8418 |       8760 | Rdr |16  2a  20  08                                                           |     |           
      10158 |      10500 | Rdr |16  2a  20  08                                                           |     |           
      13878 |      14002 | Rdr |07  19  00  00                                                           |     | IV 0x19          
      14072 |      14198 | Tag |06  3d  00                                                               |     | MIM1024          
      16468 |      16608 | Rdr |07  4f  00  00                                                           |     | IV 0x4F          
      16678 |      16804 | Tag |06  3d  00                                                               |     | MIM1024          
      24886 |      25228 | Rdr |16  2a  20  08                                                           |     |           
      26626 |      26968 | Rdr |16  2a  20  08                                                           |     |           
      30346 |      30494 | Rdr |07  7d  00  00                                                           |     | IV 0x7D          
      30564 |      30690 | Tag |06  3d  00                                                               |     | MIM1024          
      32959 |      33083 | Rdr |07  23  00  00                                                           |     | IV 0x23          
      33153 |      33279 | Tag |06  3d  00                                                               |     | MIM1024          
      41358 |      41700 | Rdr |16  2a  20  08                                                           |     |           
      43098 |      43440 | Rdr |16  2a  20  08                                                           |     |           
      46819 |      46942 | Rdr |07  51  00  00                                                           |     | IV 0x51          
      47012 |      47138 | Tag |06  3d  00                                                               |     | MIM1024          
      49408 |      49548 | Rdr |07  37  00  00                                                           |     | IV 0x37          
      49618 |      49744 | Tag |06  3d  00                                                               |     | MIM1024          
      57822 |      58164 | Rdr |16  2a  20  08                                                           |     |           
      59562 |      59904 | Rdr |16  2a  20  08                                                           |     |           
      63282 |      63406 | Rdr |07  51  00  00                                                           |     | IV 0x51          
      63476 |      63602 | Tag |06  3d  00                                                               |     | MIM1024          
      63655 |      63773 | Rdr |06  39  00  00                                                           |     | ACK 256/1024          
      63831 |      64059 | Rdr |0b  01  00  00                                                           |     | READ Byte(0)          
      64129 |      64381 | Tag |0c  77  02                                                               |     |           
      64605 |      64833 | Rdr |0b  03  00  00                                                           |     | READ Byte(1)          
[... cut ...]
     236750 |     236986 | Rdr |0b  cb  02  00                                                           |     | READ Byte(357)          
     237056 |     237308 | Tag |0c  a4  04                                                               |     |           
     237532 |     237728 | Rdr |0b  cd  02  00                                                           |     | READ Byte(358)          
     237798 |     238050 | Tag |0c  03  02                                                               |     |           
```
So, we're making progress! Thanks so far @iceman1001 and @drandreas

iceman1001, 2019-01-25:
good,  for the sake of saving the time it takes to scroll those paste,   please consider using pastebin.com for it.

I see less and less writes but more and more reads.   Try increasing the timeout evermore  120?

uhei, 2019-01-25:
> I see less and less writes but more and more reads. Try increasing the timeout evermore 120?

Yeah. That's it. Changed timeout to 120 and now it works!
Thanks a lot @iceman1001 and @drandreas!

iceman1001, 2019-01-25:
I am closing this one even if  @raphCode  has not confirmed.

closed 2019-01-25 by iceman1001

raphCode, 2019-01-25:
I was busy the last days and also had no possibility to test, but I will try with the updated code and report back here. 
The recent changes look very promising, thanks to all who contributed to this issue!

raphCode, 2019-01-29:
Yep, it works for me too, now!
Again thanks @iceman1001 and @drandreas :)

Maybe a good idea to merge the changes also into the official repo or at least the "normal"/non-rdv4 iceman fork?

iceman1001, 2019-01-29:
merged to iceman already.
will do offical later,  feel free to do it.