RfidResearchGroup/proxmark3 #91: Power up my RDV4 by a smartphone USB/OTG

Mera-balou, 2019-02-09:
I don't manage to power up my PM3 RDV4 with my phone (in order to use it with walrus app). I've an One Plus 6T smartphone and it seems that some people dit manage to with smartphones.
TBH, i've already open an issue [here](https://github.com/TeamWalrus/Walrus/issues/113#issue-408460674) in walrus repository.
My questions are : 
-  Well, did anyone succeed in powering up RDV4 with a smartphone?
- If so, with wich smartphone/cable models?
- Anyway to increase my OP 6T OTG power?

Regards

mazodude, 2019-02-11:
I have been able to use the RDV4 on a Google Pixel (1) and on a OPPO R17 Pro. Both worked fine but the LF side doesn't get enough power. 
I just used a usb-c to usb A adapter.
You can use a usb Y cable to inject power from a power bank.

iceman1001, 2019-02-13:
the usb Y cable is a good option for now.  Until the battery addon is released...

Cheeks300zx, 2019-02-13:
I use my AT&T Galaxy Note 9 with a amazon basics usb c to micro 1 feet connected to my RDV4.0. Works perfect with Walrus app and AndProx app. The phone powers the proxmark with no problem everything works on it.

iceman1001, 2019-02-13:
So it comes down to which phones that are able to power the rdv4 without extra battery.

I don't think we have a list of that.

closed 2019-02-15 by iceman1001