RfidResearchGroup/proxmark3 #100: Possible double free

sh7d, 2019-02-23:
After last changes with resolving memory leaks I'm unable to preform fully hardnested attack - using modified version of iceman old hardnested script :) 
When script runs key checking sometimes there is a nasty suprise:

> 
> [+] Using AVX SIMD core.          
> 
> 
>           
>  time    | #nonces | Activity                                                | expected to brute force          
>          |         |                                                         | #states         | time           
> ------------------------------------------------------------------------------------------------------          
>        0 |       0 | Start using 4 threads and AVX SIMD core                 |                 |          
>        0 |       0 | Brute force benchmark: 262 million (2^28,0) keys/s      | 140737488355328 |    6d          
>        1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    6d          
> #db# AcquireNonces: Auth1 error          
>        5 |     112 | Apply bit flip properties                               |  10142311514112 |   11h          
>        6 |     224 | Apply bit flip properties                               |   8605346562048 |    9h          
>        7 |     335 | Apply bit flip properties                               |   8444816916480 |    9h          
>        9 |     445 | Apply bit flip properties                               |   8386506129408 |    9h          
>       10 |     555 | Apply bit flip properties                               |   8378623459328 |    9h          
>       11 |     666 | Apply bit flip properties                               |   8378623459328 |    9h          
>       11 |     777 | Apply bit flip properties                               |   8378623459328 |    9h          
>       12 |     887 | Apply bit flip properties                               |   8378623459328 |    9h          
>       13 |     997 | Apply bit flip properties                               |   8378623459328 |    9h          
>       13 |    1108 | Apply bit flip properties                               |   8378623459328 |    9h          
>       14 |    1216 | Apply bit flip properties                               |   8378623459328 |    9h          
>       15 |    1326 | Apply bit flip properties                               |   8378623459328 |    9h          
>       17 |    1437 | Apply Sum property. Sum(a0) = 0                         |    680740257792 | 43min          
>       17 |    1545 | Apply bit flip properties                               |    299168268288 | 19min          
>       18 |    1655 | Apply bit flip properties                               |    299165908992 | 19min          
>       19 |    1764 | Apply bit flip properties                               |    299168268288 | 19min          
>       20 |    1875 | Apply bit flip properties                               |    299168268288 | 19min          
>       21 |    1982 | Apply bit flip properties                               |    299168268288 | 19min          
>       22 |    2090 | Apply bit flip properties                               |    299168268288 | 19min          
>       22 |    2090 | (1. guess: Sum(a8) = 0)                                 |    299168268288 | 19min          
>       24 |    2090 | Apply Sum(a8) and all bytes bitflip properties          |    256709115904 | 16min          
>       26 |    2090 | (2. guess: Sum(a8) = 64)                                |    978981879808 | 62min          
>       31 |    2090 | Apply Sum(a8) and all bytes bitflip properties          |    917501706240 | 58min          
>       38 |    2090 | Brute force phase:  11,99%	                             |    917216165888 | 58min          
>       43 |    2090 | Brute force phase completed. Key found: <censored>   |               0 |    0s          
> ###	check_found_keys
> #db# ChkKeys: Can't select card (ALL)          
> ###	target key B - Sector 1 block: 4 [0x04]
> [+] Using AVX SIMD core.          
> 
> 
>           
>  time    | #nonces | Activity                                                | expected to brute force          
>          |         |                                                         | #states         | time           
> ------------------------------------------------------------------------------------------------------          
>        0 |       0 | Start using 4 threads and AVX SIMD core                 |                 |          
>        0 |       0 | Brute force benchmark: 262 million (2^28,0) keys/s      | 140737488355328 |    6d          
>        1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    6d          
>        5 |     112 | Apply bit flip properties                               |    102006292480 |  6min          
>        6 |     223 | Apply bit flip properties                               |     26656464896 |  2min          
>        7 |     334 | Apply bit flip properties                               |     26204344320 |  2min          
>        8 |     444 | Apply bit flip properties                               |     23270062080 |   89s          
>        9 |     555 | Apply bit flip properties                               |     23270062080 |   89s          
>       10 |     664 | Apply bit flip properties                               |     23155558400 |   88s          
>       11 |     775 | Apply bit flip properties                               |     21135179776 |   81s          
>       12 |     886 | Apply bit flip properties                               |     21135179776 |   81s          
>       13 |     997 | Apply bit flip properties                               |     21135179776 |   81s          
>       13 |    1104 | Apply bit flip properties                               |     21135179776 |   81s          
> #db# AcquireNonces: Auth1 error          
>       14 |    1212 | Apply bit flip properties                               |     21135179776 |   81s          
>       15 |    1322 | Apply bit flip properties                               |     21135179776 |   81s          
>       16 |    1433 | Apply bit flip properties                               |     21135179776 |   81s          
>       18 |    1544 | Apply Sum property. Sum(a0) = 128                       |      2343664640 |    9s          
>       19 |    1653 | Apply bit flip properties                               |      1892741888 |    7s          
>       20 |    1762 | Apply bit flip properties                               |      1892741888 |    7s          
> #db# AcquireNonces: Auth1 error          
>       20 |    1868 | Apply bit flip properties                               |      1892741888 |    7s          
>       21 |    1868 | (Ignoring Sum(a8) properties)                           |      1892741888 |    7s          
>       27 |    1868 | Brute force phase completed. Key found: <censored>    |               0 |    0s          
> double free or corruption (out)
> Segmentation fault (core dumped)

I guess its because my shitty card is having some truble with keys checking at sector 4, key B with faster uart path, and memory is freed even if there is timeout
This message "normally" shows, when check_keys function is unable to determine newly found key:

###	check_found_keys
UART:: write time-out
[!] sending bytes to proxmark failed

iceman1001, 2019-02-23:
Nay,  it could be a bad fix. let me check it.   yup,  I added a extra free.  I have reverted it.  
Pull, compile and try again!

sh7d, 2019-02-24:
Now its works like a charm :)

closed 2019-02-24 by iceman1001