RfidResearchGroup/proxmark3 #119: [idea] link in amiibo tool software

iceman1001, 2019-03-07:
**Is your feature request related to a problem? Please describe.**
socrams amiibo tool is already added to the repo,  however it is not  used anywere.

**Describe the solution you'd like**
a nice integration with pm3 client,  like from identify a amiibo tag, being able to dump decrypted and restore encrypted.

**Describe alternatives you've considered**
or it can be added like reveng which has its own parameter style / cli.

iceman1001, 2019-09-03:
And now it is compiling.   Just need to hook it up :)

JRWR, 2020-05-16:
Any updates on this, Integration would be nice :)

iceman1001, 2020-05-16:
Not working on it,  but feel free to contribute

iceman1001, 2020-10-01:
There was some issues with old vs new dumpformat and amiibo emulation given the pwd/pack parts.
It should be fixed by now.   To note,  `hf 14a sim`  was never meant to be complete in simulating 14a based tags.

The change of mfu dump format is also a factor,  there is a note for it in the repo.   

We need verification this works well now so ppl don't need to search for blogs in order to simulate a amiibo.

Article for reference:
https://blog.neocodenetworks.com/posts/2020-04-08-amiibo-emulation/

iceman1001, 2021-04-02:
I added the linking to the pm3 client and added a skeleton fct in `hf mfu`  https://github.com/RfidResearchGroup/proxmark3/blob/b6ab166377f47dbfd92993f76ff6edbfb0a9d496/client/src/cmdhfmfu.c#L948   to show how it could be done.

I am not going to put more effort into it.   If someone want to do it,  its all there.  

Closing.

closed 2021-04-02 by iceman1001