RfidResearchGroup/proxmark3 #126: hf 15: possible truncations

doegox, 2019-03-10:
**Describe the bug**

The compiler warns us of two potential truncations in `CmdHF15Restore` when creating new commands internally:
```
cmdhf15.c: In function ‘CmdHF15Restore’:
cmdhf15.c:825:25: warning: ‘strncat’ output may be truncated copying 254 bytes from a string of length 999 [-Wstringop-truncation]
                         strncat(newCmdPrefix, param, sizeof(newCmdPrefix) - 1);
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

So in `newCmdPrefix` (255) we may have to squeeze 1000b (FILE_PATH_SIZE)

```
cmdhf15.c:900:9: note: ‘snprintf’ output between 6 and 516 bytes into a destination of size 255
         snprintf(tmpCmd, sizeof(tmpCmd), "%s u %u %s", newCmdPrefix, i, hex);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

So in `tmpCmd` (255) we may have to squeeze 4b + 255b (newCommandPrefix, itself a truncated version of FILE_PATH_SIZE, see above) + 2b + 255b (hex)

I'm not sure yet which buffers can be extended or not or if user must be warned of too long paths / too long hex.

iceman1001, 2019-03-11:
yeah,  that one..   The whole idea to is cut out part of original cmd in order to send it to other cmds internally.  Normal string operations would have been easy but C is C.  

Looking at the problem,  how to call other functions directly with reusing some data should give a better solution for it.

iceman1001, 2019-04-17:
Is this one not fixed now? time to close?

closed 2020-07-13 by iceman1001