RfidResearchGroup/proxmark3 #127: SIM: ATR wrongly captured

doegox, 2019-03-10:
**Describe the bug**

```pm3 --> sc reader
[=] ISO7618-3 ATR : 3B 67 00 00 00 00 00 00 00 90 00           

Repeating sc reader multiple times:

[=] ISO7618-3 ATR : D5 00 00 3B 67 00 00 00 00 00 00 00 90 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
[=] ISO7618-3 ATR : 3B 67 00 00 00 00 00 00 00 90 00           
[=] ISO7618-3 ATR : 3B 67 00 00 00 00 00 00 00 90 00           
[=] ISO7618-3 ATR : 3B 67 00 00 00 00 00 00 00 90 00 00 00 00 00 00 00 00 00 00 00 00           
[=] ISO7618-3 ATR : 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
[=] ISO7618-3 ATR : 2C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00           
[=] ISO7618-3 ATR : 3B 67 00 00 00 00 00 00 00 90 00           
[=] ISO7618-3 ATR : 3B 67 00 00 00 00 00 00 00 90 00 00 00 00 00 00 00 00 00 00 00 00           
[=] ISO7618-3 ATR : 3B 67 00 00 00 00 00 00 00 90 00
[=] ISO7816-3 ATR : D5 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3B 67 00 00 00 00 00 00 00 90           

pm3 --> sc list
[+] Recorded Activity (TraceLen = 41 bytes)          
[=]           
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
ISO7816-4 / Smartcard - Timings N/A yet          
          
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |          0 | Tag |d5  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00   |     |           
            |            |     |00  3b  67  00  00  00  00  00  00  00  90                               |     |
```

**Context:**
 - Linux
 - Banking card on SIM extender
-  Smart card module v3.10

iceman1001, 2019-03-10:
@merlokk  this is in your playground

merlokk, 2019-03-10:
Hi, I try to check my cards and see if I can reproduce it.
Btw @doegox  do you have latest smardcard firmware? Look sc info

iceman1001, 2019-03-10:
He has v3.10

doegox, 2019-03-10:
yep, it was in my bug description ;)
I just tried with another banking card, same behavior

doegox, 2019-03-10:
BTW `sc info` doesn't display the SIM version, only `hw status`.
There is also a strange thing:

Without a card in the reader:
```
hw status
#db# Smart card module (ISO 7816)          
#db#   version.................v3.10          
```

With a card in the reader:
```
hw status
#db# Smart card module (ISO 7816)          
#db#   version.................FAILED          
```

iceman1001, 2019-03-10:
Something is wrong with your update.
try reflash it with SIM010.BIN

doegox, 2019-03-10:
I tried reflashing several times, still same issues (ATR not sampled properly and hw status not returning sc version if there is a card).
```
pm3 --> sc upgrade f tools/simmodule/SIM010.BIN
[!] WARNING - Smartcard socket firmware upgrade.          
[!] A dangerous command, do wrong and you will brick the smart card socket          
[+] Smartcard socket firmware uploading to PM3          
..
[+] Smartcard socket firmware updating,  don't turn off your PM3!          
#db# FW 0000          
#db# FW 0080          
#db# FW 0100          
#db# FW 0180          
#db# FW 0200          
#db# FW 0280          
[+] Smartcard socket firmware upgraded successful          
```

iceman1001, 2019-03-10:
hm.. its the **failed** message when you have a card inside that makes me worried...
_hw status_ should just give you the version all the time.

Have you fiddled loose the simmodule?  Its easy to snap out of the contact

iceman1001, 2019-03-10:
Not that I think it matters but I get the 90 00,  in the end of my ATR...   

```
pm3 --> sc inf
[=] --- Smartcard Information ---------          
[=] -------------------------------------------------------------          
[=] ISO7618-3 ATR : 3B 6B 00 00 00 31 C0 64 1F 27 01 1C 07 90 00                  
[=] http://smartcard-atr.appspot.com/parse?ATR=3B6B00000031C0641F27011C079000                   
[=] ATR          
	- TB1 (Deprecated: VPP requirements) [ 0x00 ]          
	- TC1 (Extra delay between bytes required by card) [ 0x00 ]          
[!] Check summ error. Must be 0 got 0x4C          
          
[=] Historical bytes | len 0x11 | format 00
          
[=] 	Historical bytes          
   	00: 00 31 c0 64 1f 27 01 1c 07 90 00                |.1.d.'.....
          
[=] D/F (TA1)          
[=] Using default values...          
	- Di 1          
	- Fi 372          
	- F  5.0 MHz          
	- Cycles/ETU 372          
	- 10752.7 bits/sec at 4MHz          
	- 13440.9 bits/sec at Fmax (5.0MHz)          
````

doegox, 2019-03-10:
I snapped the module out and back, reflashed, still same.

doegox, 2019-03-10:
@iceman1001  I also have 9000 at end of my real ATR : 3B 67 00 00 00 00 00 00 00 90 00 
which is visible in some of the traces I gave in the initial description.

doegox, 2019-03-10:
Ok that's becoming interesting:

* Amex card:  works fine, ATR correct, hw status returns sc version
* Maestro cards (2 cards): ATR wrongly acquired, hw status returns sc version FAILED

doegox, 2019-03-10:
So try with all your contact cards, maybe one of them will allow you to reproduce the bug :)

doegox, 2019-03-10:
Amex, working fine:
[image]

The clock is running during RST.

Maestro, wrongly sampled ATR:
[image]

There is a weird 300μs drop on the RST line and the clock kind of follows the RST before eventually running as expected.

Does it ring a bell @merlokk ?

doegox, 2019-03-10:
BTW where is the smartcard module source ? I only saw .BIN so far :(

merlokk, 2019-03-10:
here may be a iso7816 cpu reset. and maybe smart card pull signals to pwr while cpu in reset state. but why...  
ill try to find a card with similar behavior...

doegox, 2019-03-10:
I don't understand how the module currently works...
* Its Vcc is High as soon as the pm3 is powered, even if no card is inserted!
* Its CLK is High when pm3 is started then it's running as soon as a `hw status` is issued, even if no card is inserted!
* Its RST is High as soon as the pm3 is powered

This sounds quite dangerous, it's powering and clocking the card while being inserted, which then will send its cold ATR while I'm even not sure IO is Hi-Z.

And `sc info` is performing a warm reset.

There is no insertion switch so I expect to have commands to choose myself when I want to power the card and when I want to stop power and eject it.
And `sc info` should maybe come with option to choose if we want a cold reset or warm reset.
Maybe with
* `sc poweron` (and get cold ATR)
* `sc reset` (and get warm ATR)
* `sc poweroff`

merlokk, 2019-03-10:
there is no power control in the current version. 
setting RST and CLK down is dangerous because of VCC
so it just performs a warm reset and then tries to communicate

doegox, 2019-03-10:
Some more captures:

Amex, ok:
[image]

Maestro, failing capture of ATR:
[image]

If I zoom and decode on the oscillo, I get properly the initial "3B" after the RST.
We see also the difference between both captures: with Amex stopped and restarted *before* the RST while for Maestro this happens exactly at the same time as the RST.

sentiprox, 2019-03-11:
in the last two pictures probably:
Yellow: RST
Blue: I/O
Pink: CLK
Is it possible that the maestro cards short-circuits its CLK to the ground when resetting?

doegox, 2019-03-11:
No that's not the issue.
Ok I found it @sentiprox @merlokk :

Here is a larger plot when there is no card:

[image]

1) the module is kind of reset, the CLK is stopped and the RST is dropping (floating)
2) the CLK is running and a proper RST is applied

Here is the same plot with the AMEK (OK):

[image]

1) the module is kind of reset, the CLK is stopped and the RST is kind of pulled up by the card itself
2) the CLK is running and a proper RST is applied
3) the card replies with its ATR

Here is the same plot with the Maestro (FAIL):

[image]

1) the module is kind of reset, the CLK is stopped and the RST is dropping (floating)
2) the card thinks it's a reset and sends its ATR
3) the proper RST is never applied

Now the interesting one: I added a pull-up resistor on the RST line:

[image]

1) the module is kind of reset, the CLK is stopped and the RST is pulled up by my resistor
2) the CLK is running and a proper RST is applied
3) the card replies with its ATR
And the PM3 gives me the proper ATR \o/

So now, the question:

Can we get the sc module having an internal pull-up on the RST line? Or avoid that RST drop ? Or ignore ATR after that first drop, wait a bit, and apply a proper reset later and then only start looking at IO?
Of course I could add a resistor but if we can fix it by sw it's way better :)

doegox, 2019-03-11:
And with the pull-up, `hw status` works again too :)

merlokk, 2019-03-11:
ok 
it might be an issue here `D5 00 00 3B 67 00 00 00 00 00 00 00 90`
`D5` - trash from data before reset

merlokk, 2019-03-11:
reset here:
https://github.com/RfidResearchGroup/proxmark3/blob/master/common/i2c.c#L693
maybe we can just remove it and check

doegox, 2019-03-11:
I tried
* Commenting out the reset
  * when I get an ATR it's the correct ATR
  * pretty unstable, often fails
* Adding 50ms delay after reset
  * when I get an ATR it's the correct ATR (most of the time)
  * a bit more stable but `hw status` fails, probably still some internal overflow ongoing

I think the only way is by changing the smartcard module core itself, to ignore crap after its reset and monitor IO only after we ask for explicit reset, and then on pm3 side add a small delay to let the first ATR finish before `GetATR`

doegox, 2019-03-11:
I also tried flushing the buffer with sc_rx_bytes in GetATR, this works fine for the Maestro but now it corrupts ATR detection of AMEX...

iceman1001, 2019-03-11:
Great findings!
So what can we do with sim module firmware / pm3 firmware to solve?

iceman1001, 2019-03-11:
So this latest firmware works v3.11   -  I have pushed it to the repo.  
The redesign / revision on hardware is not to be discussed here.

iceman1001, 2019-03-11:
ok to close?

doegox, 2019-03-11:
\o/ SIM011 successfully tested

closed 2019-03-11 by doegox

iceman1001, 2019-03-11:
@merlokk  seems to have issues...

merlokk, 2019-03-11:
v.10
ATR=3B6E00000031C065F8D602010771D68C6137
v.11:
ATR=253B6E00000031C065F8D602010771D68C613700000000000000000000
ATR=383B6E00000031C065F8D602010771D68C613700000000000000000000

merlokk, 2019-03-11:
it looks i know why.....

merlokk, 2019-03-11:
all is OK, sorry
flashed alpha module that have the same name(

doegox, 2019-03-11:
what? you didn't check the SHA512?? ;)

merlokk, 2019-03-11:
)))

iceman1001, 2019-03-11:
that is one of piwis extras in the offical repo...