RfidResearchGroup/proxmark3 #128: [Mifare] hf mf sim

vratiskol, 2019-03-10:
Hi,

I start to work on hf mf sim to :
- integrate some new Proxmark3 repo function 
- Extract all hf mf sim to a file => hfmfsim.c

This part is almost finish.
As @Icemann said to me, maybe I will need to modify some files after @philippe teuwen cleanning (thank for this !, I had a big mess with CRLF/LF files in the repo before)

Now, I'm working on Mifare mini, 1k and 4k simulation, as the current Mifare1ksim function only support 1k card.

Work in progress:
- Client side configuration (add option, extend flags param)
- arm side (ATQA / SAK managment), flags
 
 I will start to do a PR quite soon (I hope) :)

iceman1001, 2019-03-10:
Don't forget the 2K support when you are at it :)

Depends how you did the extraction,   the **hf mf sim** implementation in iceman based repo is different from offical.  So those changes needs to come with.  There is a lot of functions that will be moved from iso14443a.c -> hfmfsim.c   .  I recommend you call it:   **mifaresim.c**

doegox, 2019-03-10:
And Mifare Mini :)
I think we're done with the big cleaning, feel free to modify files @vratiskol (you can use `make style` to automatically insure your changes follow the global coding style)

vratiskol, 2019-03-15:
I push to my github a branch name [hf_mf_sim](https://github.com/vratiskol/proxmark3/tree/hf_mf_sim)
I'm currently stucked with something which looklike a memory leak during AUTH phase.
During my test, my proxmark hangs with led A and B on.
I have to unplug the usb port.
Could someone help me about this ? :)

doegox, 2019-03-15:
In switch cases line 37 & 114 you always return and so never go to your debug print, intended ?

doegox, 2019-03-15:
@vratiskol In switch cases line 37 & 114 you always return and so never go to your debug print, intended ?

vratiskol, 2019-03-15:
@

> @vratiskol In switch cases line 37 & 114 you always return and so never go to your debug print, intended ?

Wow, you are quick ! :)
Nope, no intended, I will corrected this right now

iceman1001, 2019-03-18:
Lets make sure you get this one with your fix.
https://github.com/RfidResearchGroup/proxmark3/commit/ca24b7ae74e8e99facee476527d75dedf63ce7ca

The nack/ack should be 4bits responses.

vratiskol, 2019-03-18:
> Lets make sure you get this one with your fix.
> [ca24b7a](https://github.com/RfidResearchGroup/proxmark3/commit/ca24b7ae74e8e99facee476527d75dedf63ce7ca)
> 
> The nack/ack should be 4bits responses.

Perfect ! this seems to solve my issue :)

iceman1001, 2019-03-20:
@vratiskol  you think its ready for PR?

vratiskol, 2019-03-20:
Not ready yet for a PR sadly...

I have an issue with the speed which the proxmark is replying during AUTH phase.
It's look like the PM is not waiting for the reader (my android phone) reply.
Reader repeats its queries twice and finally, the tag replied is not concidered as part of the auth.

I suppose I have to work with DELAY_ARM2AIR_AS_TAG or other DELAY variable.
Any idea ?

As the log below show, Time to reply with:
Phone as reader / PM as Tag: 9376 cycles
PM as reader / MF Card as Tag: 4736 cycles

PM as TAG and Phone as Reader:

   ```
64047318 |   64048310 | Rdr |52                                                                       |     | WUPA
   64049482 |   64051850 | Tag |04  00!                                                                  |     |
   64059064 |   64069592 | Rdr |93  70  92  fe  e7  1d  96  9a  79                                       |  ok | SELECT_UID
   64070636 |   64074156 | Tag |08  b6  dd!                                                              |     |
   64153754 |   64158458 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
   64153764 |   64158468 | Rdr |60  00  f5  7b                                                           |     | AUTH-A(0)
   64163160 |   64167896 | Tag |01  02  03  04                                                           |     | AUTH: nt  => 4736 cycles
   64174746 |   64184122 | Rdr |28  8d  64  b8  19  1f  67  42                                           |     | AUTH: nr ar (enc) => 9376 cycles
   64174722 |   64184098 | Rdr |28  8d  64  b8  19  1f  67  42                                           |     |   => 9376 cycles
   64192822 |   64197558 | Tag |d9  97  c3  dc                                                           |     |
   64262762 |   64267530 | Rdr |50  00  57  cd                                                           |  ok | HALT
   64262748 |   64267516 | Rdr |50  00  57  cd                                                           |  ok | HALT

```

Mf Card Read with PM:

```
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |        992 | Rdr |52                                                                       |     | WUPA
       2244 |       4612 | Tag |04  00                                                                   |     |
       7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
      10676 |      16500 | Tag |92  fe  e7  1d  96                                                       |     |
      19456 |      29984 | Rdr |93  70  92  fe  e7  1d  96  9a  79                                       |  ok | SELECT_UID
      31156 |      34676 | Tag |88  be  59                                                               |     |
      37120 |      41824 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
      43828 |      48564 | Tag |01  20  01  45                                                           |     | AUTH: nt
      57856 |      67168 | Rdr |fd  5e  08  6a  53  a6  52  8b                                           |     | AUTH: nr ar (enc)
      68404 |      73140 | Tag |5c  63! 4c  de!                                                          |     | AUTH: at (enc)
      79104 |      83808 | Rdr |0b  94  05  69                                                           |     |
            |            |  *  |                                              key a0a1a2a3a4a5 prng WEAK  |     |
            |            |  *  |30  00  02  A8                                                           |  ok | READBLOCK(0)
      85044 |     105908 | Tag |41  39  b8! ce  29! 58! 4c  c4! 88  25  e6  40! 86! f5  13! 95! 23  7f   |     |
            |            |  *  |92  FE  E7  1D  96  88  04  00  C0  8F  74  19  49  70  14  11  A8  B1   |  ok |
     118656 |     123360 | Rdr |db  13  af  4f                                                           |     |
            |            |  *  |50  00  57  CD                                                           |  ok | HALT
```

iceman1001, 2019-03-20:
so why is the authentication part slower on **hf mf sim** ?

iceman1001, 2019-03-20:
hm.. I see you swapped the order of the switch-case.   Hence in work  you sent AUTH1,  then you break out and of course next case is AUTH1...    You need to have continue if you want to read the reply direct..

iceman1001, 2019-03-24:
@vratiskol    Did you read this thread?   Speaking of authentication fails for **hf mf sim**
http://www.proxmark.org/forum/viewtopic.php?pid=34601#p34601

vratiskol, 2019-03-27:
@iceman1001 I finally got a "stable" version (commit in [my repo](https://github.com/vratiskol/proxmark3/tree/hf_mf_sim))
its include the modification propose by Eloff which seems to do the job correctly with my phone.
I had simulated a Mifare 1K  card and read the content with my android phone (Mifare Classic Tools app)

I'm working on RATS command right now, 
my coffe machine is complaining a lot and simulation is not working properly ;)

```
   70664504 |   70675032 | Rdr |93  70  92  fe  e7  1d  96  9a  79    |  ok | SELECT_UID
   70676076 |   70679596 | Tag |08  b6  dd!                           |     |
   70687654 |   70692422 | Rdr |e0  80  31  73                        |  ok | RATS
   70687638 |   70692406 | Rdr |e0  80  31  73                        |  ok | RATS
   70694090 |   70694730 | Tag |04                                    |     |
```

iceman1001, 2019-04-03:
how is it going @vratiskol  ?

iceman1001, 2019-04-08:
Ok,  good,  so current status is that we have @vratiskol remake and @McEloff 's first part of fixes.
Looking forward to the second part from @McEloff

iceman1001, 2019-04-08:
@McEloff   I suppose the problems you are having is just what we wanted to prevent. 
It sounds like you have problems in your local repo,  the upstream repo is fine.

iceman1001, 2019-04-10:
the offical repo still has some more fixes being push in a PR,   maybe those will be useful

vratiskol, 2019-04-10:
@McEloff I test the current version over a real reader (a candy distributor with Mifare 1K - 4 UID) and It's work really well. 
I will test it with a Mifare 4K - 4 UID quickly. 

About HALT logs, I confirm, it's not logged.
But regarding my previous post (with logs), it had been working previously
So it doesn't seems too hard to correct :) I will look at it.

iceman1001, 2019-04-11:
Let me start to say first that it was great work in adapting 14a sim/mf sim to the enhancements that was in offical repo and some of your own fixes.   Thanks.

Next thing on my mind is that I notice a lack of vision.  The sim commands could actually benefit from using the onboard flash memory both as a backup memory place and for the nonce collection.
Just bringing the same functions that are in offical repo is good but I want you to take advantage of the extras in RDV4 hardware.    When you do, you find you get a much better proxmark to use.

iceman1001, 2019-04-17:
Since the changes to **hf mf sim** seems to be working.  I will close this issue.
For the ones what want to bring some extra to the table by taking advantage of RDV4 onboard flashmem I be looking forward to a PR :)

Thanks everyone!

closed 2019-04-17 by iceman1001