RfidResearchGroup/proxmark3 #129: [idea] implement json/eml/bin everywhere

iceman1001, 2019-03-10:
**Is your feature request related to a problem? Please describe.**
As of now only parts of the commands has implemented BIN/EML/JSON formats.

**Describe the solution you'd like**
All commands that involves files,  should support BIN/EML/JSON formats. 

**Describe alternatives you've considered**
as an alternative,  lua scripts can be made.   No more python scripts.

**Additional context**
All formats we support helps when other tools needs to import/export from a proxmark generated file.
Today we have Chameleon Mini GUI,  MCT tool interoperability,   and hopefully also Project Walrus in the near future.

iceman1001, 2020-06-02:
Need a list on which more places the client is creating files in order to cover all cases.
Some `rg`magic  @doegox  ?

doegox, 2020-06-02:
read/write:
`rg fopen`
write/append:
`rg 'fopen.*"[aw]'`

iceman1001, 2020-06-02:
```rg 'fopen'|egrep -v "(mbedtls|zlib|liblua|fpga|tinycbo|whereami|armsrc|hitag2crack|deprecated)"```

iceman1001, 2020-06-04:
I don't know what but there are files created and read a little bit everywhere.
In the dependencies its not much to do about it.   Currently these left to convert to use fileutils.c,

```
client/src/cmdlfem4x.c:    if ((f = fopen(filename, "r")) == NULL) {
client/src/ui.c:            logfile = fopen(my_logfile_path, "a");
client/src/proxmark3.c:    FILE *f = fopen(path, "r");
client/src/flash.c:    fd = fopen(name, "rb");
client/src/emv/emv_pk.c:    FILE *f = fopen(fname, "r");
client/src/cmdscript.c:        FILE *f = fopen(script_path, "r");
client/src/cmdlfhitag.c.save:        f = fopen(filename, "wb");
client/src/cmdlfhitag.c:        f = fopen(filename, "wb");
client/src/cmdhfmfhard.c:                if ((fnonces = fopen(filename, "wb")) == NULL) {
client/src/cmdhfmfhard.c:        if ((fstats = fopen("hardnested_stats.txt", "a")) == NULL) {
client/src/cmdhfmf.c:    if ((f = fopen(keyFilename, "rb")) == NULL) {
client/src/cmdhfmf.c:    if ((fkeys = fopen(keyFilename, "rb")) == NULL) {
client/src/cmdhfmf.c:    if ((fdump = fopen(dataFilename, "rb")) == NULL) {
client/src/cmdhfmf.c:            f = fopen(dict_path, "r");
client/src/cmdhfmf.c:            f = fopen(dict_path, "r");
client/src/cmdhfmf.c:    if ((fnonces = fopen(filename, "wb")) == NULL) {
client/src/cmdhficlass.c:    f = fopen(filename, "wb");
client/src/cmddata.c:    FILE *f = fopen(path, "r");
client/deps/hardnested/hardnested_tables.c:    FILE *outfile = fopen(filename, "wb");
client/deps/hardnested/hardnested_bruteforce.c:    FILE *benchfile = fopen(RESOURCES_SUBDIR TEST_BENCH_FILENAME, "wb");
client/deps/hardnested/hardnested_bruteforce.c:    FILE *benchfile = fopen(path, "rb");
client/deps/amiitool/amiitool.c:        f = fopen(infile, "rb");
client/deps/amiitool/amiitool.c:            f = fopen(savefile, "rb");
client/deps/amiitool/amiitool.c:        f = fopen(outfile, "wb");
client/deps/amiitool/amiibo.c:    FILE *f = fopen(path, "rb");
```

[] - convert to use fileutils.c  for load 
[] - convert to use fileutils.c  for save
[] - convert to use json callbacks (smaller fileutils.c)
[] - convert cli commands to take any bin/eml/json files via autodetect.

iceman1001, 2022-02-05:
I realize two different things being discussed in here.
1. bin/eml/json  support in all commands that handles dumps
2. read/write to be centralized into fileutils.c, instead of having custom file access across the pm3 client.

For 1,    haven't checked which commands that still doesn't accept it.  
For 2.    some might fit into fileutils.c direct,  then we have generic binary files written / saved which would demand re-writing those fcts.

iceman1001, 2022-02-12:
MFU commands now fully support bin/eml/json format.

iceman1001, 2022-02-14:
Legic commands got some love,  eload/restore left to take json/eml

iceman1001, 2025-02-20:
Most file actions has been re-routed to fileutils.c  but still there exists a bunch here and there. 

Will not work more on it.  One of these days BIN for dump- and key-files should be deprecated.

Closing meanwhile