RfidResearchGroup/proxmark3 #130: [question] compatible with origial proxmark3

ikarus23, 2019-03-10:
Just a simple question: Is this code compatible with the original proxmark? In other words, can I flash this to my old proxmark?

merlokk, 2019-03-10:
compatible with smardcard module switched off

ikarus23, 2019-03-10:
What do you mean by "switched off"? Is there a target in the Makefile?

merlokk, 2019-03-10:
https://github.com/RfidResearchGroup/proxmark3/blob/master/client/Makefile#L94

ikarus23, 2019-03-10:
Thanks for clearing that up!

closed 2019-03-10 by ikarus23

iceman1001, 2019-03-10:
there is a thread post about it,   you would need to comment out two lines in two makefiles.

http://www.proxmark.org/forum/viewtopic.php?pid=33233#p33233

ikarus23, 2019-03-10:
Thanks, I was wandering about the armsrc.
Since I feel this will become more and more the main PM3 repo, it would be really great if there were a make target for that in the main Makefile. Something which allows me to `make old-proxmark`.

iceman1001, 2019-03-10:
My knowledge in Makefile dark arts is limited but I overheard from @doegox  that he wanted to do the same thing.  A one place switch,  like inside common/makefile.common.   It shouldn't be too hard, its down to those extra flags...   Didn't marshmellow do something like that in offical repo?

doegox, 2019-03-10:
I pushed https://github.com/RfidResearchGroup/proxmark3/commit/119e3f0ed9b6902b99fbd48609e94106c1ec4a0d which allows to define different PLATFORM. By default it's PM3RDV4.
For the moment it doesn't make much difference which other one you pick as all the others are without smartcard and without flash. So feel free to try e.g.:
```
make clean
make PLATFORM=PM3EASY
```
You can make it permanent by creating a file `Makefile.platform` with as content `PLATFORM=PM3EASY` at the root of the repo.

More info:
```
make PLATFORM=help
```
This is just a very first attempt, feel free to try and comment.

iceman1001, 2019-03-11:
**Great one!**

The 256/512 limit,  I haven't tested making a large fullimage to see if we can flash for >256kb fullimages... There is a limit check in fpgaloader I think.    There is a automatic check for 256/512 so that part should be ok.

In the near future, the  **"with fpc"** flag will be important aswell.   since in my tests I can only get 115kb serial speed.

I created a wiki page,  
https://github.com/RfidResearchGroup/proxmark3/wiki/How-to-compile-this-repo-for-non-rdv4