RfidResearchGroup/proxmark3 #133: [idea] Mifare Classic with fixed nonce

iceman1001, 2019-03-11:
**Is your feature request related to a problem? Please describe.**
There has been in the wild some clones, like Fudan and some uid changeable card which doesn't have the NACK bug and they use somewhat odd a fixed nonce.   Rendering all current Mifare classic attacks useless.  

**Describe the solution you'd like**
A new command **hf mf fixednonce** that can recover the keys from such a card

**Describe alternatives you've considered**
Some progress has been done with Fudan cards but they involve sniffing traffic and reuse.

iceman1001, 2019-10-05:
And I have a solution for it...   It was staring me in the face all the time.

iceman1001, 2019-10-08:
Now I tried some more approaches.  Even thought I had a solution which was totally wrong of me.
Since I only noticed this behaviour at my magic tags,  Gen1a,  we can just read the data off it.

Maybe add that to ```hf mf autopwn``` as well?

Sniffing the trafic between reader and tag would still work.

iceman1001, 2019-10-16:
@xtigmh  I saw in aczid's issue that you have been experimenting with a solution for fixed nonces.

Did you see two nonces?  I seen one so far but I have not tested the nested authentication.

xtigmh, 2019-10-17:
> @xtigmh I saw in aczid's issue that you have been experimenting with a solution for fixed nonces.
> 
> Did you see two nonces? I seen one so far but I have not tested the nested authentication.

I see two nonce in hardnested, only one nonce in nested authentication. I guess there are 3 nonces for this card:
nonce1:01200145
nonce2:8190c7dc(encrypted) ->7eef3586(decrypted)
nonce3: 4048e36e(encrypted)
nonce.bin gathered by proxmark3 for your reference:
……8190C7DC228190C7DC4048E36E248190C7DC……
If your card's nonce is same to me, online brutefore is also availabe.

iceman1001, 2019-10-17:
It uses the same nonce as you.  I haven't seen the second or third nonce that you have.

How do you bruteforce it?

```
[usb] pm3 --> hf 14a info
UID : 46 1D CE 03
ATQA : 00 04
SAK : 08 [2]
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1 | 1k Ev1
[=] proprietary non iso14443-4 card found, RATS not supported
[+] Magic capabilities : Gen 1a
[+] Prng detection: WEAK
[usb] pm3 --> hf mf list
[+] Recorded Activity (TraceLen = 103 bytes)
[=]                                                                                                                                                                                              
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer'
ISO14443A - All times are in carrier periods (1/13.56MHz)
Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
     0 |        992 | Rdr |52                                                                       |     | WUPA
2228 |       4596 | Tag |04  00                                                                   |     | 
7040 |       9504 | Rdr |93  20                                                                   |     | ANTICOLL
10676 |      16500 | Tag |46  1d  ce  03  96                                                       |     | 
19328 |      29792 | Rdr |93  70  46  1d  ce  03  96  5e  aa                                       |  ok | SELECT_UID
31028 |      34548 | Tag |08  b6  dd                                                               |     |
44672 |      49376 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
51380 |      56116 | Tag |01  20  01  45                                                           |     | AUTH: nt
```

xtigmh, 2019-10-17:
Sorry, you can only see second nonce in nested auth.

xtigmh, 2019-10-17:
If you send 60/61 xx xx xx to auth, tag always return 01200145, so you must know one key to do nested auth.

xtigmh, 2019-10-18:
aczid/crypto1_bs#41

iceman1001, 2019-10-18:
Yes, I read your issue over at Aczid's repo. 
I still is looking for a good way to implement a solution in the Proxmark3 client.  Some suggestion?

xtigmh, 2019-10-21:
I changed default key, nonoce 3 is never appear. It looks like online bruteforce is the only method to carck this card.

iceman1001, 2019-10-21:
I don't know of your card but mine is a Gen1A magic,  so I can read it using chinese backdoor commands. 
When you online bruteforce which data do you use?  a failed authentiation request? or?

xtigmh, 2019-10-22:
I modified nested auth, it works with one nonce, second step is tesing each key.

iceman1001, 2019-10-22:
But how do you do a nested authentication without knowing the first key to make the intial authentication?

iceman1001, 2019-10-22:
bascially j-run's nested auth solver then.
https://github.com/iceman1001/mf_nonce_brute

But it still needs a valid authentication trace...

iceman1001, 2019-10-22:
I removed the check for a valid key and fast check in order to trigger the nested part.
I get a similar nt2enc but not the same as yours.

```
[usb] pm3 --> hf mf nested 1 0 a ffffffffffff                                                                                                                        [+] Time to check 23 known keys: 0 seconds
[+] enter nested attack                                                                                                                                              #db# Nested: calibrating... ntdist=160
....
#db# Nested: calibrating... ntdist=160
#db# rtr=17 isOK=0 min=160 max=160 avg=160, delta_time=2680
#db# Nonce#1: Testing nt1=01200145 nt2enc=81e07ad8 nt2par=60
#db# Nonce#1: valid, ntdist=160
#db# Nonce#2: Testing nt1=01200145 nt2enc=81e07ad8 nt2par=60
#db# Nonce#2: dismissed (= nonce#1), ntdist=160
#db# Nonce#2: Testing nt1=01200145 nt2enc=81e07ad8 nt2par=60
#db# Nonce#2: dismissed (= nonce#1), ntdist=160
```

xtigmh, 2019-10-23:
> But how do you do a nested authentication without knowing the first key to make the initial authentication?

We should know at least one key, this is not change.

xtigmh, 2019-10-23:
nt2enc will change depends on target block key.

xtigmh, 2019-10-23:
Good job, mf_nonce_brute works well if we have trace.

```
NFC Sniffer>snoop14a
press space key to stop
[COMMAND FINISHED]
Miller.state=0, Miller.len=0
traceLen=294, Miller.output[0]=e7
NFC Sniffer>list14a
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in periods (1/3.39MHz)
      Start |        End | Src | Data (! denotes parity error, ' denotes short bytes)            | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        276 | Rdr | 26'                                                             |     | REQA
        556 |       1156 | Tag | 04  00                                                          |     |
       5280 |       5908 | Rdr | 93  20                                                          |     | ANTICOLL
       6188 |       7660 | Tag | b9  4f  da  14  38                                              |     |
      11964 |      14608 | Rdr | 93  70  b9  4f  da  14  38  7d  c1                              |  ok | SELECT_UID
      14888 |      15776 | Tag | 08  b6  dd                                                      |     |
      19528 |      20732 | Rdr | 50  00  57  cd                                                  |  ok | HALT
      31480 |      31756 | Rdr | 26'                                                             |     | REQA
    8478428 |    8478680 | Rdr | 52'                                                             |     | WUPA
    8478984 |    8479584 | Tag | 04  00                                                          |     |
    8483552 |    8486196 | Rdr | 93  70  b9  4f  da  14  38  7d  c1                              |  ok | SELECT_UID
   38415172 |   38416376 | Rdr | 60  03  6e  49                                                  |  ok | AUTH-A(3)
   38416848 |   38418032 | Tag | 01  20  01  45                                                  |     |
   38418248 |   38420604 | Rdr | 48  5f  cd  d8! 08! bf! 0b! e8!                                 | !crc| ?
   38420884 |   38422060 | Tag | e1  bd  bb  69                                                  |     |
   63203900 |   63205104 | Rdr | 1c  f4  56  9c!                                                 | !crc| ?
   63205576 |   63206760 | Tag | 8c! fd  9f  06!                                                 |     |
   63206972 |   63209304 | Rdr | 29  a0! 1d  46! 3d  e6! 7f! 23!                                 | !crc| ?
   63209608 |   63210792 | Tag | 01! 41  99! 33!                                                 |     |
   69708060 |   69709240 | Rdr | e7  cf! 83  db                                                  | !crc| ?
   69709544 |   69714760 | Tag | f1! 0e  4b  30! 93  fe  da! e3  94  a3! 8a! 77  db  0b  7a  2a! |     |
            |            |     | 02  f0!                                                         | !crc|
```
```
pm3 ~/tools/mf_nonce_brute$ ./mf_nonce_brute.exe b94fda14 8cfd9f06 1001 29a01d46 3de67f23 0111 01419933 1011
Mifare classic nested auth key recovery. Phase 1.
-------------------------------------------------
uid:            b94fda14
nt encrypted:   8cfd9f06
nt parity err:  1001
nr encrypted:   29a01d46
ar encrypted:   3de67f23
ar parity err:  0111
at encrypted:   01419933
at parity err:  1011

Bruteforce using 4 threads to find encrypted tagnonce last bytes

**** Possible key candidate ****

Key candidate: [111111111111]

Execution time: 8777 ticks
```

xtigmh, 2019-10-23:
```
pm3 ~/client$ ./proxmark3.exe com9
Prox/RFID mark3 RFID instrument
bootrom: /-suspect 2019-09-12 01:11:39
os: /-suspect 2019-10-17 03:33:01
fpga_lf.bit built for 2s30vq100 on 2015/03/06 at 07:38:04
fpga_hf.bit built for 2s30vq100 on 2019/03/20 at 08:08:07
SmartCard Slot: not available

uC: AT91SAM7S256 Rev A
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 208379 bytes (79%). Free: 53765 bytes (21%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 256K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory
proxmark3> hf mf dbg 3
#db# Debug level: 3
proxmark3> hf mf nested o 0 a ffffffffffff 4 a
--nested. sectors: 1, block no:  0, key type:A, eml:n, dmp=n checktimeout=471 us
--target block no:  4, target key type:A
#db# ISO14443A Timeout set to 1060 (10ms)
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nested: calibrating... ntdist=160
#db# rtr=17 isOK=0 min=160 max=160 avg=160, delta_time=2672
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nonce#1: Testing nt1=01200145 nt2enc=8cfd9f06 nt2par=a0
#db# Nonce#1: valid, ntdist=160
#db# ISO14443A Timeout set to 10 (0ms)
#db# ISO14443A Timeout set to 1060 (10ms)
#db# Nonce#2: Testing nt1=01200145 nt2enc=8cfd9f06 nt2par=a0
#db# Nonce#2: dismissed (= nonce#1), ntdist=160
#db# NESTED FINISHED
uid:b94fda14 trgbl=4 trgkey=0
nt=7eef3586, ks1=f212aa80
statelist[0].len=89019
after intersection:
statelist[0].len=89019
……trying auth with 89019 keys
```

xtigmh, 2019-10-24:
mf_nonce_brute get wrong key for true random card such as mifare plus.
trace:
```
NFC Sniffer>snoop14a
press space key to stop
[COMMAND FINISHED]
Miller.state=0, Miller.len=0
traceLen=306, Miller.output[0]=1e
NFC Sniffer> list14a
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in periods (1/3.39MHz)
      Start |        End | Src | Data (! denotes parity error, ' denotes short bytes)            | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        276 | Rdr | 26'                                                             |     | REQA
        556 |       1156 | Tag | 04  00                                                          |     |
       5200 |       5828 | Rdr | 93  20                                                          |     | ANTICOLL
       6104 |       7576 | Tag | e0  c8  77  e3  bc                                              |     |
      11852 |      14472 | Rdr | 93  70  e0  c8  77  e3  bc  b0  9c                              |  ok | SELECT_UID
      14772 |      15660 | Tag | 08  b6  dd                                                      |     |
      19480 |      20684 | Rdr | 50  00  57  cd                                                  |  ok | HALT
      31320 |      31596 | Rdr | 26'                                                             |     | REQA
    7020716 |    7020968 | Rdr | 52'                                                             |     | WUPA
    7021272 |    7021872 | Tag | 04  00                                                          |     |
    7025940 |    7028560 | Rdr | 93  70  e0  c8  77  e3  bc  b0  9c                              |  ok | SELECT_UID
    7028860 |    7029748 | Tag | 08  b6  dd                                                      |     |
   11460132 |   11461336 | Rdr | 60  03  6e  49                                                  |  ok | AUTH-A(3)
   11462380 |   11463556 | Tag | fa  a2  bc  00                                                  |     |
   11463780 |   11466136 | Rdr | e0  b0  c7  3b  b0! 87! a0! 14!                                 | !crc| RATS
   11466412 |   11467588 | Tag | 9f  11! 4f! 69                                                  |     |
   21983160 |   21984364 | Rdr | e6  16! c7! 36!                                                 | !crc| ?
   21985412 |   21986588 | Tag | ab! 89  c7  53                                                  |     |
   21986812 |   21989144 | Rdr | 13  bd! 42! ac! 22! 7f! 50! 72                                  | !crc| ?
   21989444 |   21990620 | Tag | 52  59! d5! b1                                                  |     |
   28331392 |   28332596 | Rdr | 1e  55! 97  81!                                                 | !crc| ?
   28332872 |   28338080 | Tag | 64! ea  bb! c7! d8! a3  b7  27  0f! 1f  7a  4c! 89  9e  74! a8  |     |
            |            |     | dd  7f!                                                         | !crc|
NFC Sniffer>
```
mf_nonce_brute: key is wrong, next cmd is right.
```
pm3 ~/tools/mf_nonce_brute$ ./mf_nonce_brute.exe e0c877e3 ab89c753 1000 13bd42ac 227f5072 1110 5259d5b1 0110 1e559781
Mifare classic nested auth key recovery. Phase 1.
-------------------------------------------------
uid:            e0c877e3
nt encrypted:   ab89c753
nt parity err:  1000
nr encrypted:   13bd42ac
ar encrypted:   227f5072
ar parity err:  1110
at encrypted:   5259d5b1
at parity err:  0110
next cmd enc:   1e559781

Bruteforce using 4 threads to find encrypted tagnonce last bytes

**** Possible key candidate ****
CMD enc(1e559781)
    dec(300bd116)       <-- Valid cmd

Key candidate: [173bffffffff]

Execution time: 8837 ticks
pm3 ~/tools/mf_nonce_brute$
```

doegox, 2019-10-24:
mf_nonce_brute is not in this repo. @iceman1001 should we add it to RRG repo ? Else this issue should rather be discussed at https://github.com/J-Run/mf_nonce_brute

ikarus23, 2019-10-26:
Hi. I've come across such a card as well. I get the same nt as @iceman1001 had in https://github.com/RfidResearchGroup/proxmark3/issues/133#issuecomment-544951144 but my nt2enc is different:
```
#db# rtr=17 isOK=0 min=160 max=160 avg=160, delta_time=2656          
#db# Nonce#1: Testing nt1=01200145 nt2enc=81c0565e nt2par=70          
#db# Nonce#1: valid, ntdist=160          
#db# Nonce#2: Testing nt1=01200145 nt2enc=81c0565e nt2par=70          
#db# Nonce#2: dismissed (= nonce#1), ntdist=160          
#db# Nonce#2: Testing nt1=01200145 nt2enc=81c0565e nt2par=70          
#db# Nonce#2: dismissed (= nonce#1), ntdist=160 
```
I'm not sure if I understand you guys correctly: Can I recover the keys if I know one key? Does it work for both PRNGs (weak/hard)? How does it work?

iceman1001, 2019-10-28:
@doegox   remember that tools/Makefile you wanted to remove stuff from?  ...thats how I link to external tools :)
This issue is about adding support or a solution for this fixed nonce tag,   how the bruteforce doesn't work is off topic.

Making a nested authentication and save the statelists,  compare,  and use the intersection to find a valid card sounds more or less lite j-run's implementation which has a offline phase1 and online against card in phase2...    It sounds like this is how @xtigmh  solves his card.

xtigmh, 2019-10-28:
Yes, it need bruteforce like j-run's implementation phase2.

doegox, 2019-10-28:
> @doegox remember that tools/Makefile you wanted to remove stuff from? ...thats how I link to external tools :)

Sigh, no comment

iceman1001, 2019-12-28:
So,  offical repo and the user @uzlonewolf  has about the same idea as @xtigmh  had.
Run nested with one known key,   get one loooong statelist and do check keys with all of them.

As a side bonus,   @pwpiwi did come up with some impressive check keys speed ups.  Its up to 206 auths / second now,  from 85,...  Massive speedup.  All in once timeout change.

I will implement a seperate command for this type of solution, since I don't like to mess up functions even more with edge cases.

ref:
https://github.com/Proxmark/proxmark3/issues/899

iceman1001, 2020-01-14:
Ok, I managed to make an implementation of **hf mf staticnested** today.
It actually works.  When the keyspace goes high up, its slow,  but it uses the improvements from old nested to check found keys against unknown sectors.

I think I pretty soon can close this one :)

iceman1001, 2020-01-14:
Done! 
https://github.com/RfidResearchGroup/proxmark3/commit/b37a4c14eb497b431f7443b9f685d7f2e222bfa0

closed 2020-01-14 by iceman1001