RfidResearchGroup/proxmark3 #134: make sytle - CRLF convertion

vratiskol, 2019-03-12:
**Describe the bug**
 The perl command wich convert CRLF is not working properly with proxspace
Loads of file are "convert" and backup.

**To Reproduce**
Run a make style or run command on proxspace env.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop (please complete the following information):**
- OS: Windows 10

I propose to use something like this to force LF:

**File:** .gitattributes
```
# Force LF
*.c text=auto eol=lf
*.h text=auto eol=lf
```

iceman1001, 2019-03-12:
yeah..  thats what happend to me too.  Had to make a complete revert on my local repo.  :) 
I think I have me git to ignorelineendings right now.

doegox, 2019-03-12:
Ok, I can't test on proxspace, please push a fix if you can.
To remove the linefeed conversion in the makefile, just remove the "\r" in the perl line and keep all the rest.
Thanks!

doegox, 2019-03-28:
https://github.com/RfidResearchGroup/proxmark3/pull/143 merged and augmented to .cpp, .py, .pl, Makefile.

Note that from what I understand, if you *need* CRLF on Windows on these files, try removing `eol=lf` from gitattributes, the `text=auto` should be enough to be cross-platform and getting CRLF on Windows and LF on Linux. You might need to delete and fetch again the workspace to see the effect.

vratiskol, 2019-03-28:
This one could be close now :)

closed 2019-03-28 by RfidResearchGroup