RfidResearchGroup/proxmark3 #135: EMV with old proxmark3

ikarus23, 2019-03-16:
I've used the new compile option `make PLATFORM=PM3OLD512` for my old proxmark3 with 512k memory. However, I get the following errors with the emv function:

```
[=] UART Setting serial baudrate 460800

 [ Proxmark3 RFID instrument ]
          

 [ CLIENT ]          
  client: iceman  
          
 [ ARM ]
 bootrom: master/v3.0.1-401-g53edb04-suspect 2018-09-13 18:14:59
      os: iceman/master/9f9ee2f1 2019-03-16 12:49:18

 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 230598 bytes (44%) Free: 293690 bytes (56%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
pm3 --> emv search
[=] Channel: CONTACTLESS          
[!!] APDU: Reply timeout.          
[!!] APDU: Reply timeout.          
UART:: write time-out
[!] sending bytes to proxmark failed          
[!!] APDU: Reply timeout.          
[-] Retry failed [A00000000305076010]. Skipped...          
UART:: write time-out
[!] sending bytes to proxmark failed          
[!!] APDU: Reply timeout.          
UART:: write time-out
[!] sending bytes to proxmark failed  
...
```

iceman1001, 2019-03-17:
Did you have an EMV based card over the antenna?

ikarus23, 2019-03-17:
Yes, normal credit card.

doegox, 2019-03-17:
I confirm the issue

doegox, 2019-03-17:
```
    switch (channel) {
        case ECC_CONTACTLESS:
            res = ExchangeAPDU14a(data, ...
```
will hardly work with my ISO14443B card...
We need to support typeB as well for EMV!

doegox, 2019-03-17:
Second issue: in that situation the proxmark (even the RDV4) is becoming unresponsive to the point it needs to be unplugged. It shouldn't.

ikarus23, 2019-03-17:
@doegox Thanks for confirming.
My card is ISO14443A. 

>in that situation the proxmark (even the RDV4) is becoming unresponsive to the point it needs to be unplugged.

Sorry for not mentioning that. I've noticed this too.

iceman1001, 2019-03-17:
ok,  the EMV commands can make my pm3 also unresponsive...  I believe there needs to be  wait for response timeout that is missing.

iceman1001, 2019-04-15:
@merlokk   if you have time ?

merlokk, 2019-04-15:
i will have it in 1-2 weeks and fix this

merlokk, 2019-07-16:
now emv not hangs. have checked with:
- no card in field
- mifare plus card in field
- get emv card from field when transaction is executed

merlokk, 2019-07-16:
I think it because higher stability of usb link

iceman1001, 2019-07-17:
Looking great! 
But part of this issues was 14B support.  I believe close this issue and start a new one for 14B EMV support etc..

closed 2019-07-17 by merlokk