RfidResearchGroup/proxmark3 #138: [idea] Ignore BCC errors in select iso1443a tags process

McEloff, 2019-03-18:
There are UID changeable Chinese ultralight ev1 tags without backdoor commands. UID changes by direct write pages 0-2. This cards not automatically recalculates BCC0/BCC1 during select tag process by reader. With wrong BCCs on card any reader cannot complete select tag for read/write. If method iso14443a_select_card(...) uses original BCC in "select tag" command, received from tag on previous step, selection process will be succeded. Then BCCs can be corrected by rewriting pages 0-2 with "hf mfu wrbl" command. For example see attached diff.
[ignore-BCC-errors-in-select-iso1443a-tags-process.txt](https://github.com/RfidResearchGroup/proxmark3/files/2980636/ignore-BCC-errors-in-select-iso1443a-tags-process.txt)

iceman1001, 2019-03-18:
Interesting concept.  The bcc check is there to make sure the communications with the tag works well.
Adding an extra option from the client would be viable,  but not as your suggested solution to blindly trust a bad BCC.   

_Support for magic gen2 cards with bad bcc._

iceman1001, 2019-03-21:
Have you tried to send **hf 14a raw** commands direct ?

**hf 14a raw -c **  without selecting tag and send the write commands direct`?

McEloff, 2019-03-21:
Without tag selecting execution of "**hf 14a raw**" failed on any card, and then proxmark has been shutdown (restarts with LED blinking): 
pm3 --> hf 14a raw -c 30 02
[!] timeout while waiting for reply.
pm3 --> hw status
[!] sending bytes to proxmark failed
Status command failed. USB Speed Test timed out
pm3 -->

Branch master (last commit ca9061bd). Compiled for PM3EVO platform.

iceman1001, 2019-03-21:
hm... interesting that it fails like that...   Have you flashed your device?   I am testing the same cmd and don't get that behavior

McEloff, 2019-03-21:
[ Proxmark3 RFID instrument ]
 [ CLIENT ]
  client: iceman
 [ ARM ]
 bootrom: iceman/master/ca9061bd-dirty-unclean 2019-03-21 22:05:16
      os: iceman/master/ca9061bd-dirty-unclean 2019-03-21 22:05:42
 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23
 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 232516 bytes (44%) Free: 291772 bytes (56%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2019-03-21:
looks up-to-date.

iceman1001, 2019-04-22:
@McEloff   I think either making a lua-script using 14a raw commands,  to ignore BCC value.
or its built in as a optional param.  Not like your patch.

McEloff, 2019-04-22:
Ok. Really add to method iso14443a_select_card() boolean parameter to switch that by command line option? There whil be not small refactoring. Or may be like 'hf mf dbg' add static variable to switch on/off selection method?

McEloff, 2019-05-09:
Released in my local repository, not actual now

doegox, 2020-06-25:
I think it's still something useful to implement, reopening this issue

doegox, 2020-09-06:
I recently faced a similar issue where ATQA was wrong and pointed to proprietary anticol.
So I implemented configuration options in https://github.com/RfidResearchGroup/proxmark3/commit/b751354ea2ae11134b974ef7f303058239ff0c5c
E.g.
[image]

closed 2020-09-06 by doegox