RfidResearchGroup/proxmark3 #139: [hf mf hardnested] loop bug in PartialSumProperty function

iceman1001, 2019-03-18:
**Describe the bug**
cmdhfmfhard.c ,  function PartialSumProperty  has a ODD_START loop.  Which **i** loop variable is the interesting one for this bug.  It goes from 0-4,  while being used for shifting in **3-i**  rending some sideeffects.  Now changing the loop is easy however the bitflip precalculated binaries is also generated with the same loop but in another file.   Changing in one place breaks the hardnested command.

```
      if (odd_even == ODD_STATE) {
            for (uint16_t i = 0; i < 5; i++) {
                part_sum ^= filter(st);
                st = (st << 1) | ((j >> (3 - i)) & 0x01) ;
            }
            part_sum ^= 1; // XOR 1 cancelled out for the other 8 bits
        }
```

**To Reproduce**
look at code sample.

**Expected behavior**
proper loops?

**Screenshots**
N/A

**Desktop (please complete the following information):**
latest source.

**Additional context**

iceman1001, 2019-04-08:
There is a main function inside  client/hardnested/hardnested_tables.c  which indicates a way of recalculating the table.

iceman1001, 2019-04-09:
With the lack of documentation in the hardnested_tables.c files we have a problem of figuring out  which files needs to be generated,  like which combos of bitflip / sumA0.   Only one person who knows the dark arts of the hardnested precalculated bitflip files,  @pwpiwi

doegox, 2019-04-09:
The generated tables differ indeed slightly between the "i < 4" and the "i < 5" versions, so we need to regenerate them. But as @iceman1001 pointed out, how to go from which executions of hardnested_tables to the actual tables present in pm3 repo? (besides the obvious zlib compression).
Current tests are done in branch https://github.com/RfidResearchGroup/proxmark3/tree/hardnested_wip

iceman1001, 2019-04-09:
Maybe the negative shift was planned?

iceman1001, 2019-04-09:
A nonce file to test with.
Use   **hf mf hard r f hf-mf-6E9AAF06-nonces**

try it with the  i < 4 vs i < 5,

[hf-mf-6E9AAF06-nonces.zip](https://github.com/RfidResearchGroup/proxmark3/files/3059708/hf-mf-6E9AAF06-nonces.zip)

iceman1001, 2019-04-09:
ok,  the failing version gives a strong indication that for each bitflip,  all sum(a8) is used.
I guess the outcome is then merged into one file per bitflip...

doegox, 2019-04-09:
Ok, I fixed the silly bug in 258332894b6e65a39b7e5e2d69ba95c3a7d90b19, still I'm interested to hear how to generate the tables.

closed 2019-04-09 by iceman1001