RfidResearchGroup/proxmark3 #141: [idea] Make hardnested the same as nested

ikarus23, 2019-03-19:
From an users perspective, I would really like if the hardnested command works as same as the nested. Just run the command, get all the keys and additional options to dump them into a file or write them to the emulator memory, etc.

iceman1001, 2019-03-19:
Feel free to add it. 
I already have my hard_autopwn script so my needs are fulfilled.

ikarus23, 2019-03-19:
> Feel free to add it.

I'm not sure if I will find the time to. It's not request, it's just an idea to improve the proxmark.

> I already have my hard_autopwn script so my needs are fulfilled.

I'm happy for you.
Sharing is caring ;) (also not a request)

doegox, 2019-08-28:
`hf mf autopwn` is filling this void now

closed 2019-08-28 by doegox

ikarus23, 2019-08-30:
Thank you! I love where this project is going!

iceman1001, 2019-08-30:
I think we all feel the love.  There is a good feeling going on here.