RfidResearchGroup/proxmark3 #149: lf t55xx - modulation tests

iceman1001, 2019-04-08:
**Describe the problems**
@doegox and me has spent some time in making lf t55xx commands better.  Especially when it comes to demodulation of the different clockspeeds.  In order to make testing easer we have created a lua script,  **test_t55xx.lua**  where you can edit in order to test a specific modulation or all of them.
Our intention is that we should get 32/32 for a test,  or  100% .  Sadly it isn't that right now. Below I have added current biphase status,  it gives a great result of 31/32.   Try ASK and you be suppresed the fail tests.  

**To Reproduce**
Steps to reproduce the behavior:
1. script run test_t55x7 

**2019-04-08T14:50:43Z remark: [SUMMARY] Success rate: 31/32 tests passed, help me improving that number!** 

**Expected behavior**
We want the output on those tests to say,  32/32

**Desktop (please complete the following information):**
-- running latest firmware/client

Please help us with testing and corrections!

merlokk, 2019-04-08:
China's clone T5577

2019-04-08T15:26:10Z remark: [SUMMARY] Success rate: 39/64 tests passed, help me improving that number!

<details> 
  <summary>log here</summary>
pm3 /pm3i/client$ ./proxmark3 com7
[=] UART Setting serial baudrate 460800

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: iceman build for RDV40 with flashmem; smartcard;

 [ ARM ]
 bootrom: iceman/master/ice_v3.1.0-891-g6967fe66-dirty-unclean 2018-06-26 09:52:13
      os: iceman/master/b0d627ba-dirty-unclean 2019-04-08 18:24:36

 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 250620 bytes (48%) Free: 273668 bytes (52%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

pm3 --> script run test_t55x7
[+] Executing: test_t55x7.lua, args ''

----------------------------------------
----------------------------------------
Wiping card

[=] Beginning Wipe of a T55xx tag (assuming the tag is not password protected)

[=] Writing page 0  block: 00  data: 0x000880E0 pwd: 0x00000000
[=] Writing page 0  block: 01  data: 0x00000000
[=] Writing page 0  block: 02  data: 0x00000000
[=] Writing page 0  block: 03  data: 0x00000000
[=] Writing page 0  block: 04  data: 0x00000000
[=] Writing page 0  block: 05  data: 0x00000000
[=] Writing page 0  block: 06  data: 0x00000000
[=] Writing page 0  block: 07  data: 0x00000000
Detecting card
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 2 - RF/32
Inverted   : No
Offset     : 32
Seq. Term. : Yes
Block0     : 0x000880E0

[=] Writing page 0  block: 01  data: 0x00000000
[=] Writing page 0  block: 02  data: 0xFFFFFFFF
[=] Writing page 0  block: 03  data: 0x80000000
[=] Writing page 0  block: 04  data: 0x00000001
ASK
Start test of ASK
[+] Found [2] possible matches for modulation.
--[1]---------------
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 0 - RF/8
Inverted   : No
Offset     : 30
[... 223 lines cut ...]

Test 00000000 == 00000000 OK
Test ffffffff == 00000000 Failed
2019-04-08T15:26:08Z remark: [ERR:READ:00181040:2] block 2: read 00000000 instead of ffffffff
Test 80000000 == 80000000 OK
Test 00000001 == 00000001 OK
2019-04-08T15:26:08Z remark: [SUMMARY:00181040] FAIL 1 test
Chip Type  : T55x7
Modulation : PSK1
Bit Rate   : 7 - RF/128
Inverted   : No
Offset     : 31
Seq. Term. : No
Block0     : 0x001C1040

Test 00000000 == 00000000 OK
Test ffffffff == 00000000 Failed
2019-04-08T15:26:09Z remark: [ERR:READ:001c1040:2] block 2: read 00000000 instead of ffffffff
Test 80000000 == 80000000 OK
Test 00000001 == 00000001 OK
2019-04-08T15:26:10Z remark: [SUMMARY:001c1040] FAIL 1 test
----------------------------------------
-----------------

</details>

merlokk, 2019-04-08:
majority of my bugs is `Test 00000000 == FFFFFFFF Failed`
Some of my cards: if PM writes there 0000... PM shows fffff..  but it writes correct. I see it from another reader

```

pm3 --> lf t5 wipe

[=] Beginning Wipe of a T55xx tag (assuming the tag is not password protected)
[=] Writing page 0  block: 00  data: 0x000880E0 pwd: 0x00000000
[=] Writing page 0  block: 01  data: 0x00000000
[=] Writing page 0  block: 02  data: 0x00000000
[=] Writing page 0  block: 03  data: 0x00000000
[=] Writing page 0  block: 04  data: 0x00000000
[=] Writing page 0  block: 05  data: 0x00000000
[=] Writing page 0  block: 06  data: 0x00000000
[=] Writing page 0  block: 07  data: 0x00000000

pm3 --> lf t5 dump
Reading Page 0:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 00 | 000880E0 | 00000000000010001000000011100000 | ....
 01 | FFFFFFFF | 11111111111111111111111111111111 | ....
 02 | FFFFFFFF | 11111111111111111111111111111111 | ....
 03 | FFFFFFFF | 11111111111111111111111111111111 | ....
 04 | FFFFFFFF | 11111111111111111111111111111111 | ....
 05 | FFFFFFFF | 11111111111111111111111111111111 | ....
 06 | FFFFFFFF | 11111111111111111111111111111111 | ....
 07 | FFFFFFFF | 11111111111111111111111111111111 | ....
Reading Page 1:
blk | hex data | binary                           | ascii
----+----------+----------------------------------+-------
 00 | 000880E0 | 00000000000010001000000011100000 | ....
 01 | E03900D0 | 11100000001110010000000011010000 | .9..
 02 | 28DC9C18 | 00101000110111001001110000011000 | (...
 03 | 00A00003 | 00000000101000000000000000000011 | ....

```

doegox, 2019-04-08:
yes normally it always writes correctly, but there are issues reading back with some modulations. E.g. if you change only block0 to a stable one (e.g. biphase, middle rate), you'll be able to read the content again.

merlokk, 2019-04-08:
i cant read at EM block( 
Manchester + RF/64
I look into code and found that it "shifts" modulation by 1/2 of RF/64 and founds 0xffff...   instead of 0x0000
only card that works - one that i have from proxmark kit)

merlokk, 2019-04-08:
[image]
))))

iceman1001, 2019-04-08:
if you used the token with the duplicator (like the blue gun)  then the tag is password protected.
Look into the dictionary file,  the three most common pwd for those duplicators is in it

iceman1001, 2019-04-08:
@merlokk  you have the old LF dual antenna on your rdv4,   it need the keyfob / tag to be direct over the antenna

iceman1001, 2019-04-08:
another tip when trying individual configs,   set the debug. 
`data setd 1`

3ldidi94, 2019-04-08:
China's clone t5577, you can see the picture of the tag
[image]

`pm3 --> script run test_t55x7
[+] Executing: test_t55x7.lua, args ''
          
----------------------------------------
----------------------------------------
Wiping card
          
[=] Beginning Wipe of a T55xx tag (assuming the tag is not password protected)
          
[=] Writing page 0  block: 00  data: 0x000880E0 pwd: 0x00000000          
[=] Writing page 0  block: 01  data: 0x00000000           
[=] Writing page 0  block: 02  data: 0x00000000           
[=] Writing page 0  block: 03  data: 0x00000000           
[=] Writing page 0  block: 04  data: 0x00000000           
[=] Writing page 0  block: 05  data: 0x00000000           
[=] Writing page 0  block: 06  data: 0x00000000           
[=] Writing page 0  block: 07  data: 0x00000000           
Detecting card
Chip Type  : T55x7          
Modulation : ASK          
Bit Rate   : 2 - RF/32          
Inverted   : No          
Offset     : 32          
Seq. Term. : Yes           
Block0     : 0x000880E0          
          
[=] Writing page 0  block: 01  data: 0x00000000           
[=] Writing page 0  block: 02  data: 0xFFFFFFFF           
[=] Writing page 0  block: 03  data: 0x80000000           
[=] Writing page 0  block: 04  data: 0x00000001           
ASK
Start test of ASK
[+] Found [2] possible matches for modulation.          
--[1]---------------          
Chip Type  : T55x7          
Modulation : ASK          
Bit Rate   : 0 - RF/8          
Inverted   : No          
Offset     : 29          
Seq. Term. : Yes           
Block0     : 0x00008040          
          
--[2]---------------          
Chip Type  : T55x7          
Modulation : BIPHASEa - (CDP)          
Bit Rate   : 0 - RF/8          
Inverted   : Yes           
Offset     : 30          
Seq. Term. : No          
Block0     : 0x000180C0          
          
can't detect modulation, skip to next config
2019-04-08T18:52:28Z remark: [ERR:DETECT:00008040] Failed to detect modulation          
2019-04-08T18:52:28Z remark: [SUMMARY:00008040] FAIL detection          
Chip Type  : T55x7          
Modulation : ASK          
Bit Rate   : 1 - RF/16          
Inverted   : No          
Offset     : 33          
Seq. Term. : Yes           
Block0     : 0x00048040          
          
Test 00000000 == FFFFFFFF Failed
2019-04-08T18:52:28Z remark: [ERR:READ:00048040:1] block 1: read FFFFFFFF instead of 00000000          
Test ffffffff == FFFFFFFF OK
Test 80000000 == 00000001 Failed
2019-04-08T18:52:29Z remark: [ERR:READ:00048040:3] block 3: read 00000001 instead of 80000000          
[... 196 lines cut ...]
2019-04-08T18:52:48Z remark: [ERR:READ:00181040:2] block 2: read 00000000 instead of ffffffff          
Test 80000000 == 80000000 OK
Test 00000001 == 00000001 OK
2019-04-08T18:52:49Z remark: [SUMMARY:00181040] FAIL 1 test          
Chip Type  : T55x7          
Modulation : PSK1          
Bit Rate   : 7 - RF/128          
Inverted   : No          
Offset     : 31          
Seq. Term. : No          
Block0     : 0x001C1040          
          
Test 00000000 == 00000000 OK
Test ffffffff == 00000000 Failed
2019-04-08T18:52:50Z remark: [ERR:READ:001c1040:2] block 2: read 00000000 instead of ffffffff          
Test 80000000 == 80000000 OK
Test 00000001 == 00000001 OK
2019-04-08T18:52:50Z remark: [SUMMARY:001c1040] FAIL 1 test          
----------------------------------------
----------------------------------------
Tests finished

2019-04-08T18:52:50Z remark: [SUMMARY] Success rate: 41/64 tests passed, help me improving that number!          
          
[+] Finished`

iceman1001, 2019-04-08:
yeah,  I see a strong tendecy with 0xFF FF FF FF gets wrong,

merlokk, 2019-04-08:
And i dont see how to fix it(
it can be fixed with using of ST terminator (but its not a fix) )
or needs to grab t5577 from beginning of transmit. there is a `0` sequence and then a code

doegox, 2019-04-08:
Yeah actually the FFFFFFFF is a corner case because you'll encounter it only when forcing a single block read (or emulating all FF UID, a bit pointless).
The only way to distinguish it from 00000000 in some modulations (ASK, PSK1?) is to decide which one is the very first transition in Manchester signal. Some other modulations don't have that problem of distinguishing Manchester symbols, e.g. Biphase, FSK, PSK2

iceman1001, 2019-04-09:
I have some more ideas to test.   Since the new noise detection is in place,   starting to sample the response a bit earlier makes use get the inital response better.   Before we have trimmed the signal  since we wanted to let the signal settle in.

merlokk, 2019-04-10:
as i saw biphase and manchester hard to separate too

TomHarkness, 2019-04-25:
Ok so this is all of the different t5577's that I have - 10 in total! Tested from left to tight and results pasted in the same order. 

I will also test the Dangerous Things XEM implant with a custom coil that will be released soon. This t5577 is kind of backdoored to allow all blocks of page 1 written. The success rate is very bad even with the custom coil.

[image]

**KEYCHAINS*

**#1**

`pm3 --> script run test_t55x7
[+] Executing: test_t55x7.lua, args ''

----------------------------------------
----------------------------------------
Wiping card

[=] Beginning Wipe of a T55xx tag (assuming the tag is not password protected)

[=] Writing page 0  block: 00  data: 0x000880E0 pwd: 0x00000000
[=] Writing page 0  block: 01  data: 0x00000000
[=] Writing page 0  block: 02  data: 0x00000000
[=] Writing page 0  block: 03  data: 0x00000000
[=] Writing page 0  block: 04  data: 0x00000000
[=] Writing page 0  block: 05  data: 0x00000000
[=] Writing page 0  block: 06  data: 0x00000000
[=] Writing page 0  block: 07  data: 0x00000000
Detecting card
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 2 - RF/32
Inverted   : No
Offset     : 31
Seq. Term. : Yes
Block0     : 0x000880E0

[=] Writing page 0  block: 01  data: 0x00000000
[=] Writing page 0  block: 02  data: 0xFFFFFFFF
[=] Writing page 0  block: 03  data: 0x80000000
[=] Writing page 0  block: 04  data: 0x00000001
ASK
Start test of ASK
[+] Found [2] possible matches for modulation.
--[1]---------------
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 0 - RF/8
Inverted   : No
Offset     : 57
Seq. Term. : Yes
Block0     : 0x00008040

--[2]---------------
Chip Type  : T55x7
Modulation : BIPHASEa - (CDP)
Bit Rate   : 0 - RF/8
Inverted   : Yes
Offset     : 58
Seq. Term. : No
Block0     : 0x000180C0

can't detect modulation, skip to next config
2019-04-25T02:17:34Z remark: [ERR:DETECT:00008040] Failed to detect modulation
2019-04-25T02:17:34Z remark: [SUMMARY:00008040] FAIL detection
Chip Type  : T55x7
Modulation : ASK
Bit Rate   : 1 - RF/16
Inverted   : No
Offset     : 29
Seq. Term. : Yes
[... 3716 lines cut ...]
can't detect modulation, skip to next config
2019-04-25T02:46:46Z remark: [ERR:DETECT:00081040] Failed to detect modulation
2019-04-25T02:46:46Z remark: [SUMMARY:00081040] FAIL detection
can't detect modulation, skip to next config
2019-04-25T02:46:46Z remark: [ERR:DETECT:000c1040] Failed to detect modulation
2019-04-25T02:46:46Z remark: [SUMMARY:000c1040] FAIL detection
can't detect modulation, skip to next config
2019-04-25T02:46:46Z remark: [ERR:DETECT:00101040] Failed to detect modulation
2019-04-25T02:46:46Z remark: [SUMMARY:00101040] FAIL detection
can't detect modulation, skip to next config
2019-04-25T02:46:47Z remark: [ERR:DETECT:00141040] Failed to detect modulation
2019-04-25T02:46:47Z remark: [SUMMARY:00141040] FAIL detection
can't detect modulation, skip to next config
2019-04-25T02:46:47Z remark: [ERR:DETECT:00181040] Failed to detect modulation
2019-04-25T02:46:47Z remark: [SUMMARY:00181040] FAIL detection
can't detect modulation, skip to next config
2019-04-25T02:46:47Z remark: [ERR:DETECT:001c1040] Failed to detect modulation
2019-04-25T02:46:47Z remark: [SUMMARY:001c1040] FAIL detection
----------------------------------------
----------------------------------------
Tests finished

2019-04-25T02:46:47Z remark: [SUMMARY] Success rate: 19/64 tests passed, help me improving that number!

[+] Finished`

iceman1001, 2021-01-27:
ASK/Biphase demod is now 32 of 32 test pass :)

iceman1001, 2026-09-12:
And with all changes lately I think it is time to actually close this one.

closed 2026-09-12 by iceman1001