RfidResearchGroup/proxmark3 #156: Proxmark3 legic sim issue

rafaelmertens, 2019-04-11:
Hi,
I have the Proxmark 3 RDV4 and tried simulation of a legic prime card.
I saw the #83 and I thought at first that I had the same issue, but the version I am running already has the timeout set to 120.

I tried changing it to 40, 80, 160, 200, 240 without any success..

Here is some basic information

```
pm3 --> hw version

Proxmark3 RFID instrument
          

 [ CLIENT ]          
 client: iceman build for RDV40 with flashmem; smartcard;  
          
 [ ARM ]
 bootrom: Version information not available
      os: Version information not available

 [ FPGA ]
 LF image built for 2s30vq100 on 2018/ 9/ 8 at 13:57:51
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev A          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 245529 bytes (47%) Free: 278759 bytes (53%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
pm3 --> hw status
#db# Memory          
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........35904          
#db# Tracing          
#db#   tracing ................0          
#db#   traceLen ...............6438          
#db# Currently loaded FPGA image          
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
#db# Flash memory          
#db#   Baudrate................24MHz          
#db#   Init....................OK          
#db#   Memory size.............2 mbits / 256kb          
#db#   Unique ID...............0xd567a882a77cbe25          
#db# Smart card module (ISO 7816)          
#db#   version.................v3.0a          
#db# LF Sampling config          
#db#   [q] divisor.............95 (125 KHz)          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config          
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............15*8 (120)          
#db# USB Speed          
#db#   Sending USB packets to client...          
#db#   Time elapsed............1500ms          
#db#   Bytes transferred.......774144          
#db#   USB Transfer Speed PM3 -> Client = 516096 Bytes/s          
#db# Various          
#db#   MF_DBGLEVEL.............1          
[... 67 lines cut ...]
      52582 |      52722 | Rdr |07  5d  00  00                                                           |     | IV 0x5D          
      52792 |      52918 | Tag |06  1d  00                                                               |     | MIM256          
      52971 |      53073 | Rdr |06  39  00  00                                                           |     | ACK 256/1024          
      53131 |      53315 | Rdr |09  01  00  00                                                           |     | READ Byte(0)          
      53385 |      53637 | Tag |0c  79  04                                                               |     |           
      53861 |      54037 | Rdr |09  03  00  00                                                           |     | READ Byte(1)          
      54107 |      54359 | Tag |0c  7d  0d                                                               |     |           
      54583 |      54735 | Rdr |09  05  00  00                                                           |     | READ Byte(2)          
      54805 |      55057 | Tag |0c  0f  06                                                               |     |           
      55281 |      55441 | Rdr |09  07  00  00                                                           |     | READ Byte(3)          
      55511 |      55763 | Tag |0c  90  0d                                                               |     |           
      55987 |      56147 | Rdr |09  09  00  00                                                           |     | READ Byte(4)          
      56217 |      56469 | Tag |0c  1f  08                                                               |     |           
      59213 |      59381 | Rdr |09  56  01  00                                                           |     | WRITE Byte(171)          
      59927 |      60071 | Rdr |09  09  00  00                                                           |     | READ Byte(4)          
      60617 |      60793 | Rdr |09  9e  01  00                                                           |     | WRITE Byte(207)          
      61171 |      61315 | Rdr |09  01  01  00                                                           |     | READ Byte(128)          
      83055 |      83195 | Rdr |07  57  00  00                                                           |     | IV 0x57          

```

When I am simulating it the LED B and C flash, but the reader doesn't react.
I've been trying to find a way of solving this, but I ran out of ideas..
Could someone point me in the right direction?
Am I doing something wrong?

iceman1001, 2019-04-11:
Your legic tag got identified.  You dumped it.   You loaded it to emulator memory. 
And you started the simulation.  The trace shows success with communicating with the reader.

Have you verified that the dump contains data?  You would need to get a trace from your tag talking with the reader in order to see what fails.  
Or dump tag,   present tag to reader, dump tag,  compare tags and identify which bytes changed.
Then look at the sim trace to see if the sim also changed those bytes...

rafaelmertens, 2019-04-11:
Hi @iceman1001,

The dump contains only a bit of data..  here is the content:
`00000000: 797d 0f90 1f60 ea48 0201 0ee3 0665 42ee  y}...`.H.....eB.
00000010: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000020: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000030: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000040: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000050: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000060: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000070: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000080: 0000 0000 0000 0000 0000 0000 0000 0000  ................
00000090: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000a0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000b0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000c0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000d0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000e0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
000000f0: 0000 0000 0000 0000 0000 0000 0000 0000  ................
`
I don't have access to the reader right now. Will check that later or tomorrow.
But if it changes some bytes, how would I be able to emulate that?

iceman1001, 2019-04-11:
the legic sim should support full sim, with both read / write.  As seen in your trace earlier.

To verify the dump command isnt broken,  dump tag twice and compare.

rafaelmertens, 2019-04-11:
I was able to get access to the reader shortly.
I simulated and dumped the tag after it. Both dumps look exactly the same... No changes..

iceman1001, 2019-04-11:
What is the output from:
`
hf legic info`

rafaelmertens, 2019-04-12:
Sorry for the delay.

Here is the output:
```
pm3 --> hf legic info
Reading tag memory 256 b...          

          

CDF: System Area          
------------------------------------------------------          
MCD: 79, MSN: 7d 0f 90, MCC: 1f OK          
DCF: 60000 (60 ea), Token Type=IM (OLE=0)          

ADF: User Area          
------------------------------------------------------          
Unsegmented card - WRP: 08, WRC: 04, RD: 0          
WRC protected area:   (I 8 | WRC 4)          

row  | data          
-----+------------------------------------------------          
[00] | 02 01 0E E3 
Remaining write protected area:  (I 12 | WRC 4 | WRP 8 | WRP_LEN 4)          

row  | data          
-----+------------------------------------------------          
[00] | 06 65 42 EE 
Remaining segment payload:  (I 16 | Remain LEN 994)          

row  | data          
-----+------------------------------------------------          
[00] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[01] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[02] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[03] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[04] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[05] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[06] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[07] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[08] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[09] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[11] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[12] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[13] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[14] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[15] | 00 00 00 00 00 00 00 00 11 8B 00 00 00 00 00 00 
[16] | 30 00 08 64 65 66 61 75 6C 74 5F 69 63 6C 61 73 
[17] | 73 5F 6B 65 79 73 2E 64 69 63 00 FF FF FF FF 08 
[18] | 6D 1C 30 00 00 00 00 00 0C 82 EC FF 3F 01 39 15 
[19] | 20 00 08 73 63 61 6E 64 69 72 2E 68 00 19 8F 08 
[20] | 4A 1A 30 00 00 00 00 00 F5 2B 91 FD B9 03 56 16 
[21] | 28 00 08 63 6D 64 6C 66 70 61 72 61 64 6F 78 2E 
[22] | 68 00 08 63 6D 64 6C 08 3D 1A 30 00 00 00 00 00 
[23] | 79 05 9B 82 7F 0A 64 16 20 00 08 63 6D 64 6C 66 
[24] | 69 6F 2E 63 00 00 44 08 0B 1A 30 00 00 00 00 00 
[25] | 77 E4 66 6A 96 BB AC 16 20 00 08 63 6D 64 68 66 
[26] | 31 34 61 2E 63 00 31 08 19 1A 30 00 00 00 00 00 
[27] | 41 7B 4E 8D 7D 6E 18 19 20 00 08 63 6D 64 68 66 
[28] | 6C 65 67 69 63 2E 63 00 2E 1A 30 00 00 00 00 00 
[29] | 81 30 9A 01 01 4E 59 1B 20 00 08 63 6D 64 6C 66 
[30] | 61 77 69 64 2E 68 00 08 57 1A 30 00 00 00 00 00 
[31] | F8 E5 58 08 F5 C0 87 1B 28 00 08 63 6D 64 6C 66 
[32] | 76 69 6B 69 6E 67 2E 63 00 2E 68 74 32 00 00 08 
[33] | 5D 1C 30 00 00 00 00 00 31 08 EB 68 82 C3 E6 1B 
[34] | 28 00 08 70 72 6F 78 67 75 69 71 74 2E 63 70 70 
[35] | 00 00 FB 8D 06 08 AC 08 5F 1C 30 00 00 00 00 00 
[36] | 42 3A A6 4F 24 A9 40 1D 20 00 08 70 72 6F 78 6D 
[37] | 61 72 6B 33 2E 63 00 08 2C 1A 30 00 00 00 00 00 
[38] | 87 23 DB 48 8F 70 BB 1E 20 00 08 63 6D 64 6C 66 
[39] | 2E 68 00 70 6D 33 5F 08 9F 1C 30 00 00 00 00 00 
[40] | 23 4E 9E F4 28 EF 67 1F 20 00 08 75 74 69 6C 2E 
[41] | 63 00 08 63 6D 64 6C 08 32 1A 30 00 00 00 00 00 
[42] | 68 C3 B7 3F 1F F8 66 20 20 00 08 63 6D 64 6C 66 
[43] | 65 6D 34 78 2E 68 00 08 1D 1A 30 00 00 00 00 00 
[44] | EA 71 9D E7 2A 03 AE 20 20 00 08 63 6D 64 68 66 
[45] | 6D 66 2E 63 00 F4 EE 08 F8 12 30 00 00 00 00 00 
[46] | 0F 21 D7 8C 10 5E F8 21 20 00 08 37 39 30 33 30 
[47] | 30 39 30 2E 62 69 6E 00 97 1A 30 00 00 00 00 00 
[48] | 8E E3 AF D9 41 97 26 22 20 00 08 67 72 61 70 68 
[49] | 2E 68 00 6E 00 00 00 08 6A 1A 30 00 00 00 00 00 
[50] | B0 37 0A 30 EF 35 A9 23 20 00 08 65 6D 6C 32 55 
[51] | 50 50 45 52 2E 73 68 00 52 1A 30 00 00 00 00 00 
[52] | 15 70 BA D5 B2 D3 03 24 28 00 08 63 6D 64 6C 66 
[53] | 73 65 63 75 72 61 6B 65 79 2E 68 00 73 69 6D 08 
[54] | 00 1A 30 00 00 00 00 00 E1 AD 23 4D 38 19 8F 26 
[55] | 20 00 08 63 6C 69 2E 63 00 B8 F7 92 1F 1B D6 08 
[56] | 36 1A 30 00 00 00 00 00 94 69 7C 89 AF 06 0E 27 
[57] | 20 00 08 63 6D 64 6C 66 67 75 61 72 64 2E 68 00 
[58] | 9B 1C 30 00 00 00 00 00 E2 7E CB 02 14 02 23 28 
[59] | 18 00 04 75 69 00 79 04 4B 1A 30 00 00 00 00 00 
[60] | D4 0C 70 7D 4B 3E 71 28 28 00 08 63 6D 64 6C 66 
[61] | 70 63 66 37 39 33 31 2E 63 00 69 2E 63 00 00 08 
[62] | 6E 1C 
-----+------------------------------------------------

```

iceman1001, 2019-04-12:
Looks like the data section is 1024 bytes while you say you have a 256byte..

rafaelmertens, 2019-04-12:
I saw that, but on the start it says "Reading tag memory 256 b..." and on hf legic read it says it was MIM256, so I thought I was a 256..
Will try to dump and emulate again with 1024.
Will let you know in a bit

rafaelmertens, 2019-04-12:
Okay.. Tried with 1024, also didn't work..

After that I did a snoop from the traffic with the real card..

pm3 --> hf snoop
#db# Buffer cleared (40000 bytes)          
#db# Skipping first 0 sample pairs, Skipping 0 triggers.
          
#db# Trigger kicked! Value: 191, Dumping Samples Hispeed now.          
#db# HF Snoop end

pm3 --> hf list
Recorded Activity (TraceLen = 40000 bytes)          
          
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
iso14443a - All times are in carrier periods (1/13.56Mhz)          
          
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |      32902 | Rdr |77! 75! 74  73! 72! 72! 71! 71! 71! 70! 70! 70! 70  70  70  70  70  70!  |     |           
            |            |     |70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70!  |     |           
            |            |     |70  70  70  70  70  70! 70! 70! 6f! 70  70  70  70  70! 70! 70! 70  70   |     |           
            |            |     |70  70  70  70! 70! 70! 6f! 40  34  37  3f! 46! 4d  53  58  5c! 5f! 62   |     |           
            |            |     |64  66  67! 69  6a! 6b  6c! 6c! 6d  6d! 6e! 6e! 6e  6e  6f! 6f! 6f! 6f   |     |           
            |            |     |6f  6f  6f! 6f! 6f! 6f! 6f! 6f  70! 6f  6f! 9c! bf  bc  ae  a0  95  8c!  |     |           
            |            |     |86  80  7d! 7a  77! 76! 74  73! 72! 72! 71! 71! 70  71  70! 70! 70  70   |     |           
            |            |     |70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70   |     |           
            |            |     |70  70! 70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70!  |     |           
            |            |     |70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70!  |     |           
            |            |     |70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70! 70  70   |     |           
            |            |     |70  70  70  70! 70! 70! 70  70  70  70  70  70! 6f  70! 70  70  70  70   |     |           
            |            |     |70  70! 70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70!  |     |           
            |            |     |70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70!  |     |           
            |            |     |70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70! 70  70   |     |           
            |            |     |70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70! 70  6f! 70  70   |     |           
            |            |     |70  70! 70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70!  |     |           
            |            |     |70! 70! 70  70  70  70  70  70! 70! 70! 70  70  70  70  70  70! 70! 70!  | !crc|           
pm3 -->

iceman1001, 2019-04-12:
ok, found that issue :)   Pull latest / compile and try HF LEGIC INFO again :)

rafaelmertens, 2019-04-12:
Here is the output

```
pm3 --> hf legic info
[+] Reading full tag memory of 256 bytes...          

          
CDF: System Area           
------------------------------------------------------          
MCD: 79, MSN: 7d 0f 90, MCC: 1f OK           
DCF: 60000 (60 ea), Token Type = IM (OLE = 0)          
------------------------------------------------------          

ADF: User Area           
------------------------------------------------------          
Unsegmented card - WRP: 08, WRC: 04, RD: 0          
WRC protected area:   (I 8 | WRC 4)          

row  | data          
-----+------------------------------------------------          
[00] | 02 01 0E E3 
-----+------------------------------------------------
          
Remaining write protected area:  (I 12 | WRC 4 | WRP 8 | WRP_LEN 4)          

row  | data          
-----+------------------------------------------------          
[00] | 06 65 42 EE 
-----+------------------------------------------------
          
Remaining segment payload:  (I 16 | Remain LEN 226)          

row  | data          
-----+------------------------------------------------          
[00] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[01] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[02] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[03] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[04] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[05] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[06] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[07] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[08] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[09] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[11] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[12] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[13] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[14] | 00 00 
-----+------------------------------------------------
          
pm3 --> 

```

iceman1001, 2019-04-12:
Good,  one little bug gone :)
Still doesn't answer why your reader doesn't answer the simulation.
Looking at your trace list from the sim,  it seem to have some missing tag responses.

pull latest, compile, back to 120? and do one more sim test against reader,  
post the trace list output here

rafaelmertens, 2019-04-12:
Okay..
So I pulled the latest, compiled and didn't change the timeout since (so it's 120).
I got a new card, just to be sure that it was not a problem with this one card...

Here is all the info and the simulations (1st with 256, 2nd with 1024 - just to be sure..)

```
pm3 --> hf legic read
[+]  UID : 79 D1 3B CD           
[+] TYPE : MIM256 card (234 bytes)          
pm3 --> hf legic info
[+] Reading full tag memory of 256 bytes...          

          
CDF: System Area           
------------------------------------------------------          
MCD: 79, MSN: d1 3b cd, MCC: 32 OK           
DCF: 60000 (60 ea), Token Type = IM (OLE = 0)          
------------------------------------------------------          

ADF: User Area           
------------------------------------------------------          
Unsegmented card - WRP: 08, WRC: 04, RD: 0          
WRC protected area:   (I 8 | WRC 4)          

row  | data          
-----+------------------------------------------------          
[00] | 02 01 0E E3 
-----+------------------------------------------------
          
Remaining write protected area:  (I 12 | WRC 4 | WRP 8 | WRP_LEN 4)          

row  | data          
-----+------------------------------------------------          
[00] | 07 10 34 7D 
-----+------------------------------------------------
          
Remaining segment payload:  (I 16 | Remain LEN 226)          

row  | data          
-----+------------------------------------------------          
[00] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[01] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[02] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[03] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[04] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[05] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[06] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[07] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[08] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[09] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[10] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[11] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[12] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[13] | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[14] | 00 00 
-----+------------------------------------------------

pm3 --> hf legic dump o _entrance_test
[+] TYPE : MIM256 card (234 bytes)          
[+] Reading tag memory 256 b...          

          
[+] Wrote 256 bytes to _entrance_test.bin 
```

1st:

```

[... 615 lines cut ...]
     261428 |     261632 | Rdr |0b  09  00  00                                                           |     | READ Byte(4)          
     261702 |     261954 | Tag |0c  32  0a                                                               |     |           
     264950 |     265146 | Rdr |0b  6e  00  00                                                           |     | WRITE Byte(55)          
     265692 |     265904 | Rdr |0b  bb  01  00                                                           |     | READ Byte(221)          
     266450 |     266654 | Rdr |0b  fa  00  00                                                           |     | WRITE Byte(125)          
     267032 |     267204 | Rdr |0b  a0  00  00                                                           |     | WRITE Byte(80)          
     288795 |     288951 | Rdr |07  7f  00  00                                                           |     | IV 0x7F          
     289021 |     289147 | Tag |06  3d  00                                                               |     | MIM1024          
     289200 |     289302 | Rdr |06  39  00  00                                                           |     | ACK 256/1024          
     289360 |     289580 | Rdr |0b  01  00  00                                                           |     | READ Byte(0)          
     289650 |     289902 | Tag |0c  79  03                                                               |     |           
     290126 |     290314 | Rdr |0b  03  00  00                                                           |     | READ Byte(1)          
     290384 |     290636 | Tag |0c  d1  05                                                               |     |           
     290860 |     291056 | Rdr |0b  05  00  00                                                           |     | READ Byte(2)          
     291126 |     291378 | Tag |0c  3b  0d                                                               |     |           
     291602 |     291814 | Rdr |0b  07  00  00                                                           |     | READ Byte(3)          
     291884 |     292136 | Tag |0c  cd  01                                                               |     |           
     292360 |     292580 | Rdr |0b  09  00  00                                                           |     | READ Byte(4)          
     292650 |     292902 | Tag |0c  32  0a                                                               |     |           
     295688 |     295884 | Rdr |0b  95  01  00                                                           |     | READ Byte(202)          
     296430 |     296610 | Rdr |0b  07  00  00                                                           |     | READ Byte(3)          
     297156 |     297368 | Rdr |0b  d8  07  00                                                           |     | WRITE Byte(1004)          
     297746 |     297918 | Rdr |0b  28  00  00                                                           |     | WRITE Byte(20)          
     307013 |     307129 | Rdr |07  21  00  00                                                           |     | IV 0x21         
```

iceman1001, 2019-04-12:
definitily some fails,   would you mind testing 240?

rafaelmertens, 2019-04-12:
okay.. Tested with 240 now, still no luck.. here is the output:

1st
```
pm3 --> hf legic dump o _test_entrance
[+] TYPE : MIM256 card (234 bytes)          
[+] Reading tag memory 256 b...          

          
[+] Wrote 256 bytes to _test_entrance.bin          
pm3 --> hf legic eload 1 _test_entrance
          
[+] Loaded 256 bytes from file: _test_entrance.bin  to emulator memory
          
pm3 --> hf legic sim 1
#db# Starting Legic emulator, press button to end          
#db# Stopped          
pm3 --> trace list legic
[+] Recorded Activity (TraceLen = 3970 bytes)          
[=]           
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer          
LEGIC - Reader Mode: Timings are in ticks (1us == 1.5ticks)
        Tag Mode: Timings are in sub carrier periods (1/212 kHz == 4.7us)          
          
      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation          
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------          
          0 |        132 | Rdr |07  53  00  00                                                           |     | IV 0x53          
        202 |        328 | Tag |06  1d  00                                                               |     | MIM256          
       2597 |       2713 | Rdr |07  21  00  00                                                           |     | IV 0x21          
       2783 |       2909 | Tag |06  1d  00                                                               |     | MIM256          
      11254 |      11378 | Rdr |07  23  00  00                                                           |     | IV 0x23          
      11448 |      11574 | Tag |06  1d  00                                                               |     | MIM256          
      13843 |      13975 | Rdr |07  2b  00  00                                                           |     | IV 0x2B          
      14045 |      14171 | Tag |06  1d  00                                                               |     | MIM256          
      34137 |      34261 | Rdr |07  0b  00  00                                                           |     | IV 0x0B          
      34331 |      34457 | Tag |06  1d  00                                                               |     | MIM256          
      34510 |      34620 | Rdr |06  39  00  00                                                           |     | ACK 256/1024          
      34678 |      34846 | Rdr |09  01  00  00                                                           |     | READ Byte(0)          
      34916 |      35168 | Tag |0c  79  04                                                               |     |           
      35392 |      35560 | Rdr |09  03  00  00                                                           |     | READ Byte(1)          
      35630 |      35882 | Tag |0c  7d  0d                                                               |     |           
      36106 |      36306 | Rdr |09  05  00  00                                                           |     | READ Byte(2)          
      36376 |      36628 | Tag |0c  0f  06                                                               |     |           
      36852 |      37012 | Rdr |09  07  00  00                                                           |     | READ Byte(3)          
      37082 |      37334 | Tag |0c  90  0d                                                               |     |           
      37558 |      37726 | Rdr |09  09  00  00                                                           |     | READ Byte(4)          
      37796 |      38048 | Tag |0c  1f  08                                                               |     |           
      40750 |      40910 | Rdr |09  70  01  00                                                           |     | WRITE Byte(184)          
      41456 |      41640 | Rdr |09  fa  01  00                                                           |     | WRITE Byte(253)          
      42186 |      42330 | Rdr |09  05  00  00                                                           |     | READ Byte(2)          
      42708 |      42876 | Rdr |09  cc  01  00                                                           |     | WRITE Byte(230)          
      51945 |      52069 | Rdr |07  43  00  00                                                           |     | IV 0x43          
      52139 |      52265 | Tag |06  1d  00                                                               |     | MIM256          
      52318 |      52436 | Rdr |06  39  00  00                                                           |     | ACK 256/1024          
      52494 |      52662 | Rdr |09  01  00  00                                                           |     | READ Byte(0)          
      52732 |      52984 | Tag |0c  79  04                                                               |     |           
      53208 |      53352 | Rdr |09  03  00  00                                                           |     | READ Byte(1)          
      53422 |      53674 | Tag |0c  7d  0d                                                               |     |           
      53898 |      54050 | Rdr |09  05  00  00                                                           |     | READ Byte(2)          
[... cut ...]
     417438 |     417650 | Rdr |0b  75  05  00                                                           |     | READ Byte(698)          
     418028 |     418208 | Rdr |0b  13  00  00                                                           |     | READ Byte(9)          
     426925 |     427057 | Rdr |07  2b  00  00                                                           |     | IV 0x2B          
     427127 |     427253 | Tag |06  3d  00                                                               |     | MIM1024 
```

iceman1001, 2019-04-12:
something nots good.  its a 256 card and demod says kind of out of bounds.
@drandreas  is most likely to figure it out. 

```
      31219 |      31415 | Rdr |0b  99  04  00    |     | READ Byte(588)          
      31961 |      32173 | Rdr |0b  ad  06  00    |     | READ Byte(854)          
      32719 |      32923 | Rdr |0b  73  04  00    |     | READ Byte(569)          
      33301 |      33505 | Rdr |0b  76  04  00    |     | WRITE Byte(571)     
```

rafaelmertens, 2019-04-12:
But that is only on the second one.. Or do you also see something like that on the first?
The second is loaded with 1024 bytes..
The first with 256 bytes.
I was just testing all possibilities and making sure I would be able to give all information necessary..

iceman1001, 2019-04-12:
aha, I see,  I missed that you used two different sizes.  Still there is a bunch of reader traffic but no pm3 sim responses.

rafaelmertens, 2019-04-15:
Yes. I saw that. But I can't tell why..
It responds fine in the beginning and then it stops.. And after that it seems to start from the beginning over and over again..
I wonder what I can be doing wrong, but there is not much to do wrong at this point...

rafaelmertens, 2019-04-15:
Hi @iceman1001,

I found the issue.
I re-imaged my proxmark again and it works now. 

Thanks for the help!

closed 2019-04-15 by iceman1001