RfidResearchGroup/proxmark3 #160: FPC - Variable length of usb-command packages

iceman1001, 2019-04-16:
Project 
https://github.com/RfidResearchGroup/proxmark3/projects/5

Variable length of usb-command packages sent over usb or fpc-usart

iceman1001, 2019-04-22:
For those who missed it,  this stuff is going on the branch  https://github.com/RfidResearchGroup/proxmark3/tree/experimental_varlen

The speed stability and overall stability in comms this work brings brings joy to my heart.  There is some docs where @doegox  has written down his findings regarding to speeds etc.

doegox, 2019-04-22:
The doc is also pretty much work-in-progress but you can already have a look and feel free to comment.
https://github.com/RfidResearchGroup/proxmark3/blob/experimental_varlen/doc/new_frame_format.txt

doegox, 2019-04-30:
It's now merged on master, supporting a mix of old and new formats.
Next step will be a slow migration to the new format...

iceman1001, 2019-05-01:
Great work!   time to close

closed 2019-05-01 by iceman1001