RfidResearchGroup/proxmark3 #164: [idea] Appveyor release include dlls

iceman1001, 2019-04-17:
**Is your feature request related to a problem? Please describe.**
The appveyor is a nice continuously integrated build server. It also has a release part configured currently where some files gets compressed into a 7zip archive.

**Describe the solution you'd like**
The compress 7zip archive should also copy some dlls needed for standalone runs.
- libgcc_s_dw2-1.dll
- libstdc++-6.dll
- libwinpthread-1.dll
- Qt5Core.dll
- Qt5Gui.dll
- Qt5Widgets.dll
- readline6.dll

**Additional context**
If this is possible,  the release enables easy running of a precompiled build together with a matching firmware.

merlokk, 2019-04-17:
It strange, but i dont see readline.dll. maybe it linked into pm3.exe?

iceman1001, 2019-04-17:
libgcc_s_dw2-1.dll missing.
[image]

libreadling6.dll missing.
[image]

iceman1001, 2019-04-17:
Excellent @merlokk !  
The artifact generated by appveyor is full functional running proxmark3 client.
The zip is 15MB,   small and lean.   So peopl who want to get just running the client for windows has another option now.

Maybe even make it push to the releases here on this page.

closed 2019-04-17 by iceman1001