RfidResearchGroup/proxmark3 #173: util.c:871:20: error: use of undeclared identifier - OS X

TomHarkness, 2019-04-24:
Seem to be getting the following error caused by the function on line 871 when compiling - OS X.

gcc -MT obj/util.o -MMD -MP -MF obj/util.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3 -DHAVE_GUI -DWITH_SMARTCARD -DWITH_FLASH -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/util.o util.c
util.c:871:20: error: use of undeclared identifier '_SC_NPROCESSORS_ONLN'
    return sysconf(_SC_NPROCESSORS_ONLN);

marcbressler, 2019-04-24:
Getting similar issues installing today:

```
➜  ~ brew install --HEAD rfidresearchgroup/proxmark3/proxmark3
==> Installing proxmark3 from rfidresearchgroup/proxmark3
==> Cloning https://github.com/RfidResearchGroup/proxmark3.git
Updating /Users/marcbressler/Library/Caches/Homebrew/proxmark3--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 1b246cd3 Force no-ms-bitfields to ensure proper struc packing on Windows
==> make clean
==> make all
Last 15 lines from /Users/marcbressler/Library/Logs/Homebrew/proxmark3/02.make:
gcc -MT obj/proxmark3.o -MMD -MP -MF obj/proxmark3.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3 -mno-ms-bitfields -DHAVE_GUI -DWITH_SMARTCARD -DWITH_FLASH -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/proxmark3.o proxmark3.c
mv -f obj/proxmark3.Td obj/proxmark3.d
gcc -MT obj/uart_posix.o -MMD -MP -MF obj/uart_posix.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3 -mno-ms-bitfields -DHAVE_GUI -DWITH_SMARTCARD -DWITH_FLASH -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/uart_posix.o ../uart/uart_posix.c
mv -f obj/uart_posix.Td obj/uart_posix.d
gcc -MT obj/uart_win32.o -MMD -MP -MF obj/uart_win32.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3 -mno-ms-bitfields -DHAVE_GUI -DWITH_SMARTCARD -DWITH_FLASH -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/uart_win32.o ../uart/uart_win32.c
mv -f obj/uart_win32.Td obj/uart_win32.d
gcc -MT obj/ui.o -MMD -MP -MF obj/ui.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3 -mno-ms-bitfields -DHAVE_GUI -DWITH_SMARTCARD -DWITH_FLASH -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/ui.o ui.c
mv -f obj/ui.Td obj/ui.d
gcc -MT obj/util.o -MMD -MP -MF obj/util.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3 -mno-ms-bitfields -DHAVE_GUI -DWITH_SMARTCARD -DWITH_FLASH -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/util.o util.c
util.c:871:20: error: use of undeclared identifier '_SC_NPROCESSORS_ONLN'
    return sysconf(_SC_NPROCESSORS_ONLN);
                   ^
1 error generated.
make[1]: *** [obj/util.o] Error 1
make: *** [client/all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/rfidresearchgroup/homebrew-proxmark3/issues

```

iceman1001, 2019-04-24:
could you two (@TomHarkness , @marcbressler )  test this suggested fix?

// determine number of logical CPU cores (use for multithreaded functions)
int num_CPUs(void) {
#if defined(_WIN32)
#include <sysinfoapi.h>
    SYSTEM_INFO sysinfo;
    GetSystemInfo(&sysinfo);
    return sysinfo.dwNumberOfProcessors;
#elif defined(__linux__)
#include <unistd.h>
    return sysconf(_SC_NPROCESSORS_ONLN);
#elif defined(__APPLE__)
#include "sys/sysctl.h"
    uint32 logicalcores = 0;
    size_t size = sizeof( logicalcores );
    sysctlbyname( "hw.logicalcpu", &logicalcores, &size, NULL, 0 );
    return logicalcores;
#else
    return 1;
#endif
}

TomHarkness, 2019-04-25:
That seems to have worked. Nice!

closed 2019-04-25 by iceman1001

iceman1001, 2019-04-25:
@TomHarkness  you sure it worked?   New issue regarding this fix on Mac OS 10.14.
https://github.com/RfidResearchGroup/proxmark3/issues/174