RfidResearchGroup/proxmark3 #174: Build error on macOS 10.14.4

nutinshell, 2019-04-25:
```
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/time.h:75:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types/_fd_def.h:57:1: error: 
      function definition is not allowed here
{
^
In file included from util.c:876:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/sysctl.h:83:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/ucred.h:96:11: error: 
      unknown type name 'u_long'; did you mean 'long'?
        volatile u_long         cr_ref;  /* reference count */
                 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/ucred.h:139:2: error: 
      unknown type name 'u_int'
        u_int   cr_version;             /* structure layout version */
        ^
In file included from util.c:876:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/sysctl.h:84:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:113:2: error: 
      unknown type name 'u_int'
        u_int   p_estcpu;        /* Time averaged value of p_cpticks. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:118:2: error: 
      unknown type name 'u_int'
        u_int   p_swtime;        /* Time swapped in or out. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:119:2: error: 
      unknown type name 'u_int'
        u_int   p_slptime;       /* Time since last blocked. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:133:2: error: 
      unknown type name 'u_char'; did you mean 'char'?
        u_char  p_priority;     /* Process priority. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:134:2: error: 
      unknown type name 'u_char'; did you mean 'char'?
        u_char  p_usrpri;       /* User-priority based on p_cpu and p_nice. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:139:2: error: 
      unknown type name 'u_short'; did you mean 'short'?
        u_short p_xstat;        /* Exit status for wait; also stop signal. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:140:2: error: 
      unknown type name 'u_short'; did you mean 'short'?
        u_short p_acflag;       /* Accounting flags. */
        ^
util.c:877:5: error: unknown type name 'uint32'; did you mean 'uint32_t'?
    uint32 logicalcores = 0;
    ^~~~~~
    uint32_t
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/_types/_uint32_t.h:31:22: note: 
      'uint32_t' declared here
typedef unsigned int uint32_t;
                     ^
11 errors generated.
make[1]: *** [obj/util.o] Error 1
make: *** [client/all] Error 2
```

any suggestion?

iceman1001, 2019-04-25:
revert the last commit in util.c

iceman1001, 2019-04-25:
hold on,  uint32  is wrong,  I pushed a fix for that..

iceman1001, 2019-04-25:
@nutinshell   does it work now?

nutinshell, 2019-04-25:
@iceman1001 no, still has 10 errors

```
In file included from util.c:876:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/sysctl.h:82:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/time.h:75:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/_types/_fd_def.h:57:1: error: function
      definition is not allowed here
{
^
In file included from util.c:876:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/sysctl.h:83:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/ucred.h:96:11: error: unknown type name
      'u_long'; did you mean 'long'?
        volatile u_long         cr_ref;  /* reference count */
                 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/ucred.h:139:2: error: unknown type name 'u_int'
        u_int   cr_version;             /* structure layout version */
        ^
In file included from util.c:876:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/sysctl.h:84:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:113:2: error: unknown type name 'u_int'
        u_int   p_estcpu;        /* Time averaged value of p_cpticks. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:118:2: error: unknown type name 'u_int'
        u_int   p_swtime;        /* Time swapped in or out. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:119:2: error: unknown type name 'u_int'
        u_int   p_slptime;       /* Time since last blocked. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:133:2: error: unknown type name
      'u_char'; did you mean 'char'?
        u_char  p_priority;     /* Process priority. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:134:2: error: unknown type name
      'u_char'; did you mean 'char'?
        u_char  p_usrpri;       /* User-priority based on p_cpu and p_nice. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:139:2: error: unknown type name
      'u_short'; did you mean 'short'?
        u_short p_xstat;        /* Exit status for wait; also stop signal. */
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/sys/proc.h:140:2: error: unknown type name
      'u_short'; did you mean 'short'?
        u_short p_acflag;       /* Accounting flags. */
        ^
10 errors generated.
make[1]: *** [obj/util.o] Error 1
make: *** [client/all] Error 2
```

SilentServices, 2019-04-25:
Having the exact same problem

iceman1001, 2019-04-25:
can't help with OSX,.. 
You can edit the util.c to return 1 instead.  It will only make hardnested to run on one thread. 
No other impact.

SilentServices, 2019-04-25:
To fix the u_ stuff add -Du_long="unsigned long" -Du_short="unsigned short" -Du_char="unsigned char" -Du_int="unsigned int" to the Makefile in /client. But the error: function definition is not allowed here error remains.

SilentServices, 2019-04-25:
Scratch the above. Just add #include <sys/types.h> to client/util.c and comment out the follwing in client/util.c like below. compiles fine now.
875 #elif defined(__APPLE__)
876 //#include <sys/sysctl.h>
877 //    uint32_t logicalcores = 0;
878 //    size_t size = sizeof( logicalcores );
879 //    sysctlbyname( "hw.logicalcpu", &logicalcores, &size, NULL, 0 );
880 //    return logicalcores;
881     return 1;
882 #else
883     return 1;
884 #endif

iceman1001, 2019-04-25:
if you comment out everything,  (return 1)  why add the include?

SilentServices, 2019-04-25:
I got more of the u_ errors so it looked reasonable. Anyway switched to Kali temporary as I need it working. Maybe I will look later into macos install again

iceman1001, 2019-04-25:
**make clean; make all** ... usually a good thing to clear out the failed compilation attempts.

closed 2019-04-25 by iceman1001