RfidResearchGroup/proxmark3 #177: hf mf chk - showing res 1 for known invalid keys

NinjaStyle82, 2019-05-01:
I am seeing results below even though the key 000000000000 is not valid for any sectors on this card. 
```
pm3 --> hf mf chk *1 ? d
[=] No key specified, trying default keys          
[ 0] ffffffffffff          
[ 1] 000000000000          
[ 2] a0a1a2a3a4a5          
[ 3] b0b1b2b3b4b5          
[ 4] c0c1c2c3c4c5          
[ 5] d0d1d2d3d4d5          
[ 6] aabbccddeeff          
[ 7] 1a2b3c4d5e6f          
[ 8] 123456789abc          
[ 9] 010203040506          
[10] 123456abcdef          
[11] abcdef123456          
[12] 4d3a99c351dd          
[13] 1a982c7e459a          
[14] d3f7d3f7d3f7          
[15] 714c5c886e97          
[16] 587ee5f9350f          
[17] a0478cc39091          
[18] 533cb6c723f6          
[19] 8fd0a4f256e9          
[20] 0000014b5c31          
[21] b578f38a5c61          
[22] 96a301bce267          
................................          
[+] Time in checkkeys: 16 seconds
          
[=] testing to read key B...          
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|001|  000000000000  | 1 |  000000000000  | 1 |          
|002|  000000000000  | 1 |  000000000000  | 1 |          
|003|  000000000000  | 1 |  000000000000  | 1 |          
|004|  000000000000  | 1 |  000000000000  | 1 |          
|005|  000000000000  | 1 |  000000000000  | 1 |          
|006|  000000000000  | 1 |  000000000000  | 1 |          
|007|  000000000000  | 1 |  000000000000  | 1 |          
|008|  000000000000  | 1 |  000000000000  | 1 |          
|009|  000000000000  | 1 |  000000000000  | 1 |          
|010|  000000000000  | 1 |  000000000000  | 1 |          
|011|  000000000000  | 1 |  000000000000  | 1 |          
|012|  000000000000  | 1 |  000000000000  | 1 |          
|013|  000000000000  | 1 |  000000000000  | 1 |          
|014|  000000000000  | 1 |  000000000000  | 1 |          
|015|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|---|----------------|---|----------------|---|          
```
Running a .dic file which has had the key 000000000000 removed, i see similar results with a different invalid key, as shown below on the same card.
```
pm3 --> hf mf chk *1 ? client/default_keys2.dic
[+] Loaded 648 keys from client/default_keys2.dic           
................................          
[+] Time in checkkeys: 43 seconds
          
[=] testing to read key B...          
|---|----------------|---|----------------|---|          
|sec|key A           |res|key B           |res|          
|---|----------------|---|----------------|---|          
|000|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|001|  000000000100  | 1 |  000000000100  | 1 |          
|002|  000000000100  | 1 |  000000000100  | 1 |          
|003|  000000000100  | 1 |  000000000100  | 1 |          
|004|  000000000100  | 1 |  000000000100  | 1 |          
|005|  000000000100  | 1 |  000000000100  | 1 |          
|006|  000000000100  | 1 |  000000000100  | 1 |          
|007|  000000000100  | 1 |  000000000100  | 1 |          
|008|  000000000100  | 1 |  000000000100  | 1 |          
|009|  000000000100  | 1 |  000000000100  | 1 |          
|010|  000000000100  | 1 |  000000000100  | 1 |          
|011|  000000000100  | 1 |  000000000100  | 1 |          
|012|  000000000100  | 1 |  000000000100  | 1 |          
|013|  000000000100  | 1 |  000000000100  | 1 |          
|014|  000000000100  | 1 |  000000000100  | 1 |          
|015|  ffffffffffff  | 1 |  ffffffffffff  | 1 |          
|---|----------------|---|----------------|---|   
```

doegox, 2019-05-01:
please test latest commit, thanks

iceman1001, 2019-05-01:
ha,  ok,  good remake @doegox ,  I have adapted mfkeys.lua and it works good again.

closed 2019-05-01 by iceman1001

NinjaStyle82, 2019-05-01:
> please test latest commit, thanks
It appears to be working perfectly. Thank you.