RfidResearchGroup/proxmark3 #183: "hf iclass clone" Results in Write Failure for Last Block

digitalentropy, 2019-05-02:
**Describe the bug**
"hf iclass clone" command appears to have an issue where the last block in the write range will fail to write.  If only one block is selected, it will fail to write.  The behavior appears to be consistent across multiple cards and multiple attempts to write.

```
pm3 --> hf iclass clone f iclass_tagdump-1ff14301f7ff12e0.bin b 07 l 07 k 0
Block |07|
 61b2f3229fb389c9 |
 MAC |c59a90ca|

#db# Write block [07] failed
#db# Clone incomplete

pm3 --> hf iclass clone f iclass_tagdump-1ff14301f7ff12e0.bin b 06 l 08 k 0
Block |06|
 030303030003e017 |
 MAC |bf308e37|

Block |07|
 61b2f3229fb389c9 |
 MAC |c59a90ca|

Block |08|
 2ad4c8211f996871 |
 MAC |e0c520dc|

#db# Write block [06] successful
#db# Write block [07] successful
#db# Write block [08] failed
#db# Clone incomplete

pm3 --> hf iclass clone f iclass_tagdump-1ff14301f7ff12e0.bin b 06 l 09 k 0
Block |06|
 030303030003e017 |
 MAC |bf308e37|

Block |07|
 61b2f3229fb389c9 |
 MAC |c59a90ca|

Block |08|
 2ad4c8211f996871 |
 MAC |e0c520dc|

Block |09|
 2ad4c8211f996871 |
 MAC |8c800100|

#db# Write block [06] successful
#db# Write block [07] successful
#db# Write block [08] successful
#db# Write block [09] failed
#db# Clone incomplete
```

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to write several or multiple blocks using the "hf iclass clone" command.

**Expected behavior**
The full range of data blocks should be successfully written when executing "hf iclass clone" command.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Microsoft Windows [Version 10.0.17763.437]
<details><summary>RDV40 Hardware Info</summary>
<p>

```code
pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman

 [ PROXMARK ]
  external flash:          present
  smartcard reader:        present
  USART for addon support: absent

 [ ARM ]
 bootrom: RRG/Iceman/master/2ec05efa 2019-05-02 11:25:24
      os: RRG/Iceman/master/2ec05efa 2019-05-02 11:25:57

 [ FPGA ]
 LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 249944 bytes (48%) Free: 274344 bytes (52%)          
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23
#db# Flash memory
#db#   Baudrate................24MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256kb
#db#   Unique ID...............0xd567a882a72ab725
#db# Smart card module (ISO 7816)
#db#   version.................v3.11
#db# LF Sampling config
#db#   [q] divisor.............95 (125 KHz)
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# LF T55XX config
#db#   [a] startgap............29*8 (232)
#db#   [b] writegap............17*8 (136)
#db#   [c] write_0.............15*8 (120)
#db#   [d] write_1.............47*8 (376)
#db#   [e] readgap.............15*8 (120)
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............1500ms
#db#   Bytes transferred.......884736
#db#   Transfer Speed PM3 -> Client = 589824  bytes/s
#db# Various
#db#   MF_DBGLEVEL.............1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db# Installed StandAlone Mode
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)
pm3 --> data tune

[=] Measuring antenna characteristics, please wait...

...

[+] LF antenna: 72.23 V - 125.00 kHz
[+] LF antenna: 39.80 V - 134.00 kHz
[+] LF optimal: 72.23 V - 125.00 kHz
[+] LF antenna is OK

[+] HF antenna: 49.00 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
```

</p>
</details>

iceman1001, 2019-05-03:
Thats an odd one.   Only the last write in different amounts of blocks.

iceman1001, 2019-09-08:
Try pulling latest code,  I pushed a fix for it.