RfidResearchGroup/proxmark3 #185: Compile error

sinihabe, 2019-05-05:
With latest pull (05.05.2019) on Raspbian GNU/Linux 9.9 (stretch) compile fails with following error

```
make[2]: Leaving directory '/root/proxmark3/client/tinycbor'
gcc -MT obj/proxmark3.o -MMD -MP -MF obj/proxmark3.Td  -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../zlib -I../uart -I../liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -g -O3 -mno-ms-bitfields -DHAVE_GUI -DWITH_SMARTCARD -DWITH_FLASH -DWITH_LF -DWITH_HITAG -DWITH_ISO15693 -DWITH_LEGICRF -DWITH_ISO14443b -DWITH_ISO14443a -DWITH_ICLASS -DWITH_FELICA -DWITH_HFSNIFF -DWITH_STANDALONE_LF_SAMYRUN -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/proxmark3.o proxmark3.c
gcc: error: unrecognized command line option '-mno-ms-bitfields'; did you mean '-fsigned-bitfields'?
Makefile:380: recipe for target 'obj/proxmark3.o' failed
make[1]: *** [obj/proxmark3.o] Error 1
make[1]: Leaving directory '/root/proxmark3/client'
Makefile:59: recipe for target 'client/all' failed
make: *** [client/all] Error 2

```

iceman1001, 2019-05-05:
try removing the offending -mno-ms-bitfields ?

iceman1001, 2019-05-05:
And which gcc version do you use?  output from gcc --version

doegox, 2019-05-05:
Apparently it requires gcc >=4.7.0 for `-mno-ms-bitfields`.
The option is mainly required for Mingw, so we might add a test in the Makefile to add the flag only on Mingw... Or we just require a gcc >= 4.7.0 (which is pretty old already... 2012)

iceman1001, 2019-05-05:
Don't we need gcc >= 4.9 for some hardnested parts ?

sinihabe, 2019-05-06:
Platform is Raspberry PI
gcc --version
gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516

doegox, 2019-05-06:
@iceman if you modify client/Makefile with
```
CFLAGS = $(ENV_CFLAGS) -std=c99 -D_ISOC99_SOURCE -DPRESETS $(INCLUDES_CLIENT) -Wall -g -O3
ifneq (,$(findstring MINGW,$(platform)))
    CFLAGS += -mno-ms-bitfields
endif
```
Does it work fine under Windows ?

iceman1001, 2019-05-06:
almost, I pushed it.

commit dd82c425 (2019-05-06): fix #185 - Only -mno-ms-bitfields on MINGW  (@doegox)
closed 2019-05-06 by iceman1001