RfidResearchGroup/proxmark3 #188: Proxmark client stuck at "[=] Communicating with PM3 over USB-CDC" for 10-15 minutes

mvjt, 2019-05-10:
**Describe the bug**
After updating ProxSpace and Proxmark 3 RDV4 to latest versions/fw it takes 10-15 minutes for the client to load completely. It gets stuck at "Communicating with PM3 over USB-CDC". 

After waiting 10-15 minutes, it finally loads as usual (pm3 prompt).  Unplugging and/or trying different USB port does not help...

Console output:
`pm3 ~/client$ ./proxmark3.exe com7

======. ===.   ===. ====.     ...iceman fork
==...==.====. ====.   ..=.      ...dedicated to RDV40
======..==.====.==. ====..
==..... ==..==..==.   ..=.    iceman@icesql.net
==.     ==. ... ==. ====..   https://github.com/rfidresearchgroup/proxmark3/
...     ...     ... .....  pre-release v4.0

Support iceman on patreon,   https://www.patreon.com/iceman1001/

[=] Using UART port com7
**[=] Communicating with PM3 over USB-CDC``**

**To Reproduce**
Steps to reproduce the behavior:
1.  Start ProxSpace v3.0 x64 (updated today to latest Git)
2.  Start Proxmark client (from this repo)
3. Gets stuck

**Expected behavior**
I´m expecting the client to load immediately

**Desktop (please complete the following information):**
 - OS: Windows 10 x64

 - hw version
`[ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman

 [ PROXMARK RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
 bootrom: RRG/Iceman// 2019-05-10 09:11:16
      os: RRG/Iceman// 2019-05-10 09:11:44

 [ FPGA ]
 LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 249999 bytes (48%) Free: 274289 bytes (52%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory
`
 - hw status

`#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23
#db# Flash memory
#db#   Baudrate................24MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256kb
#db#   Unique ID...............0xd567a882a7a67f26
#db# Smart card module (ISO 7816)
#db#   version.................v3.11
#db# LF Sampling config
#db#   [q] divisor.............95 (125 kHz)
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# LF T55XX config
#db#   [a] startgap............29*8 (232)
#db#   [b] writegap............17*8 (136)
#db#   [c] write_0.............15*8 (120)
#db#   [d] write_1.............47*8 (376)
#db#   [e] readgap.............15*8 (120)
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......273920
#db#   Transfer Speed PM3 -> Client = 547840  bytes/s
#db# Various
#db#   MF_DBGLEVEL.............1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db# Installed StandAlone Mode
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)`

 - data tune
`[=] Measuring antenna characteristics, please wait...

...

[+] LF antenna: 29.60 V - 125.00 kHz
[+] LF antenna: 47.16 V - 134.00 kHz
[+] LF optimal: 58.92 V - 139.53 kHz
[+] LF antenna is OK

[+] HF antenna: 17.63 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
  `

merlokk, 2019-05-10:
the same. maybe on msys2 something gone wrong

libin-ka, 2019-05-11:
The same problem

iceman1001, 2019-05-12:
Somewhere is a deadlock between main thread and communications thread.

McEloff, 2019-05-13:
I solved this bug locally to comment part of client code in 'comms.c' (method TestProxmark):
...
/*
#if defined(_WIN32)
                pthread_mutex_lock(&spMutex);
#endif
                int res = uart_reconfigure_timeouts(sp, UART_USB_CLIENT_RX_TIMEOUT_MS);
#if defined(_WIN32)
                pthread_mutex_unlock(&spMutex);
#endif
                if (res != PM3_SUCCESS) {
                    return res;
                }
*/
But win-client still works as 'slide show'.  On receive '#db#' text from ARM is printing slowly char-by-char. For example, type 'hf mf dbg 1', answer '#db# debug level 1' is printing char-by-char. For more effect run 'hf mf sim' with debug level 1, work some time, press button. Answer 'Emulator stopped. Tracing: ...' is printing very slowly char-by-char too.

iceman1001, 2019-05-16:
@mvjt  @McEloff @libin-ka   would you all,  pull latest,  make clean,  make  -j4 ,   and test if there is some delays still

McEloff, 2019-05-16:
Fixed

closed 2019-05-16 by iceman1001

merlokk, 2019-05-16:
Works for me

edward-p, 2019-08-29:
Similar thing happened on my Arch Linux recently.

I have a __modified Proxmark3 Easy with 512k MCU__.

I build the project using this [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=proxmark3-iceman-git) 

Run `proxmark3 /dev/pm3-0` and client stuck at "[=] Communicating with PM3 over USB-CDC" and get high CPU consumption. (Not sure if it will stuck forever)

[image]

[image]

However, I can get the output from the client when switching into stand-alone mode by pressing the button.

[image]

__To Reproduce__
Steps to reproduce the behavior:

1. Connect the USB

2. Start Proxmark client

3. Gets stuck

__Expected behavior__
I´m expecting the client to load immediately.

__Desktop (please complete the following information):__

OS: Arch Linux

hw version (Don't know how to check. Because the client gets stuck, I can do nothing. But I do flashed the latest firmware build from `6846.86790df6` with `make PLATFORM=PM3OTHER` )

iceman1001, 2019-08-29:
Ha, I didnt notice you are the same person as the package maintainer.
Great,  now we have Archlinux ,  pentoo , gentoo ppl here.  ( @ZeroChaos- ) 
So,  I notice in your package that you install fpga_compress.   Its only used when compiling. No need to install it.

There are more issues here opened regarding "make install" and paths.  I think your package will benefit if you read up on them aswell and give us your input.

iceman1001, 2019-08-29:
btw,  did you flash bootrom individually first,  reset,  then fullimage?

edward-p, 2019-08-29:
> btw, did you flash bootrom individually first, reset, then fullimage?

Yes, I do flash bootrom individually then fullimage.

```
edward•~» /usr/share/proxmark3/flash-bootrom.sh 
[=] Waiting for Proxmark3 to appear...
Session is logged into /home/edward/.proxmark3/log_20190829.txt
[+] Waiting for Proxmark3 to appear on /dev/pm3-0           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/pm3-0           
........... Found           
[=] Available memory on this board: 512K bytes
          
[=] Permitted flash range: 0x00100000-0x00180000          
[+] Loading ELF file bootrom/obj/bootrom.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00100200 (0x00000e48->0x00000e48) [R X] @0x298          
          
          
[+] Flashing... 
          
[+] Writing segments for file: bootrom/obj/bootrom.elf          
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]          
. OK           
[+]  0x00100200..0x00101047 [0xe48 / 8 blocks]          
........ OK           

          
[+] All done.           

Have a nice day!          
edward•~» /usr/share/proxmark3/flash-fullimage.sh 
[=] Waiting for Proxmark3 to appear...
Session is logged into /home/edward/.proxmark3/log_20190829.txt
[+] Waiting for Proxmark3 to appear on /dev/pm3-0           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/pm3-0           
.......... Found           
[=] Available memory on this board: 512K bytes
          
[=] Permitted flash range: 0x00102000-0x00180000          
[+] Loading ELF file armsrc/obj/fullimage.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00102000 P 0x00102000 (0x00036d70->0x00036d70) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00138d70 (0x000014c8->0x000014c8) [RW ] @0x36e04          
[=] Note: Extending previous segment from 0x36d70 to 0x38238 bytes          
          
          
[+] Flashing... 
          
[+] Writing segments for file: armsrc/obj/fullimage.elf          
[+]  0x00102000..0x0013a237 [0x38238 / 450 blocks]          
.................................................................................................................................................................................................................................................................................................................................................................................................................................................................. OK           

          
[+] All done.           

Have a nice day!
```

edward-p, 2019-08-29:
> fpga_compress
Thanks!
It will be removed in the next build.

doegox, 2019-08-29:
if you're mainainter @edward-p I advise to follow the discussions on https://github.com/RfidResearchGroup/proxmark3/issues/325 as we're trying to get closer to a "make install"

doegox, 2019-08-29:
ha you're already on it :D

edward-p, 2019-08-30:
> btw, did you flash bootrom individually first, reset, then fullimage?

There must be something wrong with the Qt5

I tried to compiled without qt5-base, then:
```
[=] Session log /home/edward/.proxmark3/log_20190830.txt

          
██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon,   https://www.patreon.com/iceman1001/          

          
[=] Using UART port /dev/pm3-0           
[=] Communicating with PM3 over USB-CDC           

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with GCC 9.1.0 OS:Linux ARCH:x86_64          

 [ PROXMARK RDV4 ]          
  external flash:                  absent           
  smartcard reader:                absent           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/4433f898 2019-08-30 20:43:50
       os: RRG/Iceman/master/4433f898 2019-08-30 20:44:02
  compiled with GCC 9.2.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 238136 bytes (45%) Free: 286152 bytes (55%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
[usb] pm3 --> 
```

Currently, Arch Linux is using [qt5-base 5.13.0](https://www.archlinux.org/packages/extra/x86_64/qt5-base/). I can't tell if it is an upstream issue or the package maintainer's mistake.

iceman1001, 2019-08-30:
Are you running on a RDV4 or an other model?
```
 [ PROXMARK RDV4 ]          
  external flash:                  absent           
  smartcard reader:                absent           
```

iceman1001, 2019-08-30:
Maybe time for me to test on some other devices.

SciLor, 2019-12-04:
> ack, you run on easy w 512kb.
> hm, still strange.

I can reproduce this issue on a easy 512k with debian (proxmox 6) too.

iceman1001, 2019-12-04:
@SciLor  Reproduce?  
- did you compile with Makefile.platform  PM3OTHER ?
- flash bootrom
- reset
- flash fullimage
- start client

SciLor, 2019-12-04:
> @SciLor Reproduce?
> 
> * did you compile with Makefile.platform  PM3OTHER ?
> * flash bootrom
> * reset
> * flash fullimage
> * start client

Yes, thats what I did.

iceman1001, 2019-12-04:
More information needed:
- GCC version

the output from 
 - the compilation
 - lsusb /  tail dmesg  after flashing fullimage
 - running the client

And you killed the modem manager?

SciLor, 2019-12-04:
```
gcc --version
gcc (Debian 8.3.0-6) 8.3.0
[...]
```
compile:
```
===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[-] CC proxmark3.c
[-] CC uart/uart_posix.c
[-] CC uart/uart_win32.c
[-] CC ui.c
[-] CC ../common/commonutil.c
[-] CC util.c
[-] CC ../common/util_posix.c
[-] CC scandir.c
[-] CC ../common/crc16.c
[-] CC comms.c
[-] CC ../common/crapto1/crapto1.c
[-] CC ../common/crapto1/crypto1.c
[-] CC mifare/mifaredefault.c
[-] CC mifare/mfkey.c
[-] CC tea.c
[-] CC fido/additional_ca.c
[-] CC fido/cose.c
[-] CC fido/cbortools.c
[-] CC fido/fidocore.c
[-] CC crypto/asn1dump.c
[-] CC crypto/libpcrypto.c
[-] CC crypto/asn1utils.c
[-] CC cliparser/argtable3.c
[-] CC cliparser/cliparser.c
[-] CC loclass/cipher.c
[-] CC loclass/cipherutils.c
[-] CC loclass/ikeys.c
[-] CC loclass/elite_crack.c
[-] CC fileutils.c
[-] CC whereami.c
[-] CC mifare/mifarehost.c
[-] CC ../common/parity.c
[-] CC ../common/crc.c
[-] CC ../common/crc64.c
[-] CC ../common/legic_prng.c
[-] CC ../common/iso15693tools.c
[-] CC prng.c
[-] CC graph.c
[-] CC cmddata.c
[-] CC ../common/lfdemod.c
[-] CC emv/crypto_polarssl.c
[-] CC emv/crypto.c
[-] CC emv/emv_pk.c
[-] CC emv/emv_pki.c
[-] CC emv/emv_pki_priv.c
[-] CC emv/test/cryptotest.c
[-] CC emv/apduinfo.c
[-] CC emv/dump.c
[-] CC emv/tlv.c
[-] CC emv/emv_tags.c
[-] CC emv/dol.c
[-] CC emv/emvjson.c
[-] CC emv/emvcore.c
[-] CC emv/test/crypto_test.c
[-] CC emv/test/sda_test.c
[... 350 lines cut ...]
./proxmark3 /dev/ttyACM0 

```
          
██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon - https://www.patreon.com/iceman1001/          
                 on paypal - https://www.paypal.me/iceman1001          

          
[=] Session log /home/daniel/.proxmark3/log_20191204.txt
[=] Using UART port /dev/ttyACM0           
[=] Communicating with PM3 over USB-CDC           

```

modem-manager is removed from my system

PS:
Installed with (with PM3OTHER in Makefile.platform) https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md

SciLor, 2019-12-04:
```
gcc --version
gcc (Debian 8.3.0-6) 8.3.0
[...]
```
compile:
```
===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
[*] MAKE client/all
[-] CC proxmark3.c
[-] CC uart/uart_posix.c
[-] CC uart/uart_win32.c
[-] CC ui.c
[-] CC ../common/commonutil.c
[-] CC util.c
[-] CC ../common/util_posix.c
[-] CC scandir.c
[-] CC ../common/crc16.c
[-] CC comms.c
[-] CC ../common/crapto1/crapto1.c
[-] CC ../common/crapto1/crypto1.c
[-] CC mifare/mifaredefault.c
[-] CC mifare/mfkey.c
[-] CC tea.c
[-] CC fido/additional_ca.c
[-] CC fido/cose.c
[-] CC fido/cbortools.c
[-] CC fido/fidocore.c
[-] CC crypto/asn1dump.c
[-] CC crypto/libpcrypto.c
[-] CC crypto/asn1utils.c
[-] CC cliparser/argtable3.c
[-] CC cliparser/cliparser.c
[-] CC loclass/cipher.c
[-] CC loclass/cipherutils.c
[-] CC loclass/ikeys.c
[-] CC loclass/elite_crack.c
[-] CC fileutils.c
[-] CC whereami.c
[-] CC mifare/mifarehost.c
[-] CC ../common/parity.c
[-] CC ../common/crc.c
[-] CC ../common/crc64.c
[-] CC ../common/legic_prng.c
[-] CC ../common/iso15693tools.c
[-] CC prng.c
[-] CC graph.c
[-] CC cmddata.c
[-] CC ../common/lfdemod.c
[-] CC emv/crypto_polarssl.c
[-] CC emv/crypto.c
[-] CC emv/emv_pk.c
[-] CC emv/emv_pki.c
[-] CC emv/emv_pki_priv.c
[-] CC emv/test/cryptotest.c
[-] CC emv/apduinfo.c
[-] CC emv/dump.c
[-] CC emv/tlv.c
[-] CC emv/emv_tags.c
[-] CC emv/dol.c
[-] CC emv/emvjson.c
[-] CC emv/emvcore.c
[-] CC emv/test/crypto_test.c
[-] CC emv/test/sda_test.c
[... 350 lines cut ...]
./proxmark3 /dev/ttyACM0 

```
          
██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon - https://www.patreon.com/iceman1001/          
                 on paypal - https://www.paypal.me/iceman1001          

          
[=] Session log /home/maria/.proxmark3/log_20191204.txt
[=] Using UART port /dev/ttyACM0           
[=] Communicating with PM3 over USB-CDC           

```

modem-manager is removed from my system

PS:
Installed with (with PM3OTHER in Makefile.platform) https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md

iceman1001, 2019-12-04:
Strange all if good,  compilation,  lsusb,  dmesg.
modem manager removed..

Suggestions:
- You got a picture of your device?
- tried different usb cable?
- tried different usb port?

SciLor, 2019-12-05:
No change :(

Changed to a different USB-Cable + Port.
https://ibb.co/pnNW8fr

```
 ./pm3-flash-bootrom 
[=] Waiting for Proxmark3 to appear...
[=] Session log /home/maria/.proxmark3/log_20191205.txt
[+] About to use the following file:          
[+]     /ramdisk/maria/proxmark3/client/../bootrom/obj/bootrom.elf          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
. Found           
[!!] ====================== OBS ! ===========================================           
[!!] Note: Your bootloader does not understand the new CMD_BL_VERSION command            
[!!] It is recommended that you first update your bootloader alone,            
[!!] reboot the Proxmark3 then only update the main firmware 
          
[=] Available memory on this board: UNKNOWN 
          
[!!] ====================== OBS ! ======================================           
[!!] Note: Your bootloader does not understand the new CHIP_INFO command            
[!!] It is recommended that you first update your bootloader alone,            
[!!] reboot the Proxmark3 then only update the main firmware 
          
[=] Permitted flash range: 0x00100000-0x00140000          
[+] Loading ELF file /ramdisk/maria/proxmark3/client/../bootrom/obj/bootrom.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00100200 (0x00000dd0->0x00000dd0) [R X] @0x298          
          
          
[+] Flashing... 
          
[+] Writing segments for file: /ramdisk/maria/proxmark3/client/../bootrom/obj/bootrom.elf          
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]          
. OK           
[+]  0x00100200..0x00100fcf [0xdd0 / 7 blocks]          
....... OK           

          
[+] All done.           

Have a nice day! 
```
```
./pm3-flash-fullimage 
[=] Waiting for Proxmark3 to appear...
[=] Session log /home/maria/.proxmark3/log_20191205.txt
[+] About to use the following file:          
[+]     /ramdisk/maria/proxmark3/client/../armsrc/obj/fullimage.elf          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
. Found           
[=] Available memory on this board: 512K bytes
          
[=] Permitted flash range: 0x00102000-0x00180000          
[+] Loading ELF file /ramdisk/maria/proxmark3/client/../armsrc/obj/fullimage.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00102000 P 0x00102000 (0x00037ed0->0x00037ed0) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00139ed0 (0x00001454->0x00001454) [RW ] @0x37f64          
[=] Note: Extending previous segment from 0x37ed0 to 0x39324 bytes          
          
          
[+] Flashing... 
          
[+] Writing segments for file: /ramdisk/maria/proxmark3/client/../armsrc/obj/fullimage.elf          
[+]  0x00102000..0x0013b323 [0x39324 / 458 blocks]          
.......................................................................................................................................................................................................................................................................................................................................................................................................................................................................... OK           

          
[+] All done.           

Have a nice day!   
```
PS: the old "iceman" firmware from the other repo works fine

iceman1001, 2019-12-05:
And you are on Debian OS...
Can you do this and post the outputs,  [accessrights](https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md#get-permissions-to-use-devttyacm0)

SciLor, 2019-12-05:
```
===================================================================
Platform name:     Proxmark3 Generic target
PLATFORM:          PM3OTHER
Platform extras:   No extra selected
Included options:  LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA NFCBARCODE HFSNIFF
Standalone mode:   LF_SAMYRUN
===================================================================
sudo adduser maria dialout
[sudo] Password for maria: 
User »maria« is already member of »dialout«.
```
and "ok"
`crw-rw-rw- 1 root dialout 166, 0 Dez  5 20:31 /dev/ttyACM0`

iceman1001, 2019-12-05:
So we tried all software related fixes.  Can't do more now.
You better stay on the old '"iceman" version.  Does offical repo work for you?

SciLor, 2019-12-05:
Yeah, that works fine.

iceman1001, 2019-12-05:
sounds like the firmware compiled with PM3OTHER on rrg/iceman doesn't get flashed ...

I have seen ppl's devices dissappear, when flashing bootrom,   that was due to one of these thre causes.
1. didn't compile with PM3OTHER.  solution was do the makefile.platform thingy...
2. ran homebrew on OSX and somehow those fw didn't get PM3OTHER. Solution was to clone repo and compile yourself.
3. modem manager 

You seem to have no of this.

doegox, 2019-12-05:
flashing seems to happen smoothly
just to be sure:

* when you run `./pm3`, it's stalled a few mins at "Communicating with PM3" then it runs fine? (as per the original description of this ticket)
* do you have the possibility to try under another Debian OS, maybe on another machine? Or booting on a live ISO like Kali? I'm trying to understand if it's sth specific to this proxmox distro or to the device you have

doegox, 2019-12-05:
I see proxmox is a virtualization platform, did you try directly on the host or on a guest VM?
Maybe it adds some abstraction layer that changes USB timings and the client gets lost...

SciLor, 2019-12-06:
If I am using the older iceman client connects fine.
If I am using the older iceman firmware with the rrg client:
```
[=] Session log /home/maria/.proxmark3/log_20191206.txt
[=] Using UART port /dev/ttyACM0           
unknown command:: 0x61334d50[!!] ERROR: cannot communicate with the Proxmark
```
I am using my proxmox host for flashing and connecting 

Using kali:
Same problem. If I am trying to flash over kali, there are problems flashing with finding the device because spice disconnecting the device, when it resets.

iceman1001, 2019-12-06:
The golden rule _never mix client and firmware_  is still valid I notice. 
Always use the same client / firmware from the same compilation.

For kali,  kill modem mananger...

CthUlhUzzz, 2020-02-25:
Problem not solved. I cloned the latest b87b031 commit and compile it on Debian 10. Client stuck on load (even in standalone mode) and one of the threads load on 100%. But on Kali linux all works fine
[image]