RfidResearchGroup/proxmark3 #189: proxmark3 client crash on macOS with pm3 RDV4 after starting

uhei, 2019-05-10:
After starting proxmark3 client it crashes with a buffer overflow:
```
tyrell:proxmark3 uhei$ PATH=/usr/bin /usr/bin/lldb client/proxmark3 /dev/tty.usbmodemiceman1 
(lldb) target create "client/proxmark3"
Current executable set to 'client/proxmark3' (x86_64).
(lldb) settings set -- target.run-args  "/dev/tty.usbmodemiceman1"
(lldb) run
Process 22554 launched: '/Users/uhei/Downloads/prox-rdv4/proxmark3/client/proxmark3' (x86_64)

          
██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon,   https://www.patreon.com/iceman1001/          

          
[=] Using UART port /dev/tty.usbmodemiceman1           
2019-05-10 12:40:27.654094+0200 proxmark3[22554:158777] detected buffer overflow
Process 22554 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff68e4a2c6 libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff68e4a2c6 <+10>: jae    0x7fff68e4a2d0            ; <+20>
    0x7fff68e4a2c8 <+12>: movq   %rax, %rdi
    0x7fff68e4a2cb <+15>: jmp    0x7fff68e44457            ; cerror_nocancel
    0x7fff68e4a2d0 <+20>: retq   
Target 0: (proxmark3) stopped.
(lldb) thread backtrace 
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff68e4a2c6 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff68effbf1 libsystem_pthread.dylib`pthread_kill + 284
    frame #2: 0x00007fff68db46a6 libsystem_c.dylib`abort + 127
    frame #3: 0x00007fff68db4819 libsystem_c.dylib`abort_report_np + 177
    frame #4: 0x00007fff68dd8cb1 libsystem_c.dylib`__chk_fail + 48
    frame #5: 0x00007fff68dd8c81 libsystem_c.dylib`__chk_fail_overflow + 16
    frame #6: 0x00007fff68dd9174 libsystem_c.dylib`__memcpy_chk + 18
    frame #7: 0x000000010000ae4a proxmark3`TestProxmark at comms.c:616:13 [opt]
    frame #8: 0x0000000100001cb2 proxmark3`main(argc=<unavailable>, argv=<unavailable>) at proxmark3.c:494:33 [opt]
    frame #9: 0x00007fff68d0f3d5 libdyld.dylib`start + 1
```

Issue seems to be here:
`memcpy(&pm3_capabilities, resp.data.asBytes, resp.length);` in `client/comms.c` at line 616

Running it on macOS 10.14.4 with a proxmark3 RDV 4.0 via USB

uhei, 2019-05-10:
`git bisect` shows 40480a49d8d3674989e2cfddb8264f6d97354c54 as first bad commit

closed 2019-05-10 by doegox

iceman1001, 2019-05-12:
Awesome fix!   Great meeting you @uhei  at HITB,    it will be very helpful having a another OSX dude to verify stuff for us since either me or @doegox  runs OSX anywhere.