RfidResearchGroup/proxmark3 #192: Unable to build os image

sh7d, 2019-05-11:
So, i tried to compile 9322ce626e36fa433bda0cd4a56be14c31aa426a, and im unable to compile os images, cause of this little suprise :)

```
arm-none-eabi-gcc -c -I../include -I../common -I. -Wall -Werror -pedantic -Wunused -std=c99 -DWITH_SMARTCARD -DWITH_FLASH -DWITH_LF -DWITH_HITAG -DWITH_ISO15693 -DWITH_LEGICRF -DWITH_ISO14443b -DWITH_ISO14443a -DWITH_ICLASS -DWITH_FELICA -DWITH_HFSNIFF -DWITH_STANDALONE_LF_SAMYRUN -DON_DEVICE -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -I. -fno-stack-protector -fno-pie -Os -mthumb -mthumb-interwork -o obj/appmain.o appmain.c 
appmain.c: In function 'PacketReceived':
appmain.c:739:64: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  739 |             setT55xxConfig(packet->oldarg[0], (t55xx_config *) packet->data.asBytes);
      |                                                                ^~~~~~
appmain.c:742:49: error: taking address of packed member of 'struct <anonymous>' may result in an unaligned pointer value [-Werror=address-of-packed-member]
  742 |             setSamplingConfig((sample_config *) packet->data.asBytes);
      |                                                 ^~~~~~
cc1: all warnings being treated as errors
make[1]: *** [../common/Makefile.common:77: obj/appmain.o] 
```

sh7d, 2019-05-11:
Well, it's looks like it's because of never version of gcc.
For now i just have to add additional cflag :)

```
diff --git a/common/Makefile.common b/common/Makefile.common
index bbdf1d0e..d2353ead 100644
--- a/common/Makefile.common
+++ b/common/Makefile.common
@@ -64,7 +64,7 @@ VPATH = . ../common ../common/crapto1 ../common/polarssl ../fpga ../zlib ../arms
 
 INCLUDES = ../include/proxmark3.h ../include/at91sam7s512.h ../include/config_gpio.h ../include/pm3_cmd.h $(APP_INCLUDES)
 
-CFLAGS =  -c $(INCLUDE) -Wall -Werror -pedantic -Wunused -std=c99 $(APP_CFLAGS) -Os
+CFLAGS =  -c $(INCLUDE) -Wall -Werror -Wno-address-of-packed-member -pedantic -Wunused -std=c99 $(APP_CFLAGS) -Os
 LDFLAGS = -nostartfiles -nodefaultlibs -Wl,-gc-sections -n
 LIBS = -lgcc
```

doegox, 2019-05-11:
Interesting, which gcc version are you using ?

doegox, 2019-05-11:
ha sorry it's in your title

doegox, 2019-05-13:
The warning is kind of legit, as a packed field could be at a boundary not reachable by a pointer.
But here "we know what we're doing" so it's more a false positive.
Still it would be cleaner to remove locally the warnings rather than disabling entirely this class of warnings...
But this packet->data.asBytes is used at many locations, I've to think what's best approach... and get a gcc 9 to test...

doegox, 2019-05-13:
I just realized the solution was easy. Since we dissociated the structs going on the wire from the internal structs, we don't need to pack the internal structs anymore.
Fixed in https://github.com/RfidResearchGroup/proxmark3/commit/d05cd5ff2b25146e3728a4f333724b3b3e1107b6

closed 2019-05-13 by doegox