RfidResearchGroup/proxmark3 #193: legic.lua // sf - safe File : Wrong File-Size

thebasti314, 2019-05-13:
I use the legic.lua script to dump and modify some third party Cash Values. 
When i try to write the binary back to a file (using the sf command), the file on the disk has a size of 3065k instead of 1024k

Steps to reproduce the behavior:
pm3 --> script run legic
 > lf xxxxxxxx.bin

1024 bytes from xxxxxxxx.bin loaded
create virtual tag from 1024 bytes

Tag-Type: SAM
3 Segment(s) found
1024 bytes for Tag processed

 > e3p
autoSelect . . 3rd Party Cash-Segment detected 
autoselected Index: 01

edit Balance? [y/n] ?y
enter new Balance≤
without comma and without currency-sign! (0-65535) (default: 100 ) 
 > 1000

edit UserID-Mapping? [y/n] ?n

edit Stamp? [y/n] ?n

 > sf
enter filename: (default: legic.temp ) 
 > test_new.bin

wrote 1024 bytes to test_new.bin

But the file on the this shows this:
3279252   4 -rw-r--r--  1 root root   3065 Mai 13 16:51 test_new.bin

pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          
 [ CLIENT ]          
  client: RRG/Iceman          

 [ PROXMARK RDV4 ]          
  external flash:                  absent           
  smartcard reader:                absent           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
 bootrom: RRG/Iceman/master/2f2796be 2019-05-08 14:28:12
      os: RRG/Iceman/master/2f2796be 2019-05-08 14:28:20

 [ FPGA ]
 LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S256 Rev D          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 242607 bytes (93%) Free: 19537 bytes ( 7%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2019-05-13:
Cool!
I haven't use this script in years since I don't have access to a legic reader.  :)
The filesize looks like it gets saved as text instead of binary.

thebasti314, 2019-05-13:
Hmm. I'm not sure about it. It seems that there are some "trash bytes" in the end of that binary

thebasti314, 2019-05-13:
Update: I can confirm: the output is saved as text. :)

iceman1001, 2019-05-14:
@thebasti314  I pushed a fix  where it will also save as BIN.    Would you mind testing it out?

thebasti314, 2019-05-14:
@iceman1001 Unfortunately it does not work. Now there are two files generated.
The first one is plain text with a filesize of 3065k. The second file is also plaintext with a filesize of 2041. It's the same content like in the first file, but without spaces.

iceman1001, 2019-05-14:
aha,  
so the bytes wasn't in byte format but as strings.  Annoying but still no problem.

thebasti314, 2019-05-14:
Right. The bytes are interpreted as strings.

iceman1001, 2019-05-14:
Test again :)

thebasti314, 2019-05-14:
The script now crashes and no second file is generated.
`
Legic command? ('h' for help - 'q' for quit) (default: h ) 
 > sf
enter filename: (default: legic.temp ) 
 > test
/proxmark3/client/scripts/legic.lua:649: attempt to concatenate local 'ext' (a nil value)
`

iceman1001, 2019-05-14:
ok,  can you send me a copy of your file.  Easier to test.

iceman1001, 2019-05-14:
its was also because you didn't have an file extension in your file name :)

thebasti314, 2019-05-14:
@iceman1001 I tryed also with extension: 
` > sf`
`enter filename: (default: legic.temp ) `
` > new.bin`
`new.bin	new1.bin`
`/pentest/wireless/proxmark3/client/lualibs/utils.lua:291: attempt to call method 'gmatch' (a nil value)`
          
I will send you an email i a few minutes...

iceman1001, 2019-05-14:
done.

closed 2019-05-14 by iceman1001