RfidResearchGroup/proxmark3 #194: Graph window loops to end if you scroll left past 0

mcd1992, 2019-05-13:
(Wow thanks github for clearing my proper issue text when you enabled this stupid new 'related issues' feature... Too lazy to re-type)

Commit https://github.com/RfidResearchGroup/proxmark3/commit/97676d3210dcb50c05ad6b870ebbe33fd8a37c84 has introduced a regression where scrolling left in the `data plot` window will cause it to loop back to the end of the graph instead of stopping at 0.

I'll look into this more later since this would be a good time to implement PgUp/Down Home/End support into the Qt graph as well..

mcd1992, 2019-05-13:
@doegox not sure if the issue jumps out at you immediately. I'm guessing its something to do with the change from `int` (signed) to `size_t` (unsigned?)

iceman1001, 2019-05-13:
hm,  jumps to end ,  but going right,  it stops,  it doesn't wrap back.

Not sure if it was intended but the wrapping around the plot like that is kind of useful.. if it had a  consistent behavior

mcd1992, 2019-05-13:
Yea its weird. I personally think it should have the hard 'edges' where it doesn't wrap around. Then we can use Home/End to jump to the beginning or end of the graph. Also wanted to add PgUp/Down as a extra binding to shift+left/right but it looks like the page left/right was never implemented fully; the `PageWidth` var never seems to get set in proxguiqt.cpp so shift+arrow does nothing.

iceman1001, 2019-05-13:
I be looking forward to your enhancements :)

closed 2019-05-13 by doegox

doegox, 2019-05-13:
I fixed the bug but feel free to propose more bindings, implement page shift, maybe why not mouse bindings as well :)

commit e4f00c52 (2019-05-13): Fix #194 Graph window loops to end if you scroll left past 0