RfidResearchGroup/proxmark3 #198: lf nedap sim segfaults

phi-el, 2019-05-20:
**Describe the bug**
'lf nedap sim' fails to simulate a given nedap ID.

**To Reproduce**
1. Start current proxmark client
2. Enter 'lf nedap sim 12345'
3. See error after some seconds

**Expected behavior**
Nedap card should be simulated

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Kali Linux Rolling 2019.1
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          

 [ PROXMARK RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
 bootrom: RRG/Iceman/master/b55e8126 2019-05-20 17:17:02
      os: RRG/Iceman/master/b55e8126 2019-05-20 17:17:15

 [ FPGA ]
 LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 248423 bytes (47%) Free: 275865 bytes (53%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory 
```
 - hw status
```
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image           
#db#   mode.................... LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32          
#db# Flash memory           
#db#   Baudrate................24MHz          
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256kb           
#db#   Unique ID...............0xd567a882a7688925          
#db# Smart card module (ISO 7816)           
#db#   version.................v3.10           
#db# LF Sampling config           
#db#   [q] divisor.............95 (125 kHz)          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............15*8 (120)          
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......309760          
#db#   Transfer Speed PM3 -> Client = 619520  bytes/s          
#db# Various           
#db#   MF_DBGLEVEL.............1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mode           
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)          
#db# Flash memory dictionary loaded           
#db#   Mifare................638 keys          
#db#   T55x7.................105 key
```

          

 - data tune
```
[=] Measuring antenna characteristics, please wait...
          
...
[+] LF antenna: 70.95 V - 125.00 kHz          
[+] LF antenna: 37.81 V - 134.00 kHz          
[+] LF optimal: 70.95 V - 125.00 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 42.63 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
```

**Additional context**
Freshly downloaded the repository, compiled everything, flashed proxmark and tried again. No success

iceman1001, 2019-05-20:
Verified and pushed a fix.  Pull latest and try again please :)

phi-el, 2019-05-20:
Client works, going to test it in ~ 10 hours against a live system, then i would close it :) Thanks :)

iceman1001, 2019-05-20:
don't have too high hopes,  the lf nedap sim can't convert to proper encoded signal.

phi-el, 2019-05-21:
Yep, isn't working against a live system. Original Bug was fixed :)

closed 2019-05-21 by phi-el