RfidResearchGroup/proxmark3 #199: 'lf em 410x_sim' uses wrong samples data

McEloff, 2019-05-20:
**Describe the bug**
Method CmdLFSim() in 'cmdlf.c' sends to ARM incorrect samples to LF-simulation. Variable GraphBuffer is an array of integer (32 bits). Method SimulateTagLowFrequency() in ARM module 'lfops.c' looks at buffer as array of uint8_t.
Method CmdLFSim() in official proxmark repo converts integer to bytes before sending USB-command:

		for (j = 0; j < USB_CMD_DATA_SIZE; j++) {
			c.d.asBytes[j] = GraphBuffer[i+j];
		}

**To Reproduce**
Steps to reproduce the behavior:
1. Starts command 'lf em 410x_sim 1122334455 32'
2. Readers not detects simulated tag as LF-tag

**Expected behavior**
'lf em 410x_sim' simulates corrected LF-tag

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 [ CLIENT ]
  client: RRG/Iceman

 [ PROXMARK RDV4 ]
  external flash:                  absent 
  smartcard reader:                absent 

 [ PROXMARK RDV4 Extras ]
  FPC USART for BT add-on support: absent 

 [ ARM ]
 bootrom: RRG/Iceman/master/fe6cbfc8-dirty-unclean 2019-05-21 01:12:35
      os: RRG/Iceman/master/fe6cbfc8-dirty-unclean 2019-05-21 01:13:04

 [ FPGA ]
 LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23

 [ Hardware ] 
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 239687 bytes (46%) Free: 284601 bytes (54%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2019-05-22:
That change had some consequence changes to do.
-with regards to return codes.
-with regards to OLD -> NG format

Would you mind testing again?

I also fixed the demod so it can demodulate the simulation data.
lf em 410x_sim 1122334455 32
press button
lf em 410x_demod

should now print the uid again.

iceman1001, 2019-05-22:
The sim work bad for me. Back to the drawing board

McEloff, 2019-05-22:
I have two "old" proxmarks - PMV3 and Easy.
First device starts 'lf em 410x_sim 1122334455 32' simulation.
On second device types 'lf search'. Second device prints "No known 125/134 kHz tags found!". Tried some times under different antenna positions. No results.

McEloff, 2019-05-22:
If apply following patch, LED_D would be flashes during transmitting LF-modulation on air. With this patch I detects that one device is responding to second device.

---
 armsrc/lfops.c | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/armsrc/lfops.c b/armsrc/lfops.c
index 18aeadb0..f3516be4 100644
--- a/armsrc/lfops.c
+++ b/armsrc/lfops.c
@@ -618,6 +618,8 @@ void SimulateTagLowFrequencyEx(int period, int gap, int ledcontrol, int numcycle
         else
             SHORT_COIL();
 
+        if (ledcontrol) LED_D_OFF();
+
         //wait until SSC_CLK goes LOW
         while (AT91C_BASE_PIOA->PIO_PDSR & GPIO_SSC_CLK) {
             WDT_HIT();
@@ -636,8 +638,6 @@ void SimulateTagLowFrequencyEx(int period, int gap, int ledcontrol, int numcycle
                 WaitUS(gap);
             }
         }
-
-        if (ledcontrol) LED_D_OFF();
     }
 OUT:
     StopTicks();
--

McEloff, 2019-05-22:
If apply following patch, LED_D would be flashes during transmitting LF-modulation on air. With this patch I detects that one device is responding to second device.
[patch.txt](https://github.com/RfidResearchGroup/proxmark3/files/3209037/patch.txt)

iceman1001, 2019-05-22:
About the same test setup as I have,   I use two RDV4,  one sim, one read.  
Also testing pm3 sim against a LF reader but its a dodgey reader so I don't trust it that much.

iceman1001, 2019-05-23:
Found the problem.  fixed in  https://github.com/RfidResearchGroup/proxmark3/commit/f8dbf6138ab5e1cf9e8cac91b179050ec77ef726

Simulation works for me now.

closed 2019-05-23 by iceman1001

McEloff, 2019-05-23:
Thanks! Now works!