RfidResearchGroup/proxmark3 #200: DV4 - smartcard reader absent

KarlLovink, 2019-05-22:
I have connected the smartcard reader to my proxmark3 RDV4.
The proxmark3 cannot find the smartcard reader for some reader.

Any idea what could be wrong?

hw version output:
[ Proxmark3 RFID instrument ]

[ CLIENT ]
client: RRG/Iceman

[ PROXMARK RDV4 ]
external flash: present
smartcard reader: absent

[ PROXMARK RDV4 Extras ]
FPC USART for BT add-on support: present

[ ARM ]
bootrom: RRG/Iceman/master/48621159 2019-05-22 15:05:30
os: RRG/Iceman/master/48621159-dirty-unclean 2019-05-22 20:52:36

[ FPGA ]
LF image built for 2s30vq100 on 2019/ 5/22 at 15: 6: 9
HF image built for 2s30vq100 on 2019/ 5/22 at 15: 6:26

[ Hardware ]
--= uC: AT91SAM7S512 Rev B
--= Embedded Processor: ARM7TDMI
--= Nonvolatile Program Memory Size: 512K bytes, Used: 250825 bytes (48%) Free: 273463 bytes (52%)
--= Second Nonvolatile Program Memory Size: None
--= Internal SRAM Size: 64K bytes
--= Architecture Identifier: AT91SAM7Sxx Series
--= Nonvolatile Program Memory Type: Embedded Flash Memory

iceman1001, 2019-05-22:
Out from: hw status

KarlLovink, 2019-05-22:
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2019/ 5/22 at 15: 6:26          
#db# Flash memory           
#db#   Baudrate................24MHz          
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256kb           
#db#   Unique ID...............0xd567a882a724a226          
#db# Smart card module (ISO 7816)           
#db#   version.................FAILED           
#db# LF Sampling config           
#db#   [q] divisor.............95 (125 kHz)          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............15*8 (120)          
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......326144          
#db#   Transfer Speed PM3 -> Client = 652288  bytes/s          
#db# Various           
#db#   MF_DBGLEVEL.............1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mode           
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)          
#db# Flash memory dictionary loaded

iceman1001, 2019-05-22:
Did you change anything in the makefiles?
Did you open your case and when you put a card in the sim module it became loose?

KarlLovink, 2019-05-22:
Did you change anything in the makefiles?
Nope
Did you open your case and when you put a card in the sim module it became loose?
Feels a bit loose yes.

iceman1001, 2019-05-22:
unpower rdv4, unplug module and connect again,   and power up pm3,  and see what the client says.
you are using fullimage/client from the same build?

KarlLovink, 2019-05-22:
unpower rdv4, unplug module and connect again, and power up pm3, and see what the client says.
Did that already several times. No result

you are using fullimage/client from the same build?
Yes.

iceman1001, 2019-05-22:
you don't have anything in the sim module now?

iceman1001, 2019-05-22:
got a picture of your setup?

KarlLovink, 2019-05-22:
No and yes. 
I tried it with and without the SIM / SAM Reader Module connected.
Also with and without a SIM card inserted.
Same results with every test.

iceman1001, 2019-05-22:
picture of setup?

KarlLovink, 2019-05-22:
Yes. Here they are
[image]
[image]

iceman1001, 2019-05-22:
backside of the sim module ?

doegox, 2019-05-22:
Would attempting to reflash the SIM module help?
```
sc upgrade f ../tools/simmodule/SIM011.BIN 
````

iceman1001, 2019-05-22:
I'm afraid our dynamic commands might not show the command anymore

iceman1001, 2019-05-22:
You could try change  line 1133 in client/cmdsmartcard.c  
from **ifpm3smartcard**  to **AlwaysAvailable**
compile the client again,  run it,  and try the upgrade command @doegox  just mentioned

KarlLovink, 2019-05-23:
Ok I have recompiled the client the AlwaysAvailable. Command still does not work. None of the smartcard command does not work because I, in my opinion, I cannot find the smartcard module.
          
[usb] pm3 --> sc
[!] This command is not available in this mode          
[[usb] pm3 --> sc upgrade f tools/simmodule/SIM011.BIN 
[!] This command is not available in this mode          
[usb] pm3 --> 

Furthermore I downgrade the proxmark3 with a version without dynamic menus and tried to upgrade it the simmodule with SIM011.BIN.

██████╗ ███╗   ███╗ ████╗     ...iceman fork
██╔══██╗████╗ ████║   ══█║      ...dedicated to RDV40
██████╔╝██╔████╔██║ ████╔╝
██╔═══╝ ██║╚██╔╝██║   ══█║    iceman@icesql.net
██║     ██║ ╚═╝ ██║ ████╔╝  https://github.com/iceman1001/proxmark3
╚═╝     ╚═╝     ╚═╝ ╚═══╝ pre v4.0

Keep iceman fork alive with a donation!           https://paypal.me/iceman1001/
MONERO: 43mNJLpgBVaTvyZmX9ajcohpvVkaRy1kbZPm8tqAb7itZgfuYecgkRF36rXrKFUkwEGeZedPsASRxgv4HPBHvJwyJdyvQuP

[=] UART Setting serial baudrate 460800

Proxmark3 RFID instrument
          

 [ CLIENT ]          
 client: iceman build for RDV40 with flashmem; smartcard;  
          
 [ ARM ]
 bootrom: iceman/master/98e24013-dirty-unclean 2018-11-05 23:12:46
      os: iceman/master/98e24013-dirty-unclean 2018-11-05 23:16:04

 [ FPGA ]
 LF image built for 2s30vq100 on 2018/11/ 5 at 23:15:38
 HF image built for 2s30vq100 on 2018/11/ 5 at 23:15:53

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 241092 bytes (46%) Free: 283196 bytes (54%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          
          
pm3 --> sc upgrade f /tmp/SIM011.BIN
[!] WARNING - Smartcard socket firmware upgrade.          
[!] A dangerous command, do wrong and you will brick the smart card socket          
[+] Smartcard socket firmware uploading to PM3          
..
[+] Smartcard socket firmware updating,  don't turn off your PM3!          
#db# FW 0000          
#db# Writing failed          
[-] Smartcard socket firmware updating failed          
pm3 --> sc
help             This help          
list             List ISO 7816 history          
info             Tag information          
reader           Act like an IS07816 reader          
raw              Send raw hex data to tag          
upgrade          Upgrade firmware          
setclock         Set clock speed          
brute            Bruteforce SFI          
pm3 --> sc info
[!] smart card select failed

iceman1001, 2019-05-23:
Hm,  in this case I would say either it is not connected to its socket correct or you got a fault simmodule,  which in that case I would suggest your RMA the module.   Where did you buy it?

KarlLovink, 2019-05-29:
I was one of the backers on kickstarter. So I think directly from Proxgrind.

iceman1001, 2019-05-29:
Give dennis at rfidresearchgroup com  an email and link to this issue in it.

iceman1001, 2019-07-28:
@KarlLovink   Has your issues been solved?

KarlLovink, 2019-07-28:
@iceman1001, sorry..... The issue has been fixed. Dennis has sent a new one. Upgrade the fw and everything is working fine now. Thanks for your support.

closed 2019-07-28 by KarlLovink