RfidResearchGroup/proxmark3 #204: Unable to extract commands help text without being connected

midnyteshade, 2019-05-25:
In the past one could extract the help text of a command without being connected, but with some of the recent offline changes it seems that changed and I'm unable to do so anymore. I.E. if I execute "hf mf chk h", chk is not available offline, but I only want to see the help on it.

iceman1001, 2019-05-25:
Dynamic commands,  dunno know if there is a good solution for it. Since not all commands has -h / --help / h parameter and in some cases **h** has a different meaning.

You can still get a complete list but its not the same as for one command you are interested in.
```
./proxmark3 -t
./proxmark3 -m 
```

midnyteshade, 2019-05-25:
Thanks for the reply, I'll see what i can come up with. Trying to keep the universal GUI working with the latest code. :)

iceman1001, 2019-06-07:
We have a plan. If device is offline,  we make the usage_help function (if it exists) can be executed instead of the actual command.   It will make the command tables a bit more complicated but solves your issue.  I understand that you need to call help texts in order to make the universal GUI to be able to extract all command and try to figure out how to use it.

midnyteshade, 2019-06-14:
That sounds great, thanks for keeping this in mind and formulating a solution.

midnyteshade, 2019-09-21:
Hi guys, any ETA on this?

iceman1001, 2019-09-21:
Noone at the moment.  Feel free to contribute!

burma69, 2019-11-03:
Clean help syntax and offline help availability is much appreciated :)

I'm trying to make it work online, but i'm not satisfied with the stability so far. Plus online help queries will spam the logfile

iceman1001, 2020-06-04:
No much going on here.  The log file is since a few months back divided into daily logs.  
No solution at hand.   Maybe time to close?

midnyteshade, 2020-07-24:
Yes you can close this. It will need more investigation into a possible solution sometime in the future to get it to work.

closed 2020-07-24 by iceman1001

doegox, 2021-04-19:
Guess what... we have it now :)
`client/proxmark3 --fulltext`
layout may change but the mechanism is there